arendst/Tasmota #21766: DNS and EQ3 problems with newer firmwares

Smandurlo, 2024-07-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have 5 ESP32 devices wired to Mitsubishi AC, controlling some EQ3 TRVs and reading temperature and humidity from BLE Mi sensors.

Tasmota 13.1 works perfectly.

However, Tasmota 13.4 introduces DNS issues: occasionally, the DNS settings change to ipaddress4 253.0.0.0 and ipaddress5 137.2 something else (I didn't write it down, sorry). I tried changing the DNS settings manually, but the issue persisted intermittently. With incorrect DNS, the time was inaccurate and the EQ3 devices weren't properly scanned. When DNS were correct, everything worked fine.

Yesterday, I upgraded to 14.1, which presented a different DNS problem. I set DNS with ipaddresses4 and ipaddress5 with my DNS and after restarting, everything initially seemed fine (correct time and DNS resolution). However, the DNS addresses showed as 0.0.0.0 while asking for them with ipaddress4 and ipaddress5. Additionally, the EQ3 devices stopped working; Tasmota could read their values, but the console always displayed FAILCONNECT, and I couldn't operate them.

I tried to hardcode the DNS inside the firmware, still no luck.

Reverting to 13.1 resolves all issues and everything works fine: I can see my correct dns and I can operate the EQ3 (DONENOTIFIED)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used: Wemos ESP32 clone
- [ ] Tasmota binary firmware version number used: 13.1 13.4 and 14.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/clima1/RESULT = {"NAME":"Wemos D1 Mini ESP32","GPIO":[0,0,1,0,1,1,0,0,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,0,0,0,0],"FLAG":0,"BASE":1}
stat/clima1/RESULT = {"Module":{"0":"Wemos D1 Mini ESP32"}}
stat/clima1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"5856":"MiEl HVAC Tx"},"GPIO17":{"5888":"MiEl HVAC Rx"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

10:31:00.591 MQT: stat/clima1/STATUS = {"Status":{"Module":0,"DeviceName":"Clima1","FriendlyName":["Clima1"],"Topic":"clima1","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:31:00.598 MQT: stat/clima1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T01:03:21","StartupUTC":"2024-07-13T07:27:39","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2023-10-15T21:55:27","SaveCount":35}}
10:31:00.604 MQT: stat/clima1/STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota32)","BuildDateTime":"2023-10-06T12:30:57","Core":"2_0_11","SDK":"4.4.5.230722","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"407/699"}}
10:31:00.610 MQT: stat/clima1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VillaTorreZisa",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C80001000600003C5A0A192800000000","000000C0","00006000","10004002","00000000"]}}
10:31:00.636 MQT: stat/clima1/STATUS4 = {"StatusMEM":{"ProgramSize":2151,"Free":728,"Heap":59,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","4C0840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,44,50,52,59,60,62,63,66,67,68,79,82,85,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
10:31:00.645 MQT: stat/clima1/STATUS5 = {"StatusNET":{"Hostname":"clima1-4756","IPAddress":"192.168.178.91","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"8.8.8.8","Mac":"A8:42:E3:CA:32:94","IP6Global":"2001:b07:2ec:6c40:aa42:e3ff:feca:3294","IP6Local":"fe80::aa42:e3ff:feca:3294","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.178.1","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
10:31:00.653 MQT: stat/clima1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.26","MqttPort":1883,"MqttClientMask":"clima1","MqttClient":"clima1","MqttUser":"Felice","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:31:00.660 MQT: stat/clima1/STATUS7 = {"StatusTIM":{"UTC":"2024-07-13T08:31:00","Local":"2024-07-13T10:31:00","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"05:39","Sunset":"20:41"}}
10:31:00.671 MQT: stat/clima1/STATUS10 = {"StatusSNS":{"Time":"2024-07-13T10:31:00","MiElHVAC":{"Temperature":28.5,"Operation":"OFF","Compressor":"OFF","roomtemp":"0300001200B8B9B9FE42000238D70000","status":"06000000000008010700004200000000","stage":"09000000020000000000000000000000"},"TempUnit":"C"}}
10:31:00.682 MQT: stat/clima1/STATUS11 = {"StatusSTS":{"Time":"2024-07-13T10:31:00","Uptime":"0T01:03:21","UptimeSec":3801,"Heap":56,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":73,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":42},"POWER":"ON","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"E8:DF:70:9D:6F:7B","Channel":6,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:10"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

just update the firware from 13.1 to 13.4 and to 14.1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I didn't check with the latest dev. I compiled the firmwares with tasmocompiler and:

#define USE_MIEL_HVAC
#define USE_MI_ESP32
#define USE_BLE_ESP32
#define USE_EQ3_ESP32

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-07-13:
There was a big networking change with Core3 and I suspect that setting manually DNS was not fully tested. It's a rare setup, we generally advise to configure DNS via DHCP, it's the most tested path. 

I can't troubleshoot right now. Please be patient

Smandurlo, 2024-07-13:
thank you for your reply! I usually use the DHCP dns, I ended up changing them manually because they didn't work properly on my ESP32 on 13.4. They were auto from DHCP, they started to behave that way after the upgrade and the manual setting was just to try to fix it. They work fine with DHCP with all my esp8266.

Jason2866, 2024-07-13:
Have you tested DNS via DHCP with latest version or not? This is not clear for me from the text you wrote.

Smandurlo, 2024-07-13:
yes, when I try ipaddress4 0.0.0.0 to use the dhcp dns, restart , but I still see 0.0.0.0. Anyway it seems they work because time is correct and I can ping google.com for example, so dns is apparently resolved.

Jason2866, 2024-07-13:
> but I still see 0.0.0.0

As designed and expected. 0.0.0.0 is the setup for DHCP and is not overriden.

Smandurlo, 2024-07-13:
O_O this is a bit confusing.
ipaddress1 2 and 3 show the addresses the DHCP gives.

ipaddress4 and 5 on esp8266 14.1 still show the DNS assigned by the DHCP server. I don't see 0.0.0.0

Anyway, ipaddress4 14.1 on esp32 shows 0.0.0.0 when I try to manual change them, so, again, it is confusing. I can't see what dns are actually used.

I think they should work as esp8266 and show the current dns used, including the dhcp dns.

Do you know anything about EQ3 not working?

Jason2866, 2024-07-13:
So you have guessed that maybe the non working BLE EQ3 is a DNS problem?
Since the underlying BLE NimBLE stack has changed with some API changes it could be that this part is not working anymore.

Since the "core" developers do not use or have BLE EQ3 devices, the driver is not tested and not known if working with actual core.

@btsimonh Do you have BLE EQ3 devices?

Smandurlo, 2024-07-13:
with wrong DNS they are not working properly for a different reason: they weren't properly scanned. So, yes, I thought something could be connected somehow to the dns problem.

I really didn't know the ble stack changed :-)

Jason2866, 2024-07-13:
Looking in Tasmota Discussions controlling the EQ3 was always problematic and never a rock solid thing. Probably it is a good idea to stay at the firmware version where this special task is working well.

MaxVonEvil, 2024-07-14:
Folks, might have stumbled over a related issue here:

In Tasmota 14.1.0 It seems the Websend command will not resolve a fqdn, even though the documentation](https://tasmota.github.io/docs/Commands/#websend) says it can. I can only get it to work by specifying an IP address. FYI: I have a node-red Http In node listening on tcp/1880. A single internal DNS (a pihole) set up via DHCP

2) **This works:** WebSend [10.x.y.z:1880]/endpoint/longpress?location=officemax&switch=m

Result:
18:20:36.650 CMD: WebSend [10.x.y.z:1880]/endpoint/longpress?location=officemax&switch=m
18:20:36.901 MQT: stat/t_wallsw_office_max/RESULT = {"WebSend":"Done"}

3) **This doesn't:** WebSend [ha.mydomain.local:1880]/endpoint/longpress?location=officemax&switch=m

Result: 
18:28:32.909 CMD: WebSend [ha.mydomain.local:1880] /endpoint/longpress?location=officemax&switch=m
18:28:43.109 MQT: stat/t_wallsw_office_max/RESULT = {"WebSend":"Connect failed"}

Isn't the above supposed to work, or am I doing something wrong?

PS: Tasmota apparently doesn't like if the MQTT broker is specified by a FQDN either. That's something I can deal with, but it kinda falls in the same bucket.

Noschvie, 2024-07-14:
> MQTT broker is specified by a FQDN

For me this feature is working, using pfSense as DNS and DHCP.

Jason2866, 2024-07-14:
mDNS is not supported `mydomain.local` and will not be implemented in Tasmota.

MaxVonEvil, 2024-07-14:
So, while apparently being marked off topic, I have to state for the record that I'm not using multicast dns, I'm using a pihole with a proper dns responding on tcp/53. It's IP is displayed as expected on the info page.

*** To whoever is moderating this thread, it seemed related, so what's the play here? Do you prefer this opened as a separate issue?

Jason2866, 2024-07-14:
@MaxVonEvil yes, but useless since your log shows clearly `.local`. This is mDNS.
If you use real DNS NEVER use this extension.

MaxVonEvil, 2024-07-14:
Ah, that would explain it. I'll change .local to something else and give it a whirl.
Now back to your regular scheduled discussions ;)

btsimonh, 2024-07-15:
I've had successful operation of BLE with EQ3 with 14+ dev branch.
It was a side effect of what I was doing (working on webcam), and since BLE and webcam can't really play nice because of the network load of streaming video, I'm not running BLE at the moment.  But I did use EQ3 BLE success or failure as a measure of whether BLE was working or not (and it did....).
I'm not sure how DNS could affect BLE, unless the DNS config is causing a LOT of wifi packets when it's not working right.

Smandurlo, 2024-07-15:
thank you for you reply. I will check on it whit the next rel. I need my AC to work properly at the moment and I am not into playing with dev now :-p

DNS problem doesn't affect the BLE, but it the esp32 cannot  connect to internet, the esp32 doesn't activaly scan the EQ3. It is like some sort of "cron" doesn't start. I thought it could be related to it.

Smandurlo, 2024-07-16:
so, the DNS problem is not a real problem (dhcp dns on esp32 will be shown as 0.0.0.0 and on esp8266 will be shown as current dhcp dns) and EQ3 seems to be fixed in the latest dev version.

Is it correct?

Jason2866, 2024-07-16:
@Smandurlo  The author of the EQ3 driver has it working with actual Tasmota. So yes.

Smandurlo, 2024-07-16:
then it is fixed and the dns is not a real issue. Closed :-)

closed 2024-07-16 by Smandurlo