arendst/Tasmota #21776: TelePeriod funktioniert nicht bei IR-Leseköpfen

Steve2901, 2024-07-15:
Bei meinem ISKRA 175 Stromzähler werden die Stromdaten mehrfach pro Sekunde aktualisiert (Display).... also so 1-3x.... kann ich nicht genau sagen wie oft....

Problem was ich habe ist, dass mein MQTT Server im Sekundentakt (oder noch öfters) mit den Daten zugebombt wird.
Mit TelePeriod 10....30 etc habe ich das schon versucht, es hat nichts gebracht. Die Daten werden im Sekundentakt an meinen MQTT Server übertragen und da ich im ioBroker das Topic aboniert habe werden die Datenpunkte auch ständig im Sekundentakt aktualisiert. ISt das so gewollt oder ein Bug? Kann man nicht einstellen in welchem Bereich er die Daten Senden soll?

Wäre super wenn es da ne Lösung gäbe. Verwendet wird Tasmota 14.1.0 auf einem Wemos D1 Mini.

Ich danke schonmal herzlichst für die Bearbeitung des Problems.

VG Steve

[image]
[image]

sfromis, 2024-07-15:
TelePeriod is for telemetry messages, which are automatically published by Tasmota irrespective of what happens. This does **not** apply to any message having the `tele/` prefix, and IR messages are generally based on something being received.

I don't know if this applies to what you'e seeing, but the meter metrics does have an indicator to force immediate publish (or not) of data, instead of at TelePeriod.
https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-metrics

Steve2901, 2024-07-15:
So it wouldn't be a bad thing if the ESP records the data from the counter every second, which is a good thing, because if data is flowing then the ESP should "process" it. I am actually concerned with the transmission interval to the MQTT server. I'm missing an option where you can set that the data is only sent after a certain time. So if I set 3 seconds, then everything that was received before the 3 seconds should be discarded and only the value that is in the ESP's memory at 3 seconds should be sent to the MQTT server. I think this would not be a huge effort to implement. I don't have to constantly bombard my MQTT server or I don't need the data from the MQTT server in the ioBroker every second. Would be really great if there was something like this to set.... Is this feasible?

Translated with DeepL.com (free version)

Steve2901, 2024-07-15:
Soweit ich weiß kann man das bei der Software
"AI-on-the-edge-device" auch machen. Da kann man einstellen in welchem Intervall er den Zähler auslesen soll, die Daten verarbeiten soll und dann senden soll. Sowas wäre in Tasmota bei MQTT super :)

As far as I know, you can do this with the software
"AI-on-the-edge-device" software. You can set the interval at which it should read the counter, process the data and then send it. Something like that would be great in Tasmota with MQTT :)

[image]