arendst/Tasmota #21778: 14.1.0 and 14.1.0.3 only finds 1 or 2 accesspoints when scanning for networks and won't connect

jeroenst, 2024-07-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I upgrade from 13.4.0 to 14.1.0.3 Tasmota won't connect again. Wifi scan only shows 1 or 2 networks.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NODEMCU
- [x] Tasmota binary firmware version number used: 13.4.0 and 13.4.0.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: https://tasmota.github.io/install/
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
0:03:45.992 SRC: Serial
00:03:45.995 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
00:03:46.040 SRC: Backlog
00:03:46.042 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
00:03:46.048 RSL: RESULT = {"NAME":"IR","GPIO":[1,1,1,1,1,1,1,1,1,1,1056,1,1,1],"FLAG":0,"BASE":18}
00:03:46.290 SRC: Backlog
00:03:46.293 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
00:03:46.297 RSL: RESULT = {"Module":{"0":"IR"}}
00:03:46.491 WIF: Attempting connection...
00:03:46.541 SRC: Backlog
00:03:46.544 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
00:03:46.552 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1056":"IRsend1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
00:03:47.493 WIF: Connect failed with AP timeout

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:04:35.466 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
00:04:35.473 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"home/TASMOTA-IR-BEDROOM","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":1,"PowerRetain":1,"InfoRetain":1,"StateRetain":1,"StatusRetain":1}}
00:04:35.509 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin.gz","RestartReason":"External System","Uptime":"0T00:04:37","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":69,"BCResetTime":"2022-06-16T14:19:24","SaveCount":174,"SaveAddress":"F9000"}}
00:04:35.539 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(ir)","BuildDateTime":"2024-06-05T08:23:19","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"369/699"}}
00:04:35.557 RSL: STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["J&M IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["002082E9","2805C80001000600003C5A0A002800000000","000000E8","00206000","80004180","00000000"]}}
00:04:35.586 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":662,"Free":340,"Heap":33,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080","00000000"],"Drivers":"1,2,!4,!5,7,9,10,12,62,!68","Sensors":"1"}}
00:04:35.616 RSL: STATUS5 = {"StatusNET":{"Hostname":"TASMOTA-IR-BEDROOM","IPAddress":"0.0.0.0","Gateway":"172.30.30.254","Subnetmask":"255.255.254.0","DNSServer1":"172.30.30.254","DNSServer2":"0.0.0.0","Mac":"A4:E5:7C:B8:7B:C8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:04:35.640 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.jst-it.nl","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B87BC8","MqttUser":"jeroen","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:04:35.662 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:04:35Z","Local":"1970-01-01T00:04:35","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"20:13","Sunset":"05:47"}}
00:04:35.679 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:04:35"}}
00:04:35.686 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:04:35","Uptime":"0T00:04:37","UptimeSec":277,"Vcc":2.981,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.002 HDW: ESP8266EX
00:00:00.051 CFG: Loaded from flash at FA, Count 161
00:00:00.056 QPC: Count 1
00:00:00.061 SRC: Restart
00:00:00.062 Project tasmota - Tasmota Version 14.1.0(release-ir)-2_7_6(2024-06-03T11:53:03)
00:00:00.115 WIF: Attempting connection...
00:00:00.118 WIF: Network (re)scan started...
00:00:01.061 WIF: Attempting connection...
00:00:02.063 WIF: Attempting connection...
00:00:03.068 WIF: Attempting connection...
[... 104 lines cut ...]
                    ESSID:"J&M Gasten"
          Cell 26 - Address: 48:29:52:59:52:8A
                    Frequency:5.24 GHz (Channel 48)
                    ESSID:"P&G"
          Cell 27 - Address: F0:9F:C2:D8:2E:7A
                    Frequency:5.24 GHz (Channel 48)
                    ESSID:"J&M"
          Cell 28 - Address: F6:9F:C2:D8:2E:7A
                    Frequency:5.24 GHz (Channel 48)
                    ESSID:"J&M Gasten"
          Cell 29 - Address: 6C:5A:B0:31:D9:4A
                    Frequency:2.427 GHz (Channel 4)
                    ESSID:"zooitjeongeregeld_Guest"
          Cell 30 - Address: 30:E1:71:4A:17:DE
                    Frequency:2.452 GHz (Channel 9)
                    ESSID:"DIRECT-DC-HP ENVY 5540 series"
          Cell 31 - Address: 48:29:52:59:52:86
                    Frequency:2.452 GHz (Channel 9)
                    ESSID:"P&G"
```

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-07-15:
Well, when I upgrade (including now to 14.1.0.3), I'm not seeing any connection issues on my network, and all my AP are popping up when scanning. Your network may be different, but I'm not seeing some sort of inherent issue with Tasmota.

jeroenst, 2024-07-15:
Strange, I had to downgrade to 13.4.0 otherwise I don't get any connection.

What information can I provide for debugging?

Jason2866, 2024-07-15:
Whitespaces and Special characters are not supported in SSID and Password with Tasmota.
Can make issues. Idk if it is the case here. Just where I would start.

sfromis, 2024-07-15:
In this case, the SSID above is `J&M IOT`.

However, I can admit to using a SSID including a space, but in some contexts the complexities of multiple levels of parsing and escaping does mean that such characters does increase the risk of trouble, which can be hard to debug when connections fail.

jeroenst, 2024-07-15:
But these are valid ssid names which can be in the air and cannot airways be controlled so imo they should not compromise the connection of tasmota. 

Beside that it works fine with tasmota 13.4.0 and prior versions.

The other strange thing is that tasmota minimal 14.1.0.3 connected without any issue.

So I think it's a configuration option.

I will try a device from scratch to see if I can find which config option causes this issue.

jeroenst, 2024-07-16:
I just fully erased and flashed a nodemcu with tasmota 14.1.0 and it connects to WiFi without any problem.

So it has to be a setting that breaks wifi.

Jason2866, 2024-07-16:
Probably wifi calibration data.

jeroenst, 2024-07-16:
I did a reset 3 for that, but it didn't make any difference

jeroenst, 2024-07-16:
I copied the configuration of the nodemcu that didn't connect to wifi after upgrading to the erased nodemcu and it works fine.

jeroenst, 2024-07-16:
Trying to upgrade another device results in the same issue. Upgrade from 13.4.0 to 14.1.0 is not stable.

closed 2024-07-16 by jeroenst

jeroenst, 2024-08-21:
I had 3 other devices that had this issue when upgrading to 14.1.0, so something is wrong...