arendst/Tasmota #21788: LVGL/haspmota and edisplay (somehow missing refresh)

timehlers, 2024-07-17:
### PROBLEM DESCRIPTION
Hi,

I tried LVGL/hspmota with my edisplay and at first I got nothing working whatever I tried. I always got a white display. By chance I found out that I get the expected output, when I use "DisplayText[]" in the console, means I get the Text from the command + the missing text from LVGL. So I read a bit in the code, and I think for some reason the epaper displays need a "Updateframe()", while normal displays don't need that for drawing the content.

If I add in tasmota/tasmota_xdrv_driver/xdrv_54_lvgl.ino in the main screen refresh function "lv_flush_callback":

renderer->Updateframe();

as last action in the function, I finally see the spash screen and the configured output. I am not sure, if this is enough, breaks something or is the correct place to put the Updateframe().

What do you think? Has somebody an epaper running with haspmota and I am missing something else?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-C3
- [x] Tasmota binary firmware version number used: 13.4.0 and 14.1.0.
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py and firmware update function
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:07:15.176 RSL: RESULT = {"NAME":"ESP32C3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
11:07:15.382 RSL: RESULT = {"Module":{"1":"ESP32C3"}}
11:07:15.585 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"832":"SSPI MISO"},"GPIO2":{"1024":"Display Rst"},"GPIO3":{"928":"SSPI CS"},"GPIO4":{"896":"SSPI SCLK"},"GPIO5":{"864":"SSPI MOSI"},"GPIO6":{"6210":"Option A3"},"GPIO7":{"928":"SSPI CS"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:08:47.636 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:08:47.845 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:08:48.046 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:09:24.073 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"esp32_869EA8","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:09:24.108 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"esp32","OtaUrl":"http://XXX","RestartReason":"Vbat power on reset","Uptime":"0T00:02:26","StartupUTC":"2024-07-17T10:06:58","Sleep":50,"CfgHolder":4617,"BootCount":384,"BCResetTime":"2024-06-29T13:19:22","SaveCount":586}}
11:09:24.142 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(tasmota32)","BuildDateTime":"2024-07-16T15:33:33","Core":"3_0_0","SDK":"5.1.4.240529","CpuFrequency":160,"Hardware":"ESP32-C3 v0.4","CR":"441/699"}}
11:09:24.157 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.0.85","LogPort":514,"SSId":["XXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004002","00000000"]}}
11:09:24.200 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2839,"Free":40,"Heap":52,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9FBA17DF","0011A021","B7F7BFCF","05DA9BC4","22360C47","480840D2","20200000","84BC482D","C10A80B1","00000015"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,13,!14,!16,!17,!20,!21,!24,29,!34,38,50,52,54,!59,!60,62,!63,!66,!67,!68,!71,!73,79,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
11:09:24.272 RSL: STATUS5 = {"StatusNET":{"Hostname":"esp32-869EA8-7848","IPAddress":"192.168.0.144","Gateway":"192.168.0.85","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"80:65:99:86:9E:A8","IP6Global":"","IP6Local":"fe80::8265:99ff:fe86:9ea8%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
11:09:24.321 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.85","MqttPort":1883,"MqttClientMask":"ESP32_%06X","MqttClient":"ESP32_869EA8","MqttUser":"","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:09:24.338 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-07-17T10:09:24Z","Local":"2024-07-17T11:09:24","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:05","Sunset":"20:46"}}
11:09:24.366 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-07-17T11:09:24"}}
11:09:24.372 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-07-17T11:09:24","Uptime":"0T00:02:26","UptimeSec":146,"Heap":50,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":100,"MqttCount":0,"Berry":{"HeapUsed":6,"Objects":63},"POWER":"ON","Wifi":{"AP":1,"SSId":"XXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":3,"Mode":"HT20","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
11:13:19.695 HTP: Berry Console
11:13:21.956 LVG: LVGL was already initialized
11:13:21.999 BRY: GC from 13108 to 8632 bytes, objects freed 57/82 (in 2 ms) - slots from 236/259 to 93/244
11:13:22.002 HSP: parsing line '{"page":1,"id":0,"bg_color":"#000000","bg_grad_color":"#000000","bg_grad_dir":1,"text_color":"#FFFFFF"}'
11:13:22.023 HSP: parsing line '{"page":1,"id":10,"obj":"label","x":1,"y":15,"w":140,"text":"Test","align":0}'
11:13:22.062 HSP: parsing line '{"page":1,"id":11,"obj":"label","x":1,"y":30,"w":140,"text":"Test2","align":0}'
11:13:22.099 HSP: parsing line '{"page":1,"id":110,"obj":"line","x":1,"y":33,"h":2,"line_width":238,"line_color":"#FFFFFF"}'
11:13:22.112 BRY: GC from 17312 to 9759 bytes, objects freed 112/120 (in 3 ms) - slots from 355/366 to 128/366
11:13:22.143 HSP: parsing line '{"page":1,"id":110,"obj":"line","x":1,"y":50,"h":2,"line_width":238,"line_color":"#000000"}'
11:13:22.211 LVG: Refreshed 11840 pixels in 11 ms (1076 pix/ms)
11:13:22.225 LVG: Refreshed 11840 pixels in 10 ms (1184 pix/ms)
11:13:22.241 LVG: Refreshed 11840 pixels in 14 ms (845 pix/ms)
11:13:43.862 CMD: DisplayText[]Test
11:13:43.864 SRC: WebConsole from 192.168.0.135
11:13:43.865 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 6, Pld -99, Data '[]Test'
11:13:44.819 RSL: RESULT = {"DisplayText":"[]Test"}

```

### TO REPRODUCE
- configure uDisplay with epaper on a binary with LVGL and haspmota
- write pages.jsonl in the filesystem with some test text.
- start Haspmota in berry with:
```lua
import haspmota
haspmota.start()
```
- screen stays white
- go to console and write "DisplayText[]Test" and the Test line, including the text from haspmota will appear

### EXPECTED BEHAVIOUR
Get content of LVGL/haspmota as soon as started.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-07-17:
Testing....

arendst, 2024-07-17:
I see the same.

Let's add your suggestion like:
```
  uint32_t pixels_len = width * height;
  uint32_t chrono_start = millis();
  renderer->setAddrWindow(area->x1, area->y1, area->x1+width, area->y1+height);
  renderer->pushColors((uint16_t *)color_p, pixels_len, true);
  renderer->setAddrWindow(0,0,0,0);
  renderer->Updateframe();
  uint32_t chrono_time = millis() - chrono_start;

  lv_disp_flush_ready(disp);

  if (pixels_len >= 10000 && (!renderer->lvgl_param.use_dma)) {
    if (HighestLogLevel() >= LOG_LEVEL_DEBUG_MORE) {
      AddLog(LOG_LEVEL_DEBUG_MORE, D_LOG_LVGL "Refreshed %d pixels in %d ms (%i pix/ms)", pixels_len, chrono_time,
              chrono_time > 0 ? pixels_len / chrono_time : -1);
    }
  }
```
and see if others notice any trouble with it. As far as I can tell the ``renderer->Updateframe()`` function specifically tests for EPS screens and does perform the update. Other screens seems to be unaffected....

timehlers, 2024-07-18:
Hi, ok.

So you also bought one of the WeAct displays, nice. :) The speed is not the best. I have 2.9 inch Weact displays, too. They are fully compatible with the WS_epaper29_v2_display.ini file. From feeling they seem to be the same speed, maybe a bit quicker, but they are also smaller.

With the splash screen I see now, that the Logo is still a bit visible on the 4.2 inch display after wiping the splash. Don't know if that can be tuned with the ini-file. The controller seems not to allow voltage tuning, like the 2.9 inch V2 does.

I'll close the issue here for now.

closed 2024-07-18 by timehlers