arendst/Tasmota #21793: Sonoff 4Ch Pro Toggles on and right off

boilermanc, 2024-07-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have flashed the latest bin onto a Sonoff 4Ch Pro. A little complicated but pretty amazing that really smart people figured that out. When I choose the Sonoff 4Ch Pro config get a config setup with only 3 gpios. However, it will create the 4 toggle buttons. When I toggle the button on, it immediately goes off.  I have tried about every command that I can find. So I cannot set the gpios on the provided template since its not the right template. I tried to create a generic template and no joy with that as well. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____Sonoff 4Ch Pro
- [ ] Tasmota binary firmware version number used: _____[Tasmota 14.1.0 (release-tasmota)](https://bit.ly/tasmota)
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: _____that little device that you plug into rx and tx and then grounded across with a wire etc...
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
4:21:17.048 SRC: Backlog
04:21:17.051 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
04:21:17.059 RSL: RESULT = {"GPIO0":{"17":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"21":"None"},"GPIO5":{"22":"None"},"GPIO9":{"23":"None"},"GPIO10":{"24":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"0":"None"},"GPIO14":{"226":"Relay3"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
 Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
04:22:27.051 SRC: Backlog
04:22:27.054 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
04:22:27.058 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
04:22:27.263 SRC: Backlog
04:22:27.266 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
04:22:27.270 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
04:22:27.514 SRC: Backlog
04:22:27.517 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
04:22:27.521 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
04:22:57.808 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
04:22:57.814 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota_9F4B28","ButtonTopic":"","Power":"100","PowerLock":"000","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
04:22:57.853 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:09:36","StartupUTC":"2024-07-18T03:13:21","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2024-07-18T03:28:40","SaveCount":92,"SaveAddress":"F9000"}}
04:22:57.883 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(release-tasmota)","BuildDateTime":"2024-06-03T11:53:14","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"380/699"}}
04:22:57.901 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["alternateHome2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
04:22:57.936 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
04:22:57.973 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9F4B28-2856","IPAddress":"192.168.12.117","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.12.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:9F:4B:28","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
04:22:57.995 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F4B28","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
04:22:57.017 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-07-18T03:22:57Z","Local":"2024-07-18T04:22:57","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:07","Sunset":"20:45"}}
04:22:57.035 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-07-18T04:22:57"}}
04:22:57.042 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-07-18T04:22:57","Uptime":"0T00:09:36","UptimeSec":576,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"alternateHome2.4","BSSId":"C6:E5:32:9F:A1:46","Channel":2,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I guess flash the same board. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
when you toggle the button on the relay stays on. and when you go to the configuration page for the device you see the correct configuration. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bernardesarthur, 2024-07-21:
Did you check if your device is use PulseTime? This behavior looks like PulseTime. Write PulseTime in your console and post results here.

bernardesarthur, 2024-07-21:
Also you can test this template: `{"NAME":"Sonoff 4CHPROR3","GPIO":[17,255,255,255,23,22,18,19,21,56,20,24,0],"FLAG":0,"BASE":23}`

Jason2866, 2024-07-23:
Not reproducable.