arendst/Tasmota #21794: Tasmota Decies Reboot -> IP SCAN

ChrSch84, 2024-07-18:
Hello 

i have a strange Problem. 
A few of my Tasmota Devices reboot after making a IP Scan in my network. First i thought my device reboot (behind Tasmota) but after a while i found out if i make a ip scan with "http" in the Programm Advanced IP Scanner. The Tasmota devices reboots.

And the Strange thing is that is not the same Hardware! Mix of "Sonoff Pow", "NOUS A1T","EIGHTREE".

Here are the Settings from Advanced IP Scanner.
"
[image]
"

Here a short List of my Devices: not all Reboot!

"
[image]
"

I could not find any hint why they reboot. 

Ideas? Hints?

Thanks for the Help.

------------------------------------------------------------------------------------------------------------

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-07-18:
Primitive tools like IP scanners may hammer the device with calls, and thereby overloading it. Some devices are not able to power sustained "spamming" by bad requests.

ChrSch84, 2024-07-18:
But why not all Devices? And why did it work a few months back. Used this Scanner for Years and had no Problem with Tasmota.

sfromis, 2024-07-18:
Some devices have a better internal power supply than others, and can survive your DOS attack. And this is not about what Tasmota can handle, but about what your hardware can handle. Comparing back in time can be random factors, or just that devices degrade with time.

ChrSch84, 2024-07-18:
- Old and new Devices are affekted.
- ITs an IP Scan not a DDoS Attack on the device.
- Its only if i select http/https all other "DDos" Attacks are fine.

sfromis, 2024-07-18:
An IP scan is effectively a DOS attack. I did not say DDOS.

ChrSch84, 2024-07-18:
ok fine DOS Attack...

Whatever...

but why is the device rebooting. If i make a "ARP" Request from the Router or Switch there is no Reboot. The IP Scan should be the same thing. And as i say its not every Tasmota Device!

So is it possible to Log the event? Or does someone know something about it?

sfromis, 2024-07-18:
Well, that depends on what your crude tool is doing when it scans via http. It may be making a lot of requests. Way different from just an ARP request.

And when it is not every Tasmota device, that just supports my hypothesis that it is not about what Tasmota can handle, but about what weak hardware can handle. Thus, is not an "event" that Tasmota knows about, and even if it could log it, that log would be gone immediately when the device croaks.