arendst/Tasmota #21807: Multiple OTA upgrade fail

Smandurlo, 2024-07-20:
### PROBLEM DESCRIPTION

A couple of weeks ago, I flashed a Shelly 2.5 from 13.1 to 14.1 minimal. After the reboot, it didn't come back online. There was no way to recover it, so I had to flash it again with the serial adapter. Okay, sh*t happens.

Two days ago, I flashed another Shelly 2.5 to 14.1 minimal, and it failed again. I noticed it was very warm, similar to the first one. Using the serial adapter, I saw it was stuck in a reboot loop. I flashed it again with the serial adapter. Coincidence? Maybe...

Now, I tried again: two more Shelly 2.5 devices died after the OTA update to minimal.

I honestly don't want to risk it with other types of devices because they don't have an accessible serial connection unless I solder it, and I don't want to reflash the remaining Shelly 2.5 devices.

What is wrong?

Thanks.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [ ] Tasmota binary firmware version number used: 13.1 -> 14.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-07-20:
When you have the setup for serial flashing, you could also use it to have a serial console via a terminal program, to provide info for debugging.

FTR, I've seen no trouble with minimal on my Shelly 2.5 devices, just upgraded one of them.

arendst, 2024-07-21:
Cannot reproduce either. My Shelly 2.5 updates without issues.

What is your update process?
Do you load the minimal yourselves or do you use the recommended procedure and let tasmota load the minimal version if needed.

Smandurlo, 2024-07-22:
So, I took another shelly 2.5. Flashed with serial adapter and tasmota.bin 14.1 with tasmotize. Configured with shelly 2.5 template. Then I went to the firmware upgrade section, inserted the minimal version otaurl for the OTA upgrade: it didn't return online anymore.

12:01:40.025 MQT: Connected
12:01:40.029 MQT: tele/tv/LWT = Online (retained)
12:01:40.031 MQT: cmnd/tv/POWER = 
12:01:40.502 MQT: tele/tv/STATE = {"Time":"2024-07-22T12:01:40","Uptime":"0T00:00:15","UptimeSec":15,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":236,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"2C:91:AB:67:FB:D1","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:01:41.254 MQT: stat/tv/UPGRADE = {"Upgrade":"Successful. Restarting"}
12:01:41.774 MQT: stat/tv/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Exception","Uptime":"0T00:00:16","StartupUTC":"2024-07-22T10:01:25","Sleep":50,"CfgHolder":4617,"BootCount":90,"BCResetTime":"2021-07-19T12:42:34","SaveCount":1040,"SaveAddress":"FB000"}}
12:01:41.823 MQT: stat/tv/RESULT = {"Time":"2024-07-22T12:01:41","Uptime":"0T00:00:16","UptimeSec":16,"Heap":23,"SleepMode":"Dynamic","S[00]Y‰:50,"LoadAvg":79,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"2C:91:AB:67:FB:D1","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:01:41.874 MQT: stat/tv/STATUS11 = {"StatusSTS":{"Time":"2024-07-22T12:01:41","Uptime":"0T00:00:16","UptimeSec":16,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":79,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"2C:91:AB:67:FB:D1","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}
12:01:42.519 APP: Restarting

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
@cp:0
ld
e:
 ets Jan  8 2013,rst cause:3, boot mode:(3,7)

ets_main.c

arendst, 2024-07-22:
I don't know what's going on here but in the short log you provided there are plenty of things which are not OK.

1 - Before, during and after the upgrade there are status messages requested continually. This is useless and a no-go and probably screwed up the flashing.
2 - In status 1 there is an exception reported before the upgrade is performed. Investigate why this happens.
3 - You manually load ``tasmota-minimal.bin.gz``. Why oh why? There is no need for a user to load minimal. Just set OtaUrl to the correct binary (``http://ota.tasmota.com/tasmota/release/tasmota.bin.gz``) and all should go as planned. Tasmota will load minimal if needed.

I suggest you'll do the following:
1 - Serial flash 14.1 again
2 - Set OtaUrl correctly (``OtaUrl http://ota.tasmota.com/tasmota/release/tasmota.bin.gz``)
3 - Set seriallog to 3 (``Seriallog 3``)
4 - Perform command ``Status 0``
5 - Start logging while executing command ``Upload 1``
6 - Provide log here

sfromis, 2024-07-22:
And make sure to kill whatever tool is doing those extra `Status` command automatically, maybe it is TasmoAdmin.

Smandurlo, 2024-07-22:
Just to clarify before proceeding with the steps @arendst requested: I typically use Tasmocompiler to create my Tasmota binary. Sometimes, I need to add functionalities and, other times, I want to remove unnecessary features like Domoticz, which can clutter the UI and consume memory.

When I have the compiled bin.gz file, I cannot upload it directly due to insufficient space. Therefore, I temporarily modify the OTA URL to the minimal version and then manually upgrade by uploading my bin.gz file.

I'm not sure why those specific statuses are requested as I don't use TasmoAdmin.

If you could tell me what might be causing this issue, I can check my MQTT server or perhaps use some Tasmota commands to diagnose it.

If you @arendst  or @sfromis  could provide the correct procedure for uploading my compiled bin.gz file from Tasmocompiler, I would be happy to try that method. But honestly until now, I haven't encountered this problem.

arendst, 2024-07-22:
OK. Makes sense although as you strip the default binary try to make it as small as possible so it doesn't need the minimal phase at all ;-)

Anyway, if you do not use TasmoAdmin you'll need to look first where the commands come from. EAnable logging level 3 and see the source of the messages. Likely it's MQTT so then find the reason why retained messages where enabled. Then remove these from your MQTT broker.