arendst/Tasmota #21810: XPT2046 not working with universal driver

Jumagiera, 2024-07-21:
Display and touch works fine at Vers. 13.4

But after flashing with Vers. 14.0 or 14.1 only the display works.
While displaycalibrate the display reacts on touches.

Neither the succesful nor failed screen is shown.
Console after calibration:
15:35:57.194 MQT: stat/tasmota_31D608/RESULT = {"DisplayRotate":0}
15:36:05.656 TS : Calibration (-42,-20) - (-42,-20) - (-42,-20) - (-42,-20)
15:36:05.659 BRY: Exception> 'divzero_error' - division by zero
15:36:05.660 stack traceback:
15:36:05.661 : in function calc_geometry
15:36:05.661 : in function finish
15:36:05.662 : in function do_next_corner
15:36:05.663 : in function <lambda>
15:36:05.664 : in function run_deferred
15:36:05.664 : in function event

Console after startup:
00:00:00.080 BRY: Berry initialized, RAM used 5341 bytes
00:00:00.498 UT : XPT2046
00:00:00.498 DSP: Configured display 'ILI9341'
00:00:00.507 Project tasmota - 14.1 Version 14.1.0(release-lvgl-haspmota)-3_0_0(2024-06-03T11:56:02)
00:00:00.514 LVG: LVGL initialized
00:00:00.600 TAP: Loaded Tasmota App 'DisplayCalibrate.tapp'
00:00:01.242 BRY: Successfully loaded 'autoexec.be'
00:00:02.098 WIF: Connecting to AP1 HaB Channel 1 BSSId 7C:FF:4D:9C:6A:4A in mode HT40 as tasmota-31D608-5640...
00:00:04.498 WIF: Connected
00:00:04.758 HTP: Web server active on tasmota-31D608-5640 with IP address 192.168.
…

Configuration:

[image]

Display.ini is from here without changes:
https://github.com/arendst/Tasmota/blob/development/tasmota/displaydesc/ILI9341_xpt_display.ini

Or with changing this line:
:UTI,XPT2046,S1,,-1, #In my display.ini there are two stars (*). They are not shown here.
Or like this:
:UTI,XPT2046,S1,33,-1,36
All three variations don’t work and effects the same as described above.
 
The logs shown that the touchscreen does not work since it provided the same coordinates for all 4 points

gemu2015, 2024-07-21:
universal touch spi controller is currently only supported when on the same bus as the display.
changing this requires some essential changes of several source files and needs to be tested. since i do not have that hardware i will have to rewire an existing display for testing. that will take a while since i am short of time.

in the meantime since the old xpt driver ist still there you may enable this one. it will work with the universal display driver like before.

gemu2015, 2024-07-22:
is fixed now. however depending on your hardware you must adjust the UT descriptor
the berry tap needs the :M section in descriptor, so add a default 

:M,340,3900,300,3800

then use this UT descriptor to start with

:UTI,XPT2046,S1,*,*,*
CPR 00
RT
:UTT
XPT 300
RT
:UTX
MV 0 2
RT
:UTY
MV 2 2
RT
#

the example decriptor from Tasmota/displaydesc is for a very special hardware and can't use the :M scaler

Jumagiera, 2024-07-22:
Great. Many thanks!
 I was just starting to have a deeper look on the code myself. Anyway. Great work.
I will have some time later this week and test it with my hardware.

patpommi, 2024-07-30:
@Jumagiera:
Hi, Have you tested whether it works in the meantime?
Unfortunately, it hasn't worked for me with the CYD yet. With various settings.

Jumagiera, 2024-07-30:
It is working for me.
Here my setup:
{"NAME":"ESP32-SUNTON","GPIO":[6210,1,800,1,448,1,1,1,672,704,736,768,1,1,1,1,0,992,1,1,0,737,1,1,0,0,0,0,705,10944,4768,1,11008,0,0,673],"FLAG":0,"BASE":1}

Display.ini

:H,ILI9341,240,320,16,SPI,1,*,*,*,*,21,*,*,40
:S,2,1,1,0,40,20
:r,1
:I
EF,3,03,80,02
CF,3,00,C1,30
ED,4,64,03,12,81
E8,3,85,00,78
CB,5,39,2C,00,34,02
F7,1,20
EA,2,00,00
C0,1,23
C1,1,10
C5,2,3e,28
C7,1,86
36,1,48
37,1,00
3A,1,55
B1,2,00,18
B6,3,08,82,27
F2,1,00
26,1,01
E0,0F,0F,31,2B,0C,0E,08,4E,F1,37,07,10,03,0E,09,00
E1,0F,00,0E,14,03,11,07,31,C1,48,08,0F,0C,31,36,0F
11,80
29,80
:o,28
:O,29
:A,2A,2B,2C,16
:R,36
:0,48,00,00,05
:1,28,00,00,01
:2,88,00,00,02
:3,E8,00,00,00
:i,20,21
:r,2
:UTI,XPT2046,S2,33,,*
CPR 00
RT
:UTT
XPT 300
RT
:UTX
MV 2 2
DBG 1
RT
:UTY
MV 0 2
DBG 1
RT
:M,3600,300,200,3600
#

Calibration seam not to work correct. The M: numbers are not perfect but it works.

Jumagiera, 2024-07-30:
And the DBG part can be removed of cause

Jumagiera, 2024-07-30:
I`m using 
Program Version	14.1.0.4 (895fbb2-lvgl-haspmota)
Build Date & Time	2024-07-23T13:41:50

sfromis, 2024-07-30:
An important difference to your original template screenshot is now having gpio 33 as "TS SPI CS".

patpommi, 2024-08-02:
With this settings it works for me also.
That´s great!!!
Now i can use my CYDs really!

Thank you folks.
And have a nice weekend!

PR #21814 (2024-07-22): fix resistive Touch xpt for 2 spi busses
## Description:

**Related issue (if applicable):** fixes #21810

also fixes cache rewrite of push colors for lvgl and rgb displays

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.7
  - [x] The code change is tested and works with Tasmota core ESP32 V.3.0.2
  - [x] I accept the [CLA](https://github.com/arendst/ [... cut]