arendst/Tasmota #21819: High packet loss after some time and frequent MQTT reconnects

Andrei-Iosifescu123, 2024-07-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After some time, the tasmota device has high packet loss (around 20-30%) and it frequently reconnects to MQTT.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266-01
- [x] Tasmota binary firmware version number used: 14.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Serial
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:28:32.173 MQT: stat/tasmota_E3B1F0/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:28:32.387 MQT: stat/tasmota_E3B1F0/RESULT = {"Module":{"0":"Generic"}}
19:28:32.642 MQT: stat/tasmota_E3B1F0/RESULT = {"GPIO0":{"256":"Relay_i1"},"GPIO1":{"0":"None"},"GPIO2":{"1216":"AM2301"},"GPIO3":{"32":"Button1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:29:10.956 MQT: stat/tasmota_E3B1F0/STATUS = {"Status":{"Module":0,"DeviceName":"Dormitor Jos","FriendlyName":[""],"Topic":"tasmota_E3B1F0","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:29:10.992 MQT: stat/tasmota_E3B1F0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin.gz","RestartReason":"External System","Uptime":"4T07:35:22","StartupUTC":"2024-07-18T10:53:48","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2024-06-16T15:53:45","SaveCount":135,"SaveAddress":"FB000"}}
19:29:11.002 MQT: stat/tasmota_E3B1F0/STATUS2 = {"StatusFWR":{"Version":"14.1.0(release-tasmota)","BuildDateTime":"2024-06-03T11:53:14","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
19:29:11.023 MQT: stat/tasmota_E3B1F0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.0.132","LogPort":514,"SSId":["Andrei",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","000080C0","00006000","00004000","00000000"]}}
19:29:11.060 MQT: stat/tasmota_E3B1F0/STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
19:29:11.100 MQT: stat/tasmota_E3B1F0/STATUS5 = {"StatusNET":{"Hostname":"tasmota-E3B1F0-4592","IPAddress":"192.168.0.169","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"50:02:91:E3:B1:F0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:29:11.124 MQT: stat/tasmota_E3B1F0/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.132","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E3B1F0","MqttUser":"admin","MqttCount":20,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:29:11.150 MQT: stat/tasmota_E3B1F0/STATUS7 = {"StatusTIM":{"UTC":"2024-07-22T18:29:11Z","Local":"2024-07-22T19:29:11","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:11","Sunset":"20:41"}}
19:29:11.171 MQT: stat/tasmota_E3B1F0/STATUS10 = {"StatusSNS":{"Time":"2024-07-22T19:29:11","AM2301":{"Temperature":26.4,"Humidity":62.5,"DewPoint":18.6},"TempUnit":"C"}}
19:29:11.185 MQT: stat/tasmota_E3B1F0/STATUS11 = {"StatusSTS":{"Time":"2024-07-22T19:29:11","Uptime":"4T07:35:23","UptimeSec":372923,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":20,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Andrei","BSSId":"E4:C3:2A:86:BA:03","Channel":11,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Jul 21 12:42:51 tasmota-E3B1F0-4592 ESP-MQT: Attempting connection...
Jul 21 12:42:51 tasmota-E3B1F0-4592 ESP-MQT: Connect failed to 192.168.0.132:1883, rc -2. Retry in 20 sec
Jul 21 12:43:10 tasmota-E3B1F0-4592 ESP-MQT: Attempting connection...
Jul 21 12:43:10 tasmota-E3B1F0-4592 ESP-MQT: Connected
```

### TO REPRODUCE
_Steps to reproduce the behavior:_ Connect esp8266 to Wi-Fi and wait for a few hours, the device will lose MQTT connection and will reconnect shortly after (turning unavailable in HA).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._ The Device is stable without losing the MQTT connection.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I have another Tasmota device (13.2.0) which does not experience this problem.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-07-22:
Those "steps to reproduce" does not. I'm having many ESP8266 devices connected for weeks without seeing any MQTT connection loss, using Tasmota version 14.1.

Andrei-Iosifescu123, 2024-07-22:
> Those "steps to reproduce" does not. I'm having many ESP8266 devices connected for weeks without seeing any MQTT connection loss, using Tasmota version 14.1.

What router are you using?

sfromis, 2024-07-22:
Google Wifi.

Andrei-Iosifescu123, 2024-07-22:
Do you have a mesh system by any chance? I'm using OpenWrt with 2 APs

sfromis, 2024-07-22:
Yes, Google Wifi is a mesh system. I have 3 nodes.

bernardesarthur, 2024-07-22:
I  also have many devices Tasmota like @sfromis and I use **UniFi Network here with no problems**. No one MQTT Reconnection and no one packet loss. Your problem sounds like a WiFi problem.