arendst/Tasmota #21830: SDCard file manager does not show size of SDCard

hagenbuch, 2024-07-23:
### PROBLEM DESCRIPTION

On Tasmocompiler, I've compiled (german languag) Tasmota32 for my Olimex ESP32-POE (chip is ESP32-D0WDQ6 v1.0), Ethernet works and it is able to use my SDCard, can read Bosch sensors (compiled it that way). So far so good.

It does also display files on the SDcard and in the filemanager, I am able to switch between both filesystems. When browsing built-in flash, the size is shown correctly (304 kByte) but when the contents of the SDCard is supposed to be shown, it reads only "Größe 0,000 MB - Frei 0,000 MB   [SDCard]"
I hope I can attach a screenshot.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Olime ESP32-POE
- [x] Tasmota binary firmware version number used: 14.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-07-23:
With a 2 GB SD card, I'm seeing: `Size 1979.600 MB - Free 1979.592 MB`
What file system is your card formatted with? Should be FAT32.

hagenbuch, 2024-07-24:
Thanks for answering so quickly!

Yes, the card is FAT32 formatted but it is 8 GByte, this might be a problem. However, I can still put stuff on it while on my desktop machine (Linux Mint) and am able to download a file even from a folder but I can not upload anything via the Tasmota file manager while in SDCard mode, I am able to upload (small enough) files to the built in flash.

Just in order to test, I reformatted the SDHC with Linux's disk manager but apparently it is not possible to "shrink" the "volume" or partition only a part of the card.

Another really minor thing (that I have to say only as I am an UI developer and Tasmota is already great so I hope to improve:

In File manager, the switching to internal flash (and back to SDCard) works by clicking on "SDCard" because it contains a link. However, we have to click the link "SDCard" to get **away** from the SDCard. THis might be a little confusing for beginners. Maybe just show where we are as text wothout link and then add a "toggle" button?

Jason2866, 2024-07-24:
The issue you encounter is not caused by Tasmota. Tasmota uses the provided functions from Arduino for accessing SD Card. The issue should be opened at espressif32 github.
For Feature Requests please open a new thread in discussions

hagenbuch, 2024-07-24:
Thank you but sorry to come back, do you mean this repo?

https://github.com/tasmota/platform-espressif32

I could not find Espressif's github aparently.. glad for any help!

sfromis, 2024-07-24:
Over 32 GB would be where I'd "expect" trouble with FAT32, not smaller than that. I have a vague recollection of some brands working better than others, but do not remember details.

Here's what I got after formatting an 8 GB FAT32 partition on a 128 GB SD cart using Gnome Disks under Linux: `Size 7797.872 MB - Free 7797.864 MB`
(Obviously, there's the usual thing about what one GB really means as a number)

One risk zone may be if your setup can't deliver good power to the SD card.

Jason2866, 2024-07-24:
I have 16Gbyte SD Cards from Sandisk which work perfectly fine with PC and mobile Phone.
Not working with Arduino. Not only the size matters, others stuff idk too.
Have 32Gbyte SD Cards working perfectly fine with Arduino and useable 32Gbyte

hagenbuch, 2024-07-24:
@Jason2866 that means 16 and 32 GByte SDCards do not only show up in Tasmota's file manager but are writeable as well? Also thanks for the right repo!

@sfromis many thanks I'll examine both of your hints!

I am connected to a USB power supply that is able to serve about 65 W in total.. but I will add a beefy electrolytic capacitor just to test / and test with PoE, maybe that changes the outcome.

sfromis, 2024-07-24:
The power situation is not "just" about your power supply or an added decoupling capacitor, but also how much is available at 3.3V, from the voltage regulator. Basically about the total power budget at that potential, and a SD card (with the MCU running) might even need 100 mA, but power consumption can vary a lot across different cards.

Jason2866, 2024-07-24:
I don't believe power supply is the reason for showing "0".
Yes, when lucky and a compatible SD Card with 32Gbyte can be used. Read/Write files and correct sizes are shown.

Jason2866, 2024-07-24:
Maybe updating the Arduino core to the newer maintained libs from https://github.com/greiman/SdFat/tree/master solves the issues?

hagenbuch, 2024-07-24:
Thanks to all of you: I tested various SDCards: 8, 16 GB from Intenso, one 32 GB from Sandisk, all show the same (0,000 MB), even if I use PoE, no other current consumption. I'm also using Tasmota on a "Cheap Yellow Display" but could not make the SDCard run there, despite swapping software serial around. What confuses my a little is that on the Olimex, I am using I2C (SCL, SDA) to access the SDCard, the Cheap Yellow Display seems to connect the SDCard with SPI: SCLK 18, MISO 19, MOSI 23, CS 5. May I need to ise I2C but then what is SDA, what is SCL?

Jason2866, 2024-07-24:
SD Cards are connected via SPI. There is a nice wiring diagram here https://github.com/espressif/arduino-esp32/tree/master/libraries/SD

hagenbuch, 2024-07-24:
Thanks but this is how Olimex wires the SDCard (in my revision at least) and I am able to read stuff..
[image]

sfromis, 2024-07-24:
I'd not expect I2C to be relevant to using a SD card from Tasmota. What is that "I am using" based on? Did you configure I2C pins for the SD card? Silk screen labeling on the board tells nothing, as ESP32 allow basically any pin anywhere (except for analog and Ethernet). Also, it is not too uncommon that SPI pins are labeled SDA/SCL for MOSI/CLK. 

Inserting the 128/8 card mentioned already in a "CYD family" Sunton ESP32-8048S070 showed it working there too. Not sure what you mean by "swapping software serial around", software serial is not used on ESP32.

Another detail. I've seen issues inserting a SD card, but working after re-seating the card. There seemed to be a bit of transverse wiggling room in the spring-less socket.

hagenbuch, 2024-07-24:
> Did you configure I2C pins for the SD card?

Yes, and the protocol shows the Olimex board understood:

`00:00:00.001 HDW: ESP32-D0WDQ6 v1.0 
00:00:00.036 UFS: FlashFS mounted with 304 kB free
00:00:00.054 CFG: Loaded from File, Anzahl 53
00:00:00.066 QPC: Count 1
00:00:00.070 I2C: Bus1 using GPIO16(SCL) and GPIO13(SDA)
00:00:00.133 UFS: SDCard mounted (SDIO 1-bit) with 0 kB free
00:00:00.134 SCR: ufilesystem found
00:00:00.172 BRY: Berry initialized, RAM used 3184 bytes`

Jason2866, 2024-07-24:
To track down where exactly the problem is the IDF example should be used with the problem SD Cards. If this works the issue is in Arduino code (which is an layer over IDF and uses the functions from IDF via APIs). If not IDF does not support "some" cards
https://github.com/espressif/esp-idf/tree/release/v5.1/examples/storage/sd_card/sdspi

hagenbuch, 2024-07-24:
OK I will try to "go back" to Arduino and report. It might take a little while.. thanks to all of you!

Jason2866, 2024-07-24:
But yes, wrong configuring GPIOs can end in half working SD Card

hagenbuch, 2024-07-24:
The Olimex only offers pins 5 (which I interpret as "chip select", maybe wrongly so?), then pin 14 (Clock, obvious) and then pin 15 where I am unsure whether to configure it a s MOSI or MISO..

hagenbuch, 2024-07-26:
I'll give you an update:

I installed micropython 1.23 fpr ESP32 on the same Olimex board. Wifi and Ethernet work.

As an example, I provided an empty, FAT32 formatted SDHC of 4GBytes, it contained just an empty folder and a file in it.

Micropython is able to read and write the disk. In order to query the total space, I have to do this

    import uos
    uos.statvfs('/flash')
    (4096, 4096, 968088, 968083, 968083, 0, 0, 0, 0, 255)

The third integer ist the total size, meaning 4096 * 968088 = 3781,594 MBytes. So, I would conclude that mPy is able to read the value. According to the latest release notes here:

https://micropython.org/download/OLIMEX_ESP32_POE/

it seems to use IDF version v5.0.5 or v5.2. Not exactly sure if that gives you the right hints.

hagenbuch, 2024-07-27:
Different devices but I guess, most of them Linux. I will try the official formatter on Windows but I am on my way to a week of holidays, will report back. Thanks so far!