arendst/Tasmota #21873: THR320D latching relay Triggered at boot

drbios, 2024-07-30:
### PROBLEM DESCRIPTION
Hi, I have a THR320D that triggers the main relay at boot time.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
20:09:39.407 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
20:09:39.422 SRC: Backlog
20:09:39.423 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
20:09:39.426 RSL: RESULT = {"NAME":"Sonoff THR320D","GPIO":[32,0,0,0,226,9280,0,0,0,321,0,576,320,9184,9216,9312,0,0,9313,9248,0,1,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:11:13.561 CMD: Status 0
20:11:13.563 SRC: WebConsole from 192.168.1.42
20:11:13.564 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
20:11:13.568 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_C37B80","ButtonTopic":"0","Power":"01","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:11:13.575 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:05:24","StartupUTC":"2024-07-30T19:05:49","Sleep":50,"CfgHolder":4617,"BootCount":123,"BCResetTime":"2024-07-30T02:09:23","SaveCount":475}}
20:11:13.579 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0.4(eb71fcf-tasmota32)","BuildDateTime":"2024-07-30T11:39:54","Core":"3_0_2","SDK":"5.1.4.240708","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"378/699"}}
20:11:13.582 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BOTH1002",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00002080","00006000","00004000","00000000"]}}
20:11:13.597 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1997,"Free":882,"Heap":148,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000014"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,82,!86,87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
20:11:13.603 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-C37B80-7040","IPAddress":"192.168.1.173","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"E8:6B:EA:C3:7B:80","IP6Global":"","IP6Local":"fe80::ea6b:eaff:fec3:7b80%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
20:11:13.606 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C37B80","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:11:13.609 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-07-30T19:11:13Z","Local":"2024-07-30T20:11:13","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:21","Sunset":"20:31"}}
20:11:13.616 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-07-30T20:11:13"}}
20:11:13.620 BRY: GC from 7464 to 4084 bytes, objects freed 9/43 (in 0 ms) - slots from 47/91 to 47/91
20:11:13.625 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-07-30T20:11:13","Uptime":"0T00:05:24","UptimeSec":324,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BOTH1002","BSSId":"70:A7:41:D2:AA:29","Channel":1,"Mode":"HT20","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:07"}}}
20:11:24.969 WIF: Checking connection...

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
PowerOnstate 0
SetOption0 1
SetOption63 1
Power1 1

Cut the enery of the unit
The main relay makes a fake turn on for a thousandth of a second, literally turning on and off quickly once and then continuing off according to the PowerOnState 0 setting.

### EXPECTED BEHAVIOUR
main relay do not trips at boot

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I have tested firmware from 12.x to developer and this phenomenon occurs in all of them. It is as if something affects pins 19 and 22 where the main bi-stable relay is connected.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-07-31:
> The main relay makes a fake turn on for a thousandth of a second,

Are you sure it does a turn on?

From what I understand from a bi-stable relay it stays in the position it last was when triggered. In your scenario you turn on the relay. Then power is removed. Using the settings for PowerOnState you expect the relay to be turned off once power is restored. 

To me it seems that's exactly what happens; once power is restored the still turned on relay is powered off as soon as Tasmota is started. Between power restore and Tasmota being able to turn off the relay (taking several microseconds of course) you'll notice power is available from the relay but Tasmota did not power on the relay.

Unless you have measured a power on signal is sent to the relay at power restoration I don't see an issue here.

I don't have the THR320D but my POWR320D does exactly that; When power is removed no click is heard hence the relay stays on. Once power is restored the relay is turned off by Tasmota several microseconds later.

drbios, 2024-07-31:
> Are you sure it does a turn on?
>
> From what I understand from a bi-stable relay it stays in the position it last was when triggered. In your scenario you turn on the relay. Then power is removed. [...]

i hope you can understand my lousy english
https://youtu.be/Bu-IL0nFojw?feature=shared

arendst, 2024-07-31:
Your english is fine. In your video happens exactly what I tried to tell above:

After power is restored your actuater will be actuated because the bi-stable relay is still ON by hardware design. Then if tasmota is powered up it will power down it's relay due to ``PowerOnState 0`` and the actuater will also power off.

This is inherent to the bi-stable relay. Nothing Tasmota can do about this.

drbios, 2024-07-31:
> Your english is fine. In your video happens exactly what I tried to tell above:
>
> After power is restored your actuater will be actuated because the bi-stable relay is still ON by hardware design. [...]

I understand, I'm used to moestable relays that don't turn on when you set the PowerOnState 0 option. In my case I was looking to use this THR320D in a transfer board that had to keep the relays off if the power came back to the board and not do this fake power on during boot like this unit does. I will use a DUal r3 instead that has moestable relays and that always stay off during boot with the PowerOnState 0 option. Thank you very much for your patience.

https://www.youtube.com/watch?v=jG0_Hlrgd6A&ab_channel=JedAnderson