arendst/Tasmota #21875: [Matter] adding Tasmota device to Google Home creates two devices

hazbot72, 2024-07-31:
### PROBLEM DESCRIPTION
I have a new smart plug device, Athom plug v3  https://www.athom.tech/blank-1/tasmota-esp32-c3-uk-plug-v3

It supports Matter in Tasmota. The plug has relay for on/off and also has power monitoring.

It comes pre-installed with Tasmota and I have updated the version to 14.10. 

When I add this device to Google Home via Matter, the commissioning is successful but I see two new devices in Google Home instead of one.

Both new devices have the same name, that of the smart plug. One device has an icon similar to other smart plugs, i.e. the "power outlet" icon. It shows the current on/off state of the relay and I can switch the plug on/off by tapping the icon.

The second new device in Google Home has what appears to be a "thermometer / lightbulb" icon. Pressing this icon doesn't perform any action but opens the settings page in Google Home for that device.

If I delete the second device from Google Home, it deletes both devices.

I suspect the second icon might be an endpoint for the power monitoring and I would like to remove this.

However if I access the Matter configuration page in the Tasmota gui, and look at the configured endpoints, I can only see one endpoint, which is the relay. There does not appear to be a way to remove the second endpoint.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Athom plug V3
- [x] Tasmota binary firmware version number used: 14.10
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: pre-installed
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

10:25:46.587 CMD: Backlog Template; Module; GPIO 255
10:25:46.613 MQT: stat/plug17/RESULT = {"NAME":"Athom Plug V3","GPIO":[0,0,0,32,0,224,576,0,0,0,0,0,0,0,0,0,0,0,0,0,3104,0],"FLAG":0,"BASE":1}
10:25:46.841 MQT: stat/plug17/RESULT = {"Module":{"0":"Athom Plug V3"}}
10:25:47.048 MQT: stat/plug17/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"0":"None"},"GPIO5":{"224":"Relay1"},"GPIO6":{"576":"LedLink_i"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"3104":"CSE7766 Rx"},"GPIO21":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

10:23:56.970 CMD: Status 0
10:23:56.972 SRC: WebConsole from 10.208.13.128
10:23:56.973 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
10:23:56.978 MQT: stat/plug17/STATUS = {"Status":{"Module":0,"DeviceName":"plug17","FriendlyName":["plug17"],"Topic":"plug17","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:23:56.989 MQT: stat/plug17/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T12:25:29","StartupUTC":"2024-07-30T20:58:27","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2024-07-26T09:47:53","SaveCount":44}}
10:23:56.998 MQT: stat/plug17/STATUS2 = {"StatusFWR":{"Version":"14.1.0(release-tasmota32)","BuildDateTime":"2024-06-03T11:56:24","Core":"3_0_0","SDK":"5.1.4.240529","CpuFrequency":160,"Hardware":"ESP32-C3 v0.4","CR":"397/699"}}
10:23:57.006 MQT: stat/plug17/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":1,"LogHost":"syslog.pdc","LogPort":514,"SSId":["PDCD2D",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00208009","0405C80001000680003C5A0A192800000000","08000080","00006000","00004000","00000020"]}}
10:23:57.022 MQT: stat/plug17/STATUS4 = {"StatusMEM":{"ProgramSize":2016,"Free":863,"Heap":138,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000015"],"Drivers":"1,2,3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!71,!73,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
10:23:57.037 MQT: stat/plug17/STATUS5 = {"StatusNET":{"Hostname":"plug17","IPAddress":"10.208.13.187","Gateway":"10.208.13.201","Subnetmask":"255.255.255.0","DNSServer1":"10.208.13.14","DNSServer2":"0.0.0.0","Mac":"CC:8D:A2:50:94:D4","IP6Global":"","IP6Local":"fe80::ce8d:a2ff:fe50:94d4%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.208.13.14","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
10:23:57.047 MQT: stat/plug17/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.pdc","MqttPort":1883,"MqttClientMask":"plug17","MqttClient":"plug17","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:23:57.052 BRY: GC from 35576 to 26788 bytes, objects freed 77/324 (in 2 ms) - slots from 527/561 to 316/546
10:23:57.059 MQT: stat/plug17/STATUS7 = {"StatusTIM":{"UTC":"2024-07-31T09:23:57Z","Local":"2024-07-31T10:23:57","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:29"}}
10:23:57.067 MQT: stat/plug17/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:23:57.078 MQT: stat/plug17/STATUS10 = {"StatusSNS":{"Time":"2024-07-31T10:23:57","ENERGY":{"TotalStartTime":"2024-07-26T09:47:53","Total":0.409,"Yesterday":0.204,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}}}
10:23:57.088 BRY: GC from 30251 to 25921 bytes, objects freed 48/307 (in 1 ms) - slots from 425/546 to 299/501
10:23:57.095 MQT: stat/plug17/STATUS11 = {"StatusSTS":{"Time":"2024-07-31T10:23:57","Uptime":"0T12:25:30","UptimeSec":44730,"Heap":138,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":25,"Objects":307},"POWER":"OFF","Wifi":{"AP":1,"SSId":"PDCD2D","BSSId":"E8:48:B8:74:5E:40","Channel":6,"Mode":"HT20","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Add a smart plug with energy monitoring into Google Home via Matter integration

### EXPECTED BEHAVIOUR
A single new device in Google Home which represents the smart plug

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-07-31:
I suppose your are seeing this icon 
[image]
which indicates a "Bridge" device. Tasmota uses a virtual bridge so you can add or remove dynamically endpoints.

If that bothers you, and you don't want dynamic changes, you can select "Force Static endpoints (non-bridge)" in the UI before pairing.

hazbot72, 2024-07-31:
Ah yes that was it.

Thank you for the explanation.

hazbot72, 2024-07-31:
Behaviour as intended

closed 2024-07-31 by hazbot72