arendst/Tasmota #21898: RfRaw 177 stops reporting after a while

GijsTimmers, 2024-08-03:
### PROBLEM DESCRIPTION
On my RfBridge flashed with Portisch firmware `RF-Bridge-EFM8BB1-20190220.hex`, after a while (I had it happen within 24 hours today) `RfRaw 177` just stops receiving RF codes in the log.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  Sonoff RF Bridge R2
- [x] Tasmota binary firmware version number used: 14.1.0.2 (bdfe352-tasmota) 
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmota WebInstaller
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
03:06:48.767 CMD: Backlog Template; Module; GPIO 255
03:06:48.791 MQT: stat/deurbel/RESULT = {"NAME":"Generic","GPIO":[32,3200,1,3232,1,1,0,0,1,320,1,0,0,0],"FLAG":0,"BASE":25}
03:06:48.993 MQT: stat/deurbel/RESULT = {"Module":{"25":"Sonoff Bridge"}}
03:06:49.248 MQT: stat/deurbel/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
03:08:32.845 CMD: Backlog Rule1; Rule2; Rule3
03:08:32.864 MQT: stat/deurbel/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":33,"Free":478,"Rules":"on system#boot do rfraw 177 endon"}}
03:08:33.101 MQT: stat/deurbel/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":35,"Free":476,"Rules":"ON Clock#Timer=1 DO Restart 1 ENDON"}}
03:08:33.346 MQT: stat/deurbel/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
03:09:07.834 CMD: Status 0
03:09:07.843 MQT: stat/deurbel/STATUS = {"Status":{"Module":25,"DeviceName":"Deurbel","FriendlyName":["Deurbel"],"Topic":"deurbel","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
03:09:07.855 MQT: stat/deurbel/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:11:11","StartupUTC":"2024-08-03T00:57:56","Sleep":50,"CfgHolder":4617,"BootCount":77,"BCResetTime":"2024-06-15T20:10:27","SaveCount":148,"SaveAddress":"F4000"}}
03:09:07.866 MQT: stat/deurbel/STATUS2 = {"StatusFWR":{"Version":"14.1.0.2(bdfe352-tasmota)","BuildDateTime":"2024-06-15T21:13:09","Boot":7,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"348/699"}}
03:09:07.875 MQT: stat/deurbel/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000081","00006000","00004000","00000000"]}}
03:09:07.894 MQT: stat/deurbel/STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
03:09:07.910 MQT: stat/deurbel/STATUS5 = {"StatusNET":{"Hostname":"deurbel","IPAddress":"192.168.3.22","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.3.1","DNSServer2":"0.0.0.0","Mac":"XXXX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
03:09:07.922 MQT: stat/deurbel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_XXXXX","MqttUser":"XXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:09:07.938 MQT: stat/deurbel/STATUS7 = {"StatusTIM":{"UTC":"2024-08-03T01:09:07Z","Local":"2024-08-03T03:09:07","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"XX:XX","Sunset":"XX:XX"}}
03:09:07.951 MQT: stat/deurbel/STATUS10 = {"StatusSNS":{"Time":"2024-08-03T03:09:07"}}
03:09:07.963 MQT: stat/deurbel/STATUS11 = {"StatusSTS":{"Time":"2024-08-03T03:09:07","Uptime":"0T00:11:11","UptimeSec":671,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"iot","BSSId":"XXXX","Channel":1,"Mode":"11n","RSSI":100,"Signal":-13,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
N/A

```

### TO REPRODUCE
Issue `RfRaw 177` and wait.

### EXPECTED BEHAVIOUR
Keep showing RF codes

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
There has been a similar discussion over [here](https://github.com/arendst/Tasmota/issues/7534) and [here](https://github.com/arendst/Tasmota/issues/6963) but both of them didn't really lead to isolating the problem so let's try again.

Jason2866, 2024-08-03:
Saw this behaviour too. The code responsible for decoding the RF protocols (extra MCU) is not maintained from the author (Portisch) anymore. Furthermore the RF-Bridges with this MCU type are EoL. The actual have a different not compatible MCU.
Long story short. This issue will probably not fixed.

GijsTimmers, 2024-08-05:
Alright, thanks for the info. As a workaround, I set up timers to have my device reboot every 4 hours now. If anybody knows a better workaround, please let me know.

closed 2024-09-04 by github-actions[bot]