arendst/Tasmota #21923: Gibberish RESULT when using nested subroutines from Console

JarnoBoks, 2024-08-07:
### PROBLEM DESCRIPTION
I use the Tasmota scripting option with subroutines and call the Subroutine from the Console. This subroutine calls another subroutine and the result MQT is gibberish. If i do not nest, the result string is ok. Example script: 

#ZONEA(pact)
	=#ZONE(1 pact)
	=>publish stat/%topic%/ZONEA %0pact% 		
next

#ZONE(zone act)
	if act==1 {
		=#zoneon(zone)	
	} else {
		=#zoneoff(zone)
	}
next

Console command: ZONEA 0
Results: 
12:19:42.172 MQT: stat/rainbird_irrigation/ZONEA = 0
12:19:42.176 MQT: stat/rainbird_irrigation/RESULT = {"ZONEA":"r�?"}

Expected result:
12:19:42.172 MQT: stat/rainbird_irrigation/ZONEA = 0
12:19:42.176 MQT: stat/rainbird_irrigation/RESULT = {"ZONEA":"0"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP8266
- [x ] Tasmota binary firmware version number used: 14.1
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [x] Flashing tools used: WebGui
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
12:26:26.811 CMD: Backlog Template; Module; GPIO 255
12:26:26.854 MQT: stat/rainbird_irrigation/RESULT = {"NAME":"Rainbird_Irrigation","GPIO":[259,0,263,352,258,257,0,0,261,448,260,230,256,0],"FLAG":0,"BASE":18}
12:26:27.072 MQT: stat/rainbird_irrigation/RESULT = {"Module":{"0":"Rainbird_Irrigation"}}
12:26:27.326 MQT: stat/rainbird_irrigation/RESULT = {"GPIO0":{"259":"Relay_i4"},"GPIO1":{"0":"None"},"GPIO2":{"263":"Relay_i8"},"GPIO3":{"352":"Counter1"},"GPIO4":{"258":"Relay_i3"},"GPIO5":{"257":"Relay_i2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"261":"Relay_i6"},"GPIO13":{"448":"PWM_i1"},"GPIO14":{"260":"Relay_i5"},"GPIO15":{"230":"Relay7"},"GPIO16":{"256":"Relay_i1"},"GPIO17":{"0":"None"}}
- [x] 
```lua
  Configuration output here: not sure wat is necessary

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: No rules present

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:29:45.615 MQT: stat/rainbird_irrigation/STATUS = {"Status":{"Module":0,"DeviceName":"rainbird_irrigation","FriendlyName":["rainbird_irrigation","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"rainbird_irrigation","ButtonTopic":"0","Power":"001000000","PowerLock":"000000000","PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:29:45.622 MQT: stat/rainbird_irrigation/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T00:48:44","StartupUTC":"2024-08-07T10:41:01","Sleep":50,"CfgHolder":4617,"BootCount":128,"BCResetTime":"2023-02-20T09:58:16","SaveCount":5884,"SaveAddress":"FB000"}}
12:29:45.627 MQT: stat/rainbird_irrigation/STATUS2 = {"StatusFWR":{"Version":"14.1.0(tasmota)","BuildDateTime":"2024-08-07T06:24:08","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"562/699"}}
12:29:45.632 MQT: stat/rainbird_irrigation/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Omega","ap01f_fixed"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]}}
12:29:45.646 MQT: stat/rainbird_irrigation/STATUS4 = {"StatusMEM":{"ProgramSize":668,"Free":332,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","879AC787","043E8001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
12:29:45.653 MQT: stat/rainbird_irrigation/STATUS5 = {"StatusNET":{"Hostname":"rainbird-irrigation-2546","IPAddress":"192.168.2.138","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:29:69:F2","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:29:45.659 MQT: stat/rainbird_irrigation/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.87","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2969F2","MqttUser":"MQTT_WifiSwitch","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:29:45.668 MQT: stat/rainbird_irrigation/STATUS7 = {"StatusTIM":{"UTC":"2024-08-07T11:29:45Z","Local":"2024-08-07T12:29:45","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:32","Sunset":"20:18"}}
12:29:45.675 MQT: stat/rainbird_irrigation/STATUS10 = {"StatusSNS":{"Time":"2024-08-07T12:29:45","COUNTER":{"C1":0}}}
12:29:45.683 MQT: stat/rainbird_irrigation/STATUS11 = {"StatusSTS":{"Time":"2024-08-07T12:29:45","Uptime":"0T00:48:44","UptimeSec":2924,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"ON","POWER8":"OFF","POWER9":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"ap01f_fixed","BSSId":"C2:C1:C0:D0:99:61","Channel":1,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":2,"Downtime":"0T00:01:02"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
[... 190 lines cut ...]
			=>power2 0
			=>power3 0
		case 9
			=>power5 0
			=>power6 0
	ends
	
	; Register this zone as active
	zstate[pzoneon]=1
	
	; Turn on the pump if there are no zones open (ie. ovls = 10). 
	; If there is a zone open, close it. 	
	ovls=#chckzn
	if ovls>9 {
		print turn on pump		
		;setup the next pump action to 1 and setup the timer to fire after 2 seconds
		apump=1
		tim=upsecs+2
	} else {
	    print TODO turn zone %ovls% off
		;=#zoneoff(ovls)
	}	
next

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-08-07:
In general, Tasmota does not allow you to call other commands from inside commands. While I'm not talking specifically about the scripter, what you post looks like the usual results when trying to do it other ways, like from Berry. The reason is that the command processing is not reentrant, and you get a race condition for the outer command after the inner one completed, destroying the context of the outer command.

JarnoBoks, 2024-08-07:
I kinda understand what you are saying. But that destroys the whole idea of subroutines in the scripter that can be called from the Console. I can workaround by publishing my own result, but it would be nice to suppress/remove this gibberish.

sfromis, 2024-08-07:
I'm not well-versed with the scripter, but in Berry my solution to that is to let the outer command complete, after having scheduled a timer (immediately expiring) to run the inner command.