arendst/Tasmota #21947: MessageHandler::msg_received exception: type_error;unsupported operand type(s) for >: 'nil' and 'int'

lboue, 2024-08-11:
### PROBLEM DESCRIPTION
Matter cluster discovery from Home Assistant doesn't work in my case. There is an exception in the Tasmota device logs.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: [_____](https://tasmota.github.io/install/)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

```
13:39:24.266 MTR: UDP received from [fe80::6a3a:58f8:9512:d44a%st1]:55930
13:39:24.268 MTR: decode header: local_session_id=64451 message_counter=36751386
13:39:24.271 MTR: >          Decrypted message: protocol_id:1 opcode=1 exchange_id=3691
13:39:24.273 MTR: .          Removed packet from sending list id=124563287
13:39:24.280 BRY: GC from 36676 to 26813 bytes, objects freed 72/283 (in 2 ms) - slots from 319/440 to 222/440
13:39:24.284 MTR: <Ack*      ( 64451) ack=36751386 id=124563290 
13:39:24.287 MTR: sending packet to '[fe80::6a3a:58f8:9512:d44a%st1]:55930'
13:39:24.369 BRY: GC from 37103 to 27027 bytes, objects freed 140/281 (in 3 ms) - slots from 409/440 to 221/425
13:39:24.465 BRY: GC from 37326 to 26256 bytes, objects freed 160/268 (in 3 ms) - slots from 378/440 to 219/440
13:39:24.543 BRY: GC from 37186 to 27009 bytes, objects freed 137/257 (in 3 ms) - slots from 376/440 to 214/425
13:39:24.547 MTR: sending packet to '[fe80::6a3a:58f8:9512:d44a%st1]:55930'
13:39:24.674 MTR: UDP received from [fe80::6a3a:58f8:9512:d44a%st1]:55930
13:39:24.676 MTR: decode header: local_session_id=64451 message_counter=36751387
13:39:24.680 MTR: >          Decrypted message: protocol_id:1 opcode=1 exchange_id=3691
13:39:24.681 MTR: .          Removed packet from sending list id=124563289
13:39:24.687 MTR: <Ack*      ( 64451) ack=36751387 id=124563292 
13:39:24.690 MTR: sending packet to '[fe80::6a3a:58f8:9512:d44a%st1]:55930'
13:39:24.763 BRY: GC from 37257 to 25352 bytes, objects freed 96/275 (in 2 ms) - slots from 364/425 to 233/425
13:39:24.862 BRY: GC from 35604 to 25609 bytes, objects freed 137/264 (in 3 ms) - slots from 376/425 to 223/425
13:39:24.953 BRY: GC from 36454 to 27031 bytes, objects freed 121/257 (in 2 ms) - slots from 350/425 to 214/425
13:39:24.957 MTR: sending packet to '[fe80::6a3a:58f8:9512:d44a%st1]:55930'
13:39:25.187 MTR: UDP received from [fe80::6a3a:58f8:9512:d44a%st1]:55930
13:39:25.189 MTR: decode header: local_session_id=64451 message_counter=36751388
13:39:25.193 MTR: >          Decrypted message: protocol_id:1 opcode=1 exchange_id=3691
13:39:25.195 MTR: .          Removed packet from sending list id=124563291
13:39:25.202 MTR: <Ack*      ( 64451) ack=36751388 id=124563294 
13:39:25.205 MTR: sending packet to '[fe80::6a3a:58f8:9512:d44a%st1]:55930'
13:39:25.263 BRY: GC from 37308 to 25700 bytes, objects freed 110/272 (in 2 ms) - slots from 352/425 to 223/425
13:39:25.364 BRY: GC from 35993 to 25547 bytes, objects freed 140/265 (in 3 ms) - slots from 412/455 to 224/455
13:39:25.451 MTR: MessageHandler::msg_received exception: type_error;unsupported operand type(s) for >: 'nil' and 'int'
13:39:44.321 WIF: Checking connection...
13:40:04.332 WIF: Checking connection...
13:40:15.269 HTP: Tools
13:40:17.627 HTP: Main Menu
13:40:18.988 HTP: Configuration
13:40:20.343 HTP: Configure Logging
13:40:24.298 WIF: Checking connection...
13:40:24.503 HTP: Configuration
13:40:44.312 WIF: Checking connection...
```

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-08-11:
I'm off for two weeks. I will check when I return

lboue, 2024-08-11:
Thank you for your time. Have a nice vacation.

closed 2024-08-29 by s-hadinger

lboue, 2024-08-29:
Here are the console traces with debug mode enabled for 2 examples :
- 1 EP (1 relay): [OK_Only_relay.log](https://github.com/user-attachments/files/16802463/OK_Only_relay.log)
- 2 EP (1 relay and one shutter): [KO_relay_and_shuter.log](https://github.com/user-attachments/files/16802462/KO_relay_and_shuter.log)

PR #22049 (2024-08-29): Matter fail to report Shutter status if no shutter is configured in Tasmota
## Description:

**Related issue (if applicable):** fixes #21947

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota core ESP8266 V.2.7.7
  - [x] The code change is tested and works with Tasmota core ESP32 V.3.0.4
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIB [... cut]
PR #22052 (2024-08-29): Matter fix Waterleak broken after Berry solidification optimisation #21885
## Description:

Fix Matter bug introduced by #21885 when a method is shared between two classes.

I will later remove the ability to share a methods between classes because it doesn't work anymore (and it wasn't any useful)

**Related issue (if applicable):** fixes #21947

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is teste [... cut]