arendst/Tasmota #21963: And again: The documentation page does not show the most current version 14.0

meyergru, 2024-08-14:
See title and:

https://tasmota.github.io/docs/

barbudor, 2024-08-14:
I'm going to stay polite again, probably for the last time, but if you don't have anything else to do than coming here to complain about the same item at each version, please move to ESP-Home and leave us

This is a community and you could just do the update yourself instead of complaining
At least you would give a little back to the community instead of just take and complain

meyergru, 2024-08-14:
What is your problem and BTW: who are you? I see something that is not right and report it. Isn't that what the issues are for? The last times I reported this, Theo seemed glad that I pointed it out.

I, for example, like many people, who just use Tasmota, do only look at the main Tasmota page to find out if there is a new version.

In my specific case, because I regularly publish images for a specific hardware that is not support directly (https://www.congenio.de/infos/shelly3em.html, for that matter). So much for "not giving back for the community". But I understand - I will mind my own business from here on...

barbudor, 2024-08-14:
If you were a nice guy, you would have clicked on the that f....g edit button on the top right of the doc page to make the change

But instead, less than 2 hours after the publication of the new version you come by starting you message by "And again:"

So yes, what have you given to the community ?
Where are discussions where you have answered to other users questions ?
Where are your PR on the code or on the docs ?

Writing a help page on your company web site ? Cool, you could just have written a doc page into Tasmota page instead so it doesn't benefit only to customer or rare people that are aware of your web site

Indeed you already mind only your own business by using Tasmota

meyergru, 2024-08-14:
Whatever.