arendst/Tasmota #21980: ADC Voltage no decimal (like on Current)

ACX-Com, 2024-08-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I'm using a voltage devider to get the Supply Voltage to the ADC0 of the ESP8266.
The problem is, that i can't find any settings to get native decimal Voltage Value.
If i try with Current (Ampere) setting for the ADC0, i get decimal without problems and with ADCParam i can make correct calibration.
Calibration works on Voltage setting too, but it is not handling the decimal setting like at Current.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): pure ESP8266
- [ ] Tasmota binary firmware version number used: 14.2.0 generic german (english also tested)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer and WebUI-Uprade incl. storing/restoring BackUp
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
  Configuration output here:
16:24:17.567 MQT: stat/tasmota2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:24:17.780 MQT: stat/tasmota2/RESULT = {"Module":{"18":"Generic"}}
16:24:17.988 MQT: stat/tasmota2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"10624":"ADC Spannung"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:27:17.047 MQT: stat/tasmota2/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota2","FriendlyName":["tasmota2"],"Topic":"tasmota2","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:27:17.087 MQT: stat/tasmota2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:16","StartupUTC":"2024-08-18T15:27:01","Sleep":50,"CfgHolder":4617,"BootCount":61,"BCResetTime":"2024-08-17T07:09:08","SaveCount":283,"SaveAddress":"F9000"}}
16:27:17.120 MQT: stat/tasmota2/STATUS2 = {"StatusFWR":{"Version":"14.2.0(release-tasmota)","BuildDateTime":"2024.08.14 12:41:05","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"393/699"}}
16:27:17.138 MQT: stat/tasmota2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLAN-Bast",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:27:17.173 MQT: stat/tasmota2/STATUS4 = {"StatusMEM":{"ProgramSize":649,"Free":352,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164054","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
16:27:17.213 MQT: stat/tasmota2/STATUS5 = {"StatusNET":{"Hostname":"tasmota2-5425","IPAddress":"192.168.2.167","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"48:55:19:EE:B5:31","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:27:17.236 MQT: stat/tasmota2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EEB531","MqttUser":"Admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:27:17.260 MQT: stat/tasmota2/STATUS7 = {"StatusTIM":{"UTC":"2024-08-18T15:27:17Z","Local":"2024-08-18T16:27:17","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:59"}}
16:27:17.316 MQT: stat/tasmota2/STATUS10 = {"StatusSNS":{"Time":"2024-08-18T16:27:17","ANALOG":{"Voltage":3}}}
16:27:17.324 MQT: stat/tasmota2/STATUS11 = {"StatusSTS":{"Time":"2024-08-18T16:27:17","Uptime":"0T00:00:16","UptimeSec":16,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WLAN-Bast","BSSId":"3C:37:12:81:90:97","Channel":11,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:28:19.098 MQT: stat/tasmota2/RESULT = {"WebLog":4}
16:28:19.425 CFG: in Flash gespeichert an F7, Anzahl 285, Bytes 4096
16:28:22.559 WIF: Prüfe Verbindung…
16:28:28.201 MQT: tele/tasmota2/STATE = {"Time":"2024-08-18T16:28:28","Uptime":"0T00:00:30","UptimeSec":30,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WLAN-Bast","BSSId":"3C:37:12:81:90:97","Channel":11,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:05"}}
16:28:28.263 MQT: tele/tasmota2/SENSOR = {"Time":"2024-08-18T16:28:28","ANALOG":{"Voltage":3}}
16:28:42.631 WIF: Prüfe Verbindung…

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

see problem description above.
Simple set ADC0 to ADC Voltage and try to get decimal value like at setting ADC Current.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Please see problem description above

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

i think it's not needed ;)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-08-18:
> Simple set ADC0 to ADC Voltage

There is no such "ADC Voltage" mode for ESP8266's ADC
The mode you are using in "ADC Input" and returns the raw reading of the ADSC register: 0..1023 on ESP8266 or 0..4095 on ESP32

You can make a feature request

[image]

barbudor, 2024-08-18:
The best solution for you is to use ADC Params to convert the value to milliVolts as the output of ADC Param is an integer

arendst, 2024-08-18:
@barbudor since v14.2 there is both current and voltage ADC present.

@ACX-Com try command voltres 2

barbudor, 2024-08-18:
Hi @arendst 
Sorry, I missed this one

sfromis, 2024-08-18:
Yeah, the default value for `VoltRes` is 0 decimals.

ACX-Com, 2024-08-18:
THANK YOU VERY MUCH!!!
VoltRes was the solution! 👍
...so simple 🤦‍♂️
for the next time i will make more showing to the commands in the Docu - sorry. 🧐

closed 2024-08-18 by Jason2866