arendst/Tasmota #21982: Fatal error when compiling tasmota32c3: ../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory

nikvostro, 2024-08-19:
Hello!

I've faced with the issue of compiling my own customized tasmota32c3 (based on _master_ branch) setup of tasmota 14.2.0 with the error: `../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory`. It was compiling well many versions before 14.2.0.

_as.exe_ file does exist in the specified location :
[image]

Troubleshooting steps performed:
1. PIO: Clean, close PIO and then:
   1. Deleted folders _.cache + .pio + .vscode_ (json files excluding settings) in project folder 
   2. Deleted folders in _.platformio_: _.cache + packages + platforms_
[Similar issue described](https://www.letscontrolit.com/forum/viewtopic.php?t=10242)
2. Tried to compile **_master_ branch standard _[env:tasmota32c3]_** setup - the same behavior
3. Performed cleanup as in p.1 and tried to compile again and again - no success

- Other environments **compiles well** with the current PIO and tasmota 14.2.0.
- No additional antivirus or firewall software except Windows 10 embedded.
- No interactive prompts and alerts in the Windows Defender detection log

What I'm doing wrong?

```
Compiling .pio\build\tasmota32c3\lib1a1\HTTPUpdate\HTTPUpdate.cpp.o
Compiling .pio\build\tasmota32c3\lib69a\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\tasmota32c3\lib20a\WebServer\Parsing.cpp.o
Compiling .pio\build\tasmota32c3\lib20a\WebServer\WebServer.cpp.o
riscv32-esp-elf-g++: fatal error: cannot execute 'c:/users/nikolay/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\tasmota32c3\src\tasmota.ino.cpp.o] Error 1
============================================================================================================================== [FAILED] Took 56.16 seconds ==============================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota32c3    FAILED    00:00:56.155
========================================================================================================================= 1 failed, 0 succeeded in 00:00:56.155 ========================================================================================================================= 

 *  The terminal process "C:\Users\Nikolay\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'tasmota32c3'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

arendst, 2024-08-19:
https://github.com/arendst/Tasmota/discussions/21941

closed 2024-08-19 by arendst

nikvostro, 2024-08-19:
@arendst apologies - missed those! Thank you!