arendst/Tasmota #21986: ShutterButton hold does not work

markatom, 2024-08-19:
### PROBLEM DESCRIPTION
I am trying to set up button control for my shutters. Single, double, and triple press commands are working as documented, but I am unable to make the hold command work.

* I am using Kincony A8 (KC868-A8).
* I have custom compiled firmware from `tasmota32_base` with `user_config_override.h` as follows:

```c
#define USE_I2C

#define USE_PCF8574
#define USE_PCF8574_MODE2

#define USE_ETHERNET
#define ETH_TYPE 0
#define ETH_ADDR 0
#define ETH_CLKMODE 3
```

* My `pcf8574.dat` configuration file looks like this:

```json
{"NAME":"PCF8574","BASE":1,"GPIO":[32,33,34,35,36,37,38,39,256,257,258,259,260,261,262,263]}
```

* I am using the template provided by blakadder:

```json
{"NAME":"KC868-A16 rev 1.4","GPIO":[32,1,1,1,640,608,1,1,1,1,1,1,1,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
```

Configuration used:

```
Interlock 1,2 3,4, 5,6 7,8
Interlock ON

SetOption1 1 # restrict to single to penta press and hold actions (i.e., disable inadvertent reset due to long press)
SetOption32 5 # number of 0.1 seconds to hold button before sending HOLD action message.
SetOption80 1 # enable blinds and shutters support

ShutterRelay1 1 
ShutterRelay2 3
ShutterRelay3 5
ShutterRelay4 7

ShutterMode 1 # normal two relays up/off down/off

# I am using buttons 7 and 8 to control the first shutter
# there is no particular reason for this
# I just wired it like that on my workbench to test it out
ShutterButton1 7 0 - - 0
ShutterButton1 8 100 - - 100

Restart 1
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Kincony A8 (KC868-A8)
- [x] Tasmota binary firmware version number used: 14.2.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: https://tasmota.github.io/install/
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
[... 363 lines cut ...]
22:07:03.687 SHT: Shtr1, Button 8, hold 30, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
22:07:03.737 SHT: Shtr1, Button 8, hold 31, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
22:07:03.788 SHT: Shtr1, Button 8, hold 32, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
22:07:03.838 SHT: Shtr1, Button 8, hold 33, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
22:07:03.888 SHT: Shtr1, Button 8, hold 34, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
22:07:04.035 SHT: Shtr1, Button 8, hold 35, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
22:07:04.091 SHT: Shtr1, Button 8, hold 36, dir 0, index 7, payload 1, last state 0, press counter 0, window 0
22:07:04.096 RSL: RESULT = {"Button8":{"Action":"CLEAR"}}
```

### TO REPRODUCE
Hold the button configured using ShutterButton, nothing happens.

### EXPECTED BEHAVIOUR
I expect that holding the button should start moving the shutters toward the configured position and stop upon reaching it or upon the button release.

### SCREENSHOTS
I have no screenshots to provide.

### ADDITIONAL CONTEXT
I have no additional context to provide.

**(Please, remember to close the issue when the problem has been addressed)**

Thank you for your time and for this wonderful piece of software which Tasmota is!

stefanbode, 2024-09-08:
Yes. In the good old days at esp8266 you could define a hold event on a button. I assume the documentation needs an update. As far as I remember I removed the hold event by purpose because most intuitive (like my wife) the user is expediting that the shutter moves as long as they hold and immediately stop on release. There were some side effects I removed the hold on esp32 by purpose.

markatom, 2024-09-08:
> most intuitive (like my wife) the user is expediting that the shutter moves as long as they hold and immediately stop on release

I agree that this is the most intuitive for the user and is the behavior I want. However, the issue is, that my shutters do not move while holding the button.

stefanbode, 2024-09-08:
I believe there is a good desciption in the documentation about this. On this is the shutter STARTS moving with the hold event. Therefore this should not to be too long. otherwhise it feel unnatural. Also there is a option to DISBALE the TASMOTA reset if you holde the button too long. I assume this is also not wanted.

markatom, 2024-09-08:
I am afraid we may not understand each other.

I have configured the delay between the hold and the actual sending of the HOLD event to 0.5 seconds:
```
SetOption32 5
```

I have also disabled Tasmota reset on long press:
```
SetOption1 1
```

See the attached configuration and logs I provided earlier with the description. All of this is there. I can even see the HOLD event in the attached logs:

```
22:07:00.995 RSL: RESULT = {"Button8":{"Action":"HOLD"}}
```

However, no relay is ever engaged while I am holding the button (even if I hold it for a very long time – eg. 30 seconds).

stefanbode, 2024-09-08:
Here is my button definition 

MQT: hm/status/ESP_5517A3/RESULT = {"ShutterButton1":"1 100 50 74 100 - - - - -|2 0 50 24 0 - - - - -"}

Here the expected behavior works on esp32

stefanbode, 2024-09-08:
There should be no difference if you just use button 7 and 8. In the log file the trigger on HOLD gets executed. Quite unsure what blocks the shutter to start. All triggers are met. Does normal one or double click work?

markatom, 2024-09-08:
> Does normal one or double click work?

Yes, single, double, even triple press work.

stefanbode, 2024-09-08:
Ok, looks like I have to configure it here, too. Absolutely no guess what is wrong. Will do this later afternoon:
Esp32, one shutter on relay1,2, button7,8 configured as shutterbutton, normal shutter with interlock.

This sounds so simple and straight forward.......

stefanbode, 2024-09-08:
regression ACCEPTED. Fix on the way.

stefanbode, 2024-09-08:
Small explanation: Buttons as long as not defined as shutterbuttons works without problems. As shutterbuttons are defined there is a procedure to give back control if the multipress seems not belong to the shutter. There are several multipress events that reset some settings on 5,6,7 ... presses. The if statement just gives back control already on "4", but this was used to "mark" the hold event. internally it comes as 4 presses. Now it circle back on >4 and not on >3. Tested and worked

markatom, 2024-09-08:
Thank you for the fix and explanation! Will you merge your patch into `development` branch today? I am excited to try it out later today.

markatom, 2024-09-08:
I see you already merge it. 🙂

markatom, 2024-09-08:
I am still unable to get it working. What does your configuration look like?

I double-checked I have the latest development version (14.2.0.4) and everything is set correctly up.

I also tried a different pair of buttons that have no ShutterButtons on them. I can see their corresponding SINGLE, DOUBLE, HOLD, and CLEAR events in the console, but no relay would engage.

stefanbode, 2024-09-09:
This is strange. With the changed version and YOUR shutterbutton config it should look like this when you hold a button. Line 3 MUST be there because in line 3 already on the old version there was this "press cnt 4"

```
14:34:36.642 Start shutter after long press
14:34:36.653 MULTI: SHT: Shtr1, Btn 8, hold 10, dir 0, idx 7, payload 0, last state 0, press cnt 4, window 0, simu_press 0
14:34:36.655 SHT: Shtr1, Button 8 = 4 (single=1, double=2, tripple=3, hold=4)
14:34:36.667 SHT: Shtr1 -> 100
14:34:36.668 SHT: Shtr1 -> Endpoint movement detected at 100. Set Tilt: 0
14:34:36.680 SHT: Target tilt -128 for button 8
14:34:36.682 SHT: Target tilt -128 for button 8
14:34:36.683 SHT: Pos. payload <> (0), payload 100, idx 1 (1), src 3
14:34:36.695 SHT: lastsource 3:, real 100000, target 0, tiltreal: 0, tilttarget: 0, payload 0
14:34:36.696 SRC: Shutter
14:34:36.703 MQT: hm/status/ESP_F506CC/RESULT = {"POWER2":"ON"}
14:34:36.706 MQT: hm/status/ESP_F506CC/POWER2 = ON
14:34:36.708 SHT: Shtr1 Real 100000, Start 100000, Stop 0, Dir -1, Delay 40, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
```

stefanbode, 2024-09-11:
@markatom : Can you share some logs and check again. As mentioned it should work with the new version ESP32 only, no changes to ESP8266

markatom, 2024-09-11:
Here are logs with the same setup (ESP32), development version `14.2.0.4`, right before I hold the button:

```
21:43:52.959 SHT: Shtr1, Button 8, hold 0, dir 0, index 7, payload 0, last state 1, press counter 0, window 0
21:43:52.962 BTN: Button8 multi-press 1
21:43:53.009 SHT: Shtr1, Button 8, hold 1, dir 0, index 7, payload 0, last state 0, press counter 1, window 9
21:43:53.059 SHT: Shtr1, Button 8, hold 2, dir 0, index 7, payload 0, last state 0, press counter 1, window 8
21:43:53.109 SHT: Shtr1, Button 8, hold 3, dir 0, index 7, payload 0, last state 0, press counter 1, window 7
21:43:53.159 SHT: Shtr1, Button 8, hold 4, dir 0, index 7, payload 0, last state 0, press counter 1, window 6
21:43:53.209 SHT: Shtr1, Button 8, hold 5, dir 0, index 7, payload 0, last state 0, press counter 1, window 5
21:43:53.259 SHT: Shtr1, Button 8, hold 6, dir 0, index 7, payload 0, last state 0, press counter 1, window 4
21:43:53.309 SHT: Shtr1, Button 8, hold 7, dir 0, index 7, payload 0, last state 0, press counter 1, window 3
21:43:53.359 SHT: Shtr1, Button 8, hold 8, dir 0, index 7, payload 0, last state 0, press counter 1, window 2
21:43:53.409 SHT: Shtr1, Button 8, hold 9, dir 0, index 7, payload 0, last state 0, press counter 1, window 1
21:43:53.413 RSL: RESULT = {"Button8":{"Action":"HOLD"}}
21:43:53.495 SHT: Shtr1, Button 8, hold 10, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.508 SHT: Shtr1, Button 8, hold 11, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.559 SHT: Shtr1, Button 8, hold 12, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.609 SHT: Shtr1, Button 8, hold 13, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.659 SHT: Shtr1, Button 8, hold 14, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.709 SHT: Shtr1, Button 8, hold 15, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.759 SHT: Shtr1, Button 8, hold 16, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.809 SHT: Shtr1, Button 8, hold 17, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.859 SHT: Shtr1, Button 8, hold 18, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.909 SHT: Shtr1, Button 8, hold 19, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:53.959 SHT: Shtr1, Button 8, hold 20, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.009 SHT: Shtr1, Button 8, hold 21, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.059 SHT: Shtr1, Button 8, hold 22, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.109 SHT: Shtr1, Button 8, hold 23, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.159 SHT: Shtr1, Button 8, hold 24, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.209 SHT: Shtr1, Button 8, hold 25, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.259 SHT: Shtr1, Button 8, hold 26, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.309 SHT: Shtr1, Button 8, hold 27, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.359 SHT: Shtr1, Button 8, hold 28, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.409 SHT: Shtr1, Button 8, hold 29, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.459 SHT: Shtr1, Button 8, hold 30, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.509 SHT: Shtr1, Button 8, hold 31, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.559 SHT: Shtr1, Button 8, hold 32, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.609 SHT: Shtr1, Button 8, hold 33, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.659 SHT: Shtr1, Button 8, hold 34, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.709 SHT: Shtr1, Button 8, hold 35, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.759 SHT: Shtr1, Button 8, hold 36, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.809 SHT: Shtr1, Button 8, hold 37, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.859 SHT: Shtr1, Button 8, hold 38, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.909 SHT: Shtr1, Button 8, hold 39, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:54.959 SHT: Shtr1, Button 8, hold 40, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:55.009 SHT: Shtr1, Button 8, hold 41, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:55.060 SHT: Shtr1, Button 8, hold 42, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:55.109 SHT: Shtr1, Button 8, hold 43, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:55.159 SHT: Shtr1, Button 8, hold 44, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:55.209 SHT: Shtr1, Button 8, hold 45, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:55.259 SHT: Shtr1, Button 8, hold 46, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
[... cut ...]
21:43:57.945 SHT: Shtr1, Button 8, hold 98, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:57.996 SHT: Shtr1, Button 8, hold 99, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:58.045 SHT: Shtr1, Button 8, hold 100, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
21:43:58.095 SHT: Shtr1, Button 8, hold 101, dir 0, index 7, payload 1, last state 0, press counter 0, window 0
21:43:58.100 RSL: RESULT = {"Button8":{"Action":"CLEAR"}}
```

markatom, 2024-09-11:
ShutterButton config:

```
21:55:19.272 RSL: RESULT = {"ShutterButton1":"7 0 - - - - - - -|8 100 - - - - - - -"}
```

Status 0:
```
21:56:02.369 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","","","","","","",""],"Topic":"tasmota_A8A174","ButtonTopic":"0","Power":"00000000","PowerLock":"00000000","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:56:02.375 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Vbat power on reset","Uptime":"0T00:15:26","StartupUTC":"2024-09-11T20:40:36","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2024-09-08T18:32:09","SaveCount":19}}
21:56:02.379 RSL: STATUS2 = {"StatusFWR":{"Version":"14.2.0.4(tasmota)","BuildDateTime":"2024-09-08T17:07:25","Core":"3_0_4","SDK":"5.1.4.240822","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"344/699"}}
21:56:02.383 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mcz iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C00B","0505C80001000600003C5A0A192800000000","40800080","00006000","00004000","00000000"]}}
21:56:02.391 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1470,"Free":1409,"Heap":140,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F9AC7CF","00148001","000000CF","010013C0","C0000B81","40004080","00200000","5400002C","00000080","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,12,!16,!20,!21,!24,26,27,28,29,!35,38,50,52,62,!68,82,!121","Sensors":"1,2,3,5,6,127","I2CDriver":"2,7"}}
21:56:02.398 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-A8A174-0372","IPAddress":"10.0.1.237","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.1.1","DNSServer2":"0.0.0.0","Mac":"90:38:0C:A8:A1:74","IP6Global":"","IP6Local":"fe80::9238:cff:fea8:a174%st2","Ethernet":{"Hostname":"tasmota-A8A174-0372-eth","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.0.1.1","DNSServer2":"0.0.0.0","Mac":"90:38:0C:A8:A1:77","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:56:02.404 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A8A174","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:56:02.409 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-09-11T20:56:02Z","Local":"2024-09-11T21:56:02","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:22","Sunset":"19:10"}}
21:56:02.415 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-09-11T21:56:02","Shutter1":{"Position":23,"Direction":0,"Target":23,"Tilt":0},"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
21:56:02.428 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-09-11T21:56:02","Uptime":"0T00:15:26","UptimeSec":926,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":34},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"mcz iot","BSSId":"D2:21:F9:3C:84:C5","Channel":1,"Mode":"HT20","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:56:02.435 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT2":{"Relay1":5,"Relay2":6,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT3":{"Relay1":7,"Relay2":8,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]}}}
```

stefanbode, 2024-09-12:
It is clear why it does not work in your file because you did not define the HOLD event on the shutterbutton. Please use the following commands to define the hold event:
ShutterButton1 7 0 - - 0
ShutterButton1 8 100 - - 100

markatom, 2024-09-12:
I see, sorry about that. Now I have the correct ShutterButton config, see:

```
19:45:37.803 RSL: RESULT = {"ShutterButton1":"7 0 - - 0 - - - -|8 100 - - 100 - - - -"}
```

But still, it does not work, see these logs:

```
19:46:12.277 SHT: Shtr1, Button 8, hold 0, dir 0, index 7, payload 0, last state 1, press counter 0, window 0
19:46:12.280 BTN: Button8 multi-press 1
19:46:12.327 SHT: Shtr1, Button 8, hold 1, dir 0, index 7, payload 0, last state 0, press counter 1, window 9
19:46:12.377 SHT: Shtr1, Button 8, hold 2, dir 0, index 7, payload 0, last state 0, press counter 1, window 8
19:46:12.427 SHT: Shtr1, Button 8, hold 3, dir 0, index 7, payload 0, last state 0, press counter 1, window 7
19:46:12.477 SHT: Shtr1, Button 8, hold 4, dir 0, index 7, payload 0, last state 0, press counter 1, window 6
19:46:12.527 SHT: Shtr1, Button 8, hold 5, dir 0, index 7, payload 0, last state 0, press counter 1, window 5
19:46:12.577 SHT: Shtr1, Button 8, hold 6, dir 0, index 7, payload 0, last state 0, press counter 1, window 4
19:46:12.627 SHT: Shtr1, Button 8, hold 7, dir 0, index 7, payload 0, last state 0, press counter 1, window 3
19:46:12.677 SHT: Shtr1, Button 8, hold 8, dir 0, index 7, payload 0, last state 0, press counter 1, window 2
19:46:12.727 SHT: Shtr1, Button 8, hold 9, dir 0, index 7, payload 0, last state 0, press counter 1, window 1
19:46:12.731 RSL: RESULT = {"Button8":{"Action":"HOLD"}}
19:46:12.777 SHT: Shtr1, Button 8, hold 10, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:12.827 SHT: Shtr1, Button 8, hold 11, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:12.877 SHT: Shtr1, Button 8, hold 12, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:12.927 SHT: Shtr1, Button 8, hold 13, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.028 SHT: Shtr1, Button 8, hold 14, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.085 SHT: Shtr1, Button 8, hold 15, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.135 SHT: Shtr1, Button 8, hold 16, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.185 SHT: Shtr1, Button 8, hold 17, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.235 SHT: Shtr1, Button 8, hold 18, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.286 SHT: Shtr1, Button 8, hold 19, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.335 SHT: Shtr1, Button 8, hold 20, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.385 SHT: Shtr1, Button 8, hold 21, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.435 SHT: Shtr1, Button 8, hold 22, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.485 SHT: Shtr1, Button 8, hold 23, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.535 SHT: Shtr1, Button 8, hold 24, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.585 SHT: Shtr1, Button 8, hold 25, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.635 SHT: Shtr1, Button 8, hold 26, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.685 SHT: Shtr1, Button 8, hold 27, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.735 SHT: Shtr1, Button 8, hold 28, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.785 SHT: Shtr1, Button 8, hold 29, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.835 SHT: Shtr1, Button 8, hold 30, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.885 SHT: Shtr1, Button 8, hold 31, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.935 SHT: Shtr1, Button 8, hold 32, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:13.985 SHT: Shtr1, Button 8, hold 33, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.035 SHT: Shtr1, Button 8, hold 34, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.085 SHT: Shtr1, Button 8, hold 35, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.135 SHT: Shtr1, Button 8, hold 36, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.185 SHT: Shtr1, Button 8, hold 37, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.235 SHT: Shtr1, Button 8, hold 38, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.287 SHT: Shtr1, Button 8, hold 39, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.335 SHT: Shtr1, Button 8, hold 40, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.385 SHT: Shtr1, Button 8, hold 41, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.435 SHT: Shtr1, Button 8, hold 42, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.485 SHT: Shtr1, Button 8, hold 43, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.535 SHT: Shtr1, Button 8, hold 44, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.585 SHT: Shtr1, Button 8, hold 45, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
[... cut ...]
19:46:14.835 SHT: Shtr1, Button 8, hold 50, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.885 SHT: Shtr1, Button 8, hold 51, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.935 SHT: Shtr1, Button 8, hold 52, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:14.985 SHT: Shtr1, Button 8, hold 53, dir 0, index 7, payload 1, last state 0, press counter 0, window 0
19:46:14.990 RSL: RESULT = {"Button8":{"Action":"CLEAR"}}
```

stefanbode, 2024-09-13:
Hmm, I do not have access to a testbench until monday. I am missing in your logs the section that looks like:
```
22:06:43.583 RSL: RESULT = {"WebLog":4}
22:06:47.378 MULTI: SHT: Shtr1, Btn 7, hold 13, dir 0, idx 6, payload 0, last state 0, press cnt 4, window 0, simu_press 0
22:06:47.425 SHT: Shtr1, Button 7, hold 14, dir 0, index 6, payload 0, last state 0, press counter 4, window 0
22:06:47.427 Start shutter after long press
22:06:47.428 MULTI: SHT: Shtr1, Btn 7, hold 14, dir 0, idx 6, payload 0, last state 0, press cnt 4, window 0, simu_press 0
22:06:47.475 SHT: Shtr1, Button 7, hold 15, dir 0, index 6, payload 0, last state 0, press counter 4, window 0
22:06:47.477 Start shutter after long press
```
This should get called if the window = 0 and press_counter > 0:
```
19:46:12.677 SHT: Shtr1, Button 8, hold 8, dir 0, index 7, payload 0, last state 0, press counter 1, window 2
19:46:12.727 SHT: Shtr1, Button 8, hold 9, dir 0, index 7, payload 0, last state 0, press counter 1, window 1
19:46:12.731 RSL: RESULT = {"Button8":{"Action":"HOLD"}}
19:46:12.777 SHT: Shtr1, Button 8, hold 10, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
19:46:12.827 SHT: Shtr1, Button 8, hold 11, dir 0, index 7, payload 0, last state 0, press counter 0, window 0
```
For me it looks like the window=0 does not get met.

stefanbode, 2024-09-13:
Maybe `SetOption32 10`  can make a difference to call the HOLD after one second. I assume you did not change `Setoption12 0` which is the default on any system

markatom, 2024-09-13:
@stefanbode You are right. Any value >= 6 for `SetOption32` works. Smaller interval will not trigger a relay.

stefanbode, 2024-09-13:
Great thanks for the info. Let me check if there is a reason for it.

crllsn, 2024-09-29:
If it can help you, version 12.5.0 works even with `SetOption32 5`. I did not test version 13, but version 14.2.0 and 14.2.0.6 need at least `SetOption32 6`.

stefanbode, 2024-09-29:
Yes, this is a known limitation. Maybe @arendst theo has an idea why setoption32 behave different if 5 or below.

arendst, 2024-09-29:
Except if you configure ``ButtonDebounce 500`` I have no idea why setoption32 below 6 doesn't work.

On the other hand, a hold delay of 600mS to detect a hold state look reasonable to me.

stefanbode, 2024-10-24:
@crllsn : can we close this case or do we need more investigation to get SetOption32 5 working again?

crllsn, 2024-11-23:
@stefanbode : you can close it. Changed all my configs and SetOption32 6 is reasonably ok. Thanks!

closed 2024-11-23 by arendst