arendst/Tasmota #22005: Exception when get zigbee started information

vensole, 2024-08-24:
### PROBLEM DESCRIPTION
I want to get zigbee state periodically, so i using berry to query tasmota. Here is my simple code:

import zigbee

print(zigbee.started())

And i have got this exception:

BRY: Exception> 'attribute_error' - the 'zb_coord' object has no attribute 'started'

So does anyone know what my problem is, thanks!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-08-24:
As you did not fill in any of the requested info, hard to make any guesses. I'm getting `true` when I do the same thing one a working Zigbee setup using Zigbee2Tasmota.

In case you're using it for ZHA or Zigbee2MQTT, the Tasmota device is not running anything Zigbee, merely providing a tunnel for the coordinator application to talk directly to the Zigbee chip. Hence nothing can work if you look for Zigbee-related functions in Tasmota.

s-hadinger, 2024-08-24:
`zigbee.started()` is a recent addition. But as sfromis stated, you share nothing about your environment. We can only try to guess what you have and what you want to do. Definitely not the best way to proceed.

s-hadinger, 2024-08-24:
Closing as the template was not filled. Will reopen if necessary information is provided

closed 2024-08-24 by s-hadinger