arendst/Tasmota #22007: Exception when use tuya zigbee smart knob

tasict, 2024-08-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:39:05.401 MQT: stat/ZBPro/STATUS = {"Status":{"Module":0,"DeviceName":"ZBPro","FriendlyName":["ZBPro"],"Topic":"ZBPro","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:39:05.463 MQT: stat/ZBPro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:54","StartupUTC":"2024-08-24T14:35:11","Sleep":0,"CfgHolder":4617,"BootCount":794,"BCResetTime":"2023-11-11T09:50:51","SaveCount":1165}}
22:39:05.523 MQT: stat/ZBPro/STATUS2 = {"StatusFWR":{"Version":"14.2.0.3(831597e-zbbrdgpro)","BuildDateTime":"2024-08-23T09:26:00","Core":"3_0_4","SDK":"5.1.4.240801","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"407/699"}}
22:39:05.560 MQT: stat/ZBPro/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NO46-2.4G","NO46"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000695A0A192800000000","00000080","40006082","00004000","00000000"]}}
22:39:05.615 MQT: stat/ZBPro/STATUS4 = {"StatusMEM":{"ProgramSize":1576,"Free":279,"Heap":124,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F8A87CF","00548001","000000CF","010013C0","C0000981","40024080","00200000","5400082C","00000080","00000004"],"Drivers":"1,2,!3,!4,!5,!8,9,10,12,!16,!20,!21,23,24,!27,29,!35,38,!41,50,52,!59,62,!68,82,!121","Sensors":"1,2,3,5,6,127","I2CDriver":"7"}}
22:39:05.688 MQT: stat/ZBPro/STATUS5 = {"StatusNET":{"Hostname":"ZBPro","IPAddress":"10.1.8.25","Gateway":"10.1.8.1","Subnetmask":"255.255.252.0","DNSServer1":"10.1.8.1","DNSServer2":"8.8.8.8","Mac":"40:22:D8:60:7D:04","IP6Global":"fd00:88::4222:d8ff:fe60:7d04","IP6Local":"fe80::4222:d8ff:fe60:7d04%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.1.8.1","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
22:39:05.761 MQT: stat/ZBPro/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.ctsite.tk","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_607D04","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":15}}
22:39:05.810 MQT: stat/ZBPro/STATUS7 = {"StatusTIM":{"UTC":"2024-08-24T14:39:05Z","Local":"2024-08-24T22:39:05","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+08:00","Sunrise":"05:31","Sunset":"18:19"}}
22:39:05.848 MQT: stat/ZBPro/STATUS10 = {"StatusSNS":{"Time":"2024-08-24T22:39:05"}}
22:39:05.876 MQT: stat/ZBPro/STATUS11 = {"StatusSTS":{"Time":"2024-08-24T22:39:05","Uptime":"0T00:03:54","UptimeSec":234,"Heap":121,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Berry":{"HeapUsed":26,"Objects":402},"Wifi":{"AP":2,"SSId":"NO46","BSSId":"E4:38:83:8C:F8:AE","Channel":11,"Mode":"HT20","RSSI":100,"Signal":-21,"LinkCount":1,"Downtime":"0T00:00:04"}}}
22:39:05.938 MQT: stat/ZBPro/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"LoadProhibited","EPC":"4011c806","EXCVADDR":"00000000","CallChain":["4011c803","401417c2","4013f858","40139aa1","4015e2b6","401402ac","4013f858","401384e0","40139777","40133bd0","4015ec76","4010744f","401034dd","4015f3fb","40109805","40127df7","400dea1c","400d93d1","401501ea"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
I encountered an issue while using the Zigbee device Smart Knob (Tuya ERS-10TZBVK-AA) with Tasmota 14.2.0.3 (831597e-zbbrdgpro). When performing a Short Single Press, the ZB Pro crashes and restarts.

### EXPECTED BEHAVIOUR
22:35:16.559 MQT: tele/ZBPro/INFO1 = {"Info1":{"Module":"Sonoff Zigbee Pro","Version":"14.2.0.3(831597e-zbbrdgpro)","FallbackTopic":"cmnd/DVES_607D04_fb/","GroupTopic":"cmnd/tasmotas/"}}
22:35:16.596 MQT: tele/ZBPro/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ZBPro","IPAddress":"10.1.8.25","IP6Global":"","IP6Local":"fe80::4222:d8ff:fe60:7d04%st1"}}
22:35:16.630 MQT: tele/ZBPro/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"LoadProhibited","EPC":"4011c806","EXCVADDR":"00000000","CallChain":["4011c803","401417c2","4013f858","40139aa1","4015e2b6","401402ac","4013f858","401384e0","40139777","40133bd0","4015ec76","4010744f","401034dd","4015f3fb","40109805","40127df7","400dea1c","400d93d1","401501ea"]},"BootCount":794}}
22:35:19.311 QPC: Reset

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-08-24:
It's hard to debug without a detailed log from serial. Do you have the opportunity to plug the ZBPro to serial and set `seriallog 3`?

tasict, 2024-08-25:
I suspect that there might be an issue with the Koenkk Zigbee firmware. If I try to patch the latest Koenkk firmware, CC1352P2_CC2652P_launchpad_coordinator_20240710.zip, onto the Sonoff Zigbee Pro, similar to the modifications you made earlier, how should I proceed?

s-hadinger, 2024-08-25:
I have ordered the device, but it will take weeks for delivery

tasict, 2024-08-25:
Wow, let me describe it in more detail. This device has two modes: command and event. In the default command mode, rotating the knob doesn’t cause any issues, but single, double, or hold presses all result in a crash. A quick triple press switches it to event mode, and in event mode, everything works fine. I’ll also try to find some time to dump the crash log.

s-hadinger, 2024-08-26:
Ok, any log that makes it possible to reproduce the problem is useful.

closed 2024-09-25 by github-actions[bot]

s-hadinger, 2024-09-28:
I received my devices and tested in both modes, I don't have any crash:

Command mode:
```
(switch to command mode)
11:02:05.999 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006/8004":0,"Endpoint":1,"LinkQuality":145}}}

11:02:23.468 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006!02":"","Power":2,"Endpoint":1,"LinkQuality":163}}}
11:02:26.383 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0008!02":"00190100","DimmerStepUp":25,"Endpoint":1,"LinkQuality":153}}}
11:02:28.833 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0008!02":"01190100","DimmerStepDown":25,"Endpoint":1,"LinkQuality":149}}}
11:02:31.979 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0300!4C":"01120001009900F401","ColorTempStepUp":18,"Endpoint":1,"LinkQuality":120}}}
11:02:33.506 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0300!4C":"03120001009900F401","ColorTempStepDown":18,"Endpoint":1,"LinkQuality":94}}}
11:02:37.333 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0300!04":"01C8","SatMove":1,"Endpoint":1,"LinkQuality":149}}}
11:02:38.309 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0300!01":"010F","HueMove":1,"Endpoint":1,"LinkQuality":149}}}
11:02:38.569 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0300!47":"","Endpoint":1,"LinkQuality":134}}}
```

Scene mode:
```
(switch to scene mode)
11:04:00.049 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006/8004":1,"Endpoint":1,"LinkQuality":167}}}

11:04:08.922 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006!FD":"00","LidlPower":0,"Endpoint":1,"LinkQuality":163}}}
11:04:11.685 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006!FC":"00","Power":252,"Endpoint":1,"LinkQuality":149}}}
11:04:13.661 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006!FC":"01","Power":252,"Endpoint":1,"LinkQuality":163}}}
11:04:23.978 RSL: SENSOR = {"ZbReceived":{"0x4485":{"Device":"0x4485","Name":"Rotary","0006!FD":"02","LidlPower":2,"Endpoint":1,"LinkQuality":167}}}
```

tasict, 2024-09-29:
Thanks to your response, I first updated the firmware to the 0928 daily build and tried to reproduce the reboot issue. Surprisingly, it no longer occurs on the latest version, and everything is functioning properly! Thank you so much for your assistance.