arendst/Tasmota #22020: Sonoff ZBBridge Pro: No connection to HA and no Zigbee buttons

Jambola77, 2024-08-26:
### PROBLEM DESCRIPTION
I have succesfully flashed a Sonoff ZBBridge Pro with tasmota software (14.0) and was able to setup a Sonoff Zigbee Smart Water Valve SWV. Then, after an upgrade to tasmota 14.2.0.3 (831597e-zbbrdgpro), the bridge to HomeAssistant is not working anymore and I cannot join Zigbee devices in tasmota. The menu button disappeared.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff SWV Smart Water Valve
- [x] Tasmota binary firmware version number used: 14.2.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Webflasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
```
16:44:54.503 MQT: stat/ZigbeeBridge/RESULT = {"NAME":"TCP ZBBridge Pro","GPIO":[0,0,576,0,480,0,0,0,0,1,1,5792,0,0,0,5472,0,320,5793,5504,0,640,608,32,0,0,0,0,0,1,0,0,0,0,0,0],"FLAG":0,"BASE":1}
16:44:54.724 MQT: stat/ZigbeeBridge/RESULT = {"Module":{"0":"TCP ZBBridge Pro"}}
16:44:54.932 MQT: stat/ZigbeeBridge/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5792":"Zigbee Rst1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"5472":"TCP Tx"},"GPIO20":{"0":"None"},"GPIO21":{"320":"Led_i1"},"GPIO22":{"5793":"Zigbee Rst2"},"GPIO23":{"5504":"TCP Rx"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"608":"I2C SCL1"},"GPIO27":{"32":"Button1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:45:57.624 MQT: stat/ZigbeeBridge/STATUS = {"Status":{"Module":0,"DeviceName":"ZigBridge","FriendlyName":["ZigBridge"],"Topic":"ZigbeeBridge","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:45:57.660 MQT: stat/ZigbeeBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"0T00:09:29","StartupUTC":"2024-08-26T14:36:28","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2024-08-19T07:41:15","SaveCount":32}}
16:45:57.699 MQT: stat/ZigbeeBridge/STATUS2 = {"StatusFWR":{"Version":"14.2.0.3(831597e-zbbrdgpro)","BuildDateTime":"2024-08-23T09:26:00","Core":"3_0_4","SDK":"5.1.4.240801","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"412/699"}}
16:45:57.717 MQT: stat/ZigbeeBridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HalloNachbar",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:45:57.753 MQT: stat/ZigbeeBridge/STATUS4 = {"StatusMEM":{"ProgramSize":1576,"Free":279,"Heap":155,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F8A87CF","00548001","000000CF","010013C0","C0000981","40024080","00200000","5400082C","00000080","00000004"],"Drivers":"1,2,!3,!4,!5,!8,9,10,12,!16,!20,!21,!23,24,!27,29,!35,38,41,50,52,!59,62,!68,82,!121","Sensors":"1,2,3,5,6,127","I2CDriver":"7"}}
16:45:57.806 MQT: stat/ZigbeeBridge/STATUS5 = {"StatusNET":{"Hostname":"ZigbeeBridge-1752","IPAddress":"192.168.1.247","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.100","DNSServer2":"2a04:4540:8c0b:d800:9a9b:cbff:fe96:ae3","Mac":"E4:65:B8:54:46:D8","IP6Global":"2a04:4540:8c0b:d800:e665:b8ff:fe54:46d8","IP6Local":"fe80::e665:b8ff:fe54:46d8%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.100","DNSServer2":"2a04:4540:8c0b:d800:9a9b:cbff:fe96:ae3","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
16:45:57.866 MQT: stat/ZigbeeBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5446D8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:45:57.894 MQT: stat/ZigbeeBridge/STATUS7 = {"StatusTIM":{"UTC":"2024-08-26T14:45:57Z","Local":"2024-08-26T16:45:57","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"06:59","Sunset":"20:43"}}
16:45:57.912 MQT: stat/ZigbeeBridge/STATUS10 = {"StatusSNS":{"Time":"2024-08-26T16:45:57"}}
16:45:57.923 MQT: stat/ZigbeeBridge/STATUS11 = {"StatusSTS":{"Time":"2024-08-26T16:45:57","Uptime":"0T00:09:29","UptimeSec":569,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":47},"Wifi":{"AP":1,"SSId":"HalloNachbar","BSSId":"98:9B:CB:96:0A:E5","Channel":11,"Mode":"HT20","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}}
16:45:57.962 MQT: stat/ZigbeeBridge/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"LoadProhibited","EPC":"40184b02","EXCVADDR":"00000000","CallChain":["40184aff","40174b6a","40184ddd","40184e3b","40174442","40128e67","4010fbe8","400d442f","4010ba00","40127df7","400d9049","400d920a","40114644","401417c2","4013f858","401384e0","40139777","40133bd0","4011b21b","4011e92c","40127df7","400dea1c","400d93d1","401501ea"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash the software and try to add a zigbee device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Menu buttons are active. Server runs on port 8888

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-08-26:
> Download https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file password: changeme In the installer menu, select "gcc."

What is this supposed to be? Malware?

sfromis, 2024-08-26:
When you configure it for TCP access (ser2net tunnel), Tasmota is not in any way involved in Zigbee operation, as it is all up to what's on the other end, like ZHA or Zigbee2MQTT. Hence, there will be no Zigbee buttons in the Tasmota web UI, and no pairing there either.

Did you run the command `TCPStart 8888` to have it listen to that port for the tunnel?