arendst/Tasmota #22025: KNX Alexa not available

SpeedyBlade, 2024-08-26:
Hello

why is Alexa not available when KNX Tasmota was flashed?

br

s-hadinger, 2024-08-26:
I suppose that you are talking about `tasmota-knx.bin`, which indeeds has Alexa Hue disabled.

I'm not sure of the original reason, but most certainly to make the firmware small enough. Full Hue emulation takes 16KB of Flash.

You can compile your own and enable both KNX and Alexa Hue emulation

SpeedyBlade, 2024-08-26:
yes...you are fully right ;)

But i am not able to compile my own .bin file that i can also enable both (KNX and Alexa Hue emulation).

s-hadinger, 2024-08-26:
There are many simple options to compile yourself, like TasmoCompiler 

https://tasmota.github.io/docs/Compile-your-build/#online-compilers

SpeedyBlade, 2024-08-26:
it would be better, that the orignal tasmota-knx.bin will be updated and give Alexa Hue emulation as an option ;)

s-hadinger, 2024-08-26:
Option would mean yet another firmware to the list. It's easy enough with TasmoCompiler