arendst/Tasmota #22062: Problems receiving data from vbus interface (sensor53) since Release 14.0.0

eikejenn, 2024-08-31:
Hi together, 
i have a problem with Tasmota (sensor53 metricdata receiving from Resol vbus interface) since Tasmota Version 14.0.0.

I used a self compiled Image for ESP8266
My User settings for compiling:
```
#undef  USE_RULES
#define USE_SCRIPT

#define USE_SCRIPT_JSON_EXPORT
#define USE_SCRIPT_WEB_DISPLAY
// SML stuff
#define USE_SML_M
#define USE_SML_SCRIPT_CMD
#define USE_SCRIPT_JSON_EXPORT
#define USE_SCRIPT_WEB_DISPLAY
#define SML_REPLACE_VARS
#define SML_MAX_VARS 30
#define SML_DUMP_SIZE 200
```

Compiling with Tasmota-13.3.0 is running fine.
Iam using this simple script to get four temperatures from the vbus data.
```
>D
r="1,AA10007B421000010918"

>B
=>sensor53 r

>M 1
+1,3,v,0,9600,Solar
%r%vo0sw@10,Kollektor,°C,koll,1
%r%vo2sw@10,Speicher 1,°C,sp1,1
%r%vo4sw@10,Speicher 2,°C,sp2,1
%r%vo6sw@10,Sensor 54,°C,s4,1
```

Now i want to update to the newest release (14.2.0). 
compiling with the same usersettings. After flashing the image, all metrics shows 0.0 in the web gui and the json in the mqtt also shows 0.0.
enabling the vbus dump (sensor53 d1) shows the vbus data and i think this is all correct. (i didnt change the hardware, because in 13.3.0 is all running fine.)

```
20:11:33.062 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:33.735 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:34.487 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:36.061 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:36.657 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:39.028 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:39.680 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:40.318 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f c0 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 3b 01 48 01 00 7a 
20:11:41.236 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:41.783 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:42.364 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:44.004 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:44.655 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:45.295 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:46.049 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:46.648 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:47.341 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:48.994 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:49.643 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:51.041 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 3b 
20:11:51.639 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:52.329 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:54.235 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:54.832 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 02 78 64 7f 07 00 02 13 0b 06 7f 7f 0c 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
20:11:55.359 : aa 15 00 7b 42 10 00 01 0a 12 01 08 00 00 00 76 00 64 7f 7f 0c 11 02 0a 00 00 00 73 66 01 73 02 01 22 00 00 00 00 00 7f 01 0b 00 00 00 73 00 00 00 00 00 7f 02 01 00 0b 00 71 66 01 73 02 01 22 
20:11:57.232 : aa 00 00 7b 42 20 00 05 00 00 00 00 00 00 00 1d 
20:11:57.781 : aa 10 00 7b 42 10 00 01 09 18 66 01 73 02 01 22 3b 01 48 01 00 7a 00 7f 06 00 13 86 64 00 00 7d 00 04 7e 00 00 00 00 00 7f 00 00 00 00 00 7f 68 00 00 00 00 17 
[... 13 lines cut ...]
development -> not running

The difference between can be this:
Tasmota-13.3.0: PLATFORM: Espressif 8266 (2023.4.0) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS
Tasmota-14.0.0: PLATFORM: Espressif 8266 (2024.1.1) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS

What can i do to narrow down the error?

status 0 shows:
```
20:20:52.249 MQT: TASMOTA/stat/255/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"255","ButtonTopic":"255","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:20:52.256 MQT: TASMOTA/stat/255/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:19:35","StartupUTC":"2024-08-31T19:01:17","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2024-08-31T20:01:16","SaveCount":6,"SaveAddress":"F6000"}}
20:20:52.264 MQT: TASMOTA/stat/255/STATUS2 = {"StatusFWR":{"Version":"14.2.0.3(eike-sml-vbus)","BuildDateTime":"2024-08-31T20:57:11","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"384/699"}}
20:20:52.270 MQT: TASMOTA/stat/255/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["48414A454E4E",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
20:20:52.285 MQT: TASMOTA/stat/255/STATUS4 = {"StatusMEM":{"ProgramSize":709,"Free":292,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","879AC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6,53","I2CDriver":"7"}}
20:20:52.293 MQT: TASMOTA/stat/255/STATUS5 = {"StatusNET":{"Hostname":"255-0156","IPAddress":"192.168.20.146","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.20.1","DNSServer2":"0.0.0.0","Mac":"18:FE:34:E3:80:9C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:20:52.299 MQT: TASMOTA/stat/255/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E3809C","MqttUser":"server","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
[... cut ...]
20:20:52.319 MQT: TASMOTA/stat/255/STATUS10 = {"StatusSNS":{"Time":"2024-08-31T20:20:52","Solar":{"koll":0.0,"sp1":0.0,"sp2":0.0,"s4":0.0,"r1":0,"r2":0}}}
20:20:52.327 MQT: TASMOTA/stat/255/STATUS11 = {"StatusSTS":{"Time":"2024-08-31T20:20:52","Uptime":"0T00:19:35","UptimeSec":1175,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"48414A454E4E","BSSId":"72:A7:41:3E:2F:12","Channel":6,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

Thanks for your help.
Regards
Eike

eikejenn, 2024-08-31:
I spend some time and have done some compilations on some commits.
release 13.4.0 and 13.4.0.1 is running fine.  release 13.4.0.2 has the failure.
So i checked some commits between.
```
2544cf030 Update esptool.py  (#20890)
```
based on this commit (esptool update) the vbus data is coming.
```
99f4774e1 core3 fixes (#20895)
```
The next commit (core3 fixes) is not running for me. All metrics are 0.0.
This commit changes the file [tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino].
But i cannot see, why my metrics all are 0. :-(
Regards
Eike

gemu2015, 2024-09-01:
there were changes according default string size such that string replacement fails with the newer version.
you simply must expand string size like
>D 24