arendst/Tasmota #22137: SonOff POWR320 Elite reset in loop after flash

jvilafe, 2024-09-13:
### PROBLEM DESCRIPTION
New SonOff POW Elite 20A. Working fine with original Itead firmware.
Flashed with Tasmota ok, but no tasmota Ap shows after flashing and reconnecting to 230v.
Erased flash with esptool, then, connected again USB_serial and flashed again with tasmota webflash utility (latest)
Flashed OK, disconnect from serial, connect to 230v and shows tasmota AP for a few seconds...then, AP disappears again and there is no option to show Ap again.
Logs:

[10:06:05]
[10:06:05]00:00:00.002 HDW: ESP32-D0WD-V3 v3.1 
[10:06:05]00:00:00.020 UFS: FlashFS mounted with 308 kB free
[10:06:05]00:00:00.032 CFG: Loaded from File, Count 2
[10:06:05]00:00:00.043 QPC: Count 1
[10:06:05]00:00:00.593 BRY: Berry initialized, RAM used 5060 bytes
[10:06:05]00:00:00.606 Project tasmota - Tasmota Version 14.2.0(release-tasmota32)-3_0_4(2024-08-14T12:39:45)
[10:06:06]00:00:00.010 WIF: WifiManager active for 3 minutes
[10:06:06][10:06:06]
[10:06:06]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:06:06]configsip: 0, SPIWP:0xee
[10:06:06]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:06:06]mode:DIO, clock div:2
[10:06:06]load:0x3fff0030,len:184
[10:06:06]load:0x40078000,len:13228
[10:06:06]ho 0 tail 12 room 4
[10:06:06]load:0x40080400,len:4
[10:06:06]load:0x40080404,len:2924
[10:06:06]entry 0x40080570

in loop...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): POWR320D  (Elite 20A)
- [x ] Tasmota binary firmware version number used: tried with 14.1, 12.5, and latest 14.2.0(release-tasmota32)-3_0_4(2024-08-14T12:39:45)
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool for erase flash. ESPflash and tasmota webflash for upload
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
-[10:06:05]
[10:06:05]00:00:00.002 HDW: ESP32-D0WD-V3 v3.1 
[10:06:05]00:00:00.020 UFS: FlashFS mounted with 308 kB free
[10:06:05]00:00:00.032 CFG: Loaded from File, Count 2
[10:06:05]00:00:00.043 QPC: Count 1
[10:06:05]00:00:00.593 BRY: Berry initialized, RAM used 5060 bytes
[10:06:05]00:00:00.606 Project tasmota - Tasmota Version 14.2.0(release-tasmota32)-3_0_4(2024-08-14T12:39:45)
[10:06:06]00:00:00.010 WIF: WifiManager active for 3 minutes
[10:06:06][10:06:06]
[10:06:06]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:06:06]configsip: 0, SPIWP:0xee
[10:06:06]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:06:06]mode:DIO, clock div:2
[10:06:06]load:0x3fff0030,len:184
[10:06:06]load:0x40078000,len:13228
[10:06:06]ho 0 tail 12 room 4
[10:06:06]load:0x40080400,len:4
[10:06:06]load:0x40080404,len:2924
[10:06:06]entry 0x40080570
[10:06:06]
[10:06:06]00:00:00.002 HDW: ESP32-D0WD-V3 v3.1 
[10:06:06]00:00:00.020 UFS: FlashFS mounted with 308 kB free
[10:06:06]00:00:00.032 CFG: Loaded from File, Count 2
[10:06:06]00:00:00.043 QPC: Count 2
[10:06:06]00:00:00.593 BRY: Berry initialized, RAM used 5060 bytes
[10:06:06]00:00:00.606 Project tasmota - Tasmota Version 14.2.0(release-tasmota32)-3_0_4(2024-08-14T12:39:45)
[10:06:07]00:00:00.010 WIF: WifiManager active for 3 minutes
[10:06:07][10:06:07]
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flashed with Tasmota ok, but no tasmota Ap shows after flashing and reconnecting to 230v.
Erased flash with esptool, then, connected again USB_serial and flashed again with tasmota webflash utility (latest)
Flashed OK, disconnect from serial, connect to 230v and shows tasmota AP for a few seconds...then, AP disappears again and there is no option to show Ap again.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-09-13:
The log looks very much like using a wimpy USB/serial adapter unable to power the ESP32 to boot to Wifi. What happens with mains power instead is hard to guess.....

jvilafe, 2024-09-13:
Hi

I only use USB/Serial to flash. Then, I connect the POW to 230V. First time boot after flash, POW shows AP tasmota-xxxx-xxxx for 20 or 30 seconds, I can connect to this WiFi but not to device....and Tasmota AP goes down and disappears...
It is only for first time boot up after flash...all other power cycle didn't shows the tasmota AP again.

sfromis, 2024-09-13:
You also used the USB/serial adapter to provide the logs you showed, thus no basis for guessing what happens when using mains instead.

jvilafe, 2024-09-13:
In your opinion, how can I take the logs when connected to 230V?