arendst/Tasmota #22149: Sonoff Only works on serial 14.2.0 But works normal on 12.2.0

Whininie, 2024-09-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

when connected to AC, Wifi on diffrent Sonoff devices not working on 14.2.0 but working on 11.2.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP8266
- [X] Tasmota binary firmware version number used: 14.2.0
  - [X] Pre-compiled
  - [] Self-compiled
- [X] Flashing tools used: Web Installer & Update Tool tasmota webpage
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

A good day to update all my sonoff devices, some came back online, some still offline, after hours or troubleshooting & Google i found out when i am on update 14.2.0 the wifi only works when connected to serial. if i use firmware 12.2.0 then they all working fine.
can only find old posts about it. is there a known issue of a fix?
**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-09-15:
Well-known problem with that bad old hardware, barely able to power itself, and sometimes better chances with an older firmware version. Or maybe `tasmota-lite.bin.gz`

Whininie, 2024-09-15:
> Well-known problem with that bad old hardware, barely able to power itself, and sometimes better chances with an older firmware version. Or maybe `tasmota-lite.bin.gz`

Yeah i did try the lite but no succes, what do you recommand buy new sonoffs or just the old firmware.

sfromis, 2024-09-15:
The general recommendation around upgrades is to not do it, unless you have a specific reason. "If it ain't broke, don't fix it". A higher number is not in itself a "good reason", maybe unless you have a bit of OCD :wink: 

As long as it works with older firmware, fine with that, but beware that the senescence may be  progressing, meaning that having spares available would make sense.

Whininie, 2024-09-15:
> "maybe unless you have a bit of OCD 😉

I have, never tested, but i have. so i will order new ones. because it are 3 of my 25 devices that are not working. and replacing is faster then downgrading all that stuff. and 3 with other firmware is a no go haha