arendst/Tasmota #22150: APDS9960 does not work with Tasmota

Basti149, 2024-09-16:
I want to connect an APDS 9960 to my Wemos mini.
For this I compiled Tasmota.bin with the following

#ifndef USE_APDS9960
#define USE_APDS9960 // [I2cDriver21] Enable APDS9960 Proximity Sensor (I2C address 0x39). Disables SHT and VEML6070 (+4k7 code)
#define USE_APDS9960_GESTURE // Enable APDS9960 Gesture feature (+2k code)
#define USE_APDS9960_PROXIMITY // Enable APDS9960 Proximity feature (>50 code)
#define USE_APDS9960_COLOR // Enable APDS9960 Color feature (+0.8k code)
#define USE_APDS9960_STARTMODE 0 // Default to enable Gesture mode
#endif

But still nothing is displayed in the console
This is only displayed here with i2cscan.

15:46:55.893 RSL: RESULT = {"I2CScan":"Device(s) found at 0x39"}

Do I have to enter anything else? I use Tasmota 14.x.x

sfromis, 2024-09-16:
Don't just open another thread, and "Issues" is not for asking questions. I did respond in the other thread, and what you post now is not a valid command, should be `Sensor27 1`, but only if the sensor is actually included in the binary you're running, and detecting the sensor.

Basti149, 2024-09-16:
This is a different sensor than the APD9960, I meant the APD9930 and this is not yet supported by Tasmota. That's why I want someone to integrate the sensor into Tasmota.

[Image]

That's why he couldn't detect the sensor

https://components101.com/sensors/apds-9930-digital-ambient-light-and-proximity-sensor