arendst/Tasmota #22206: Wemos SSD1306 128x64 0.96 not working with Universal Display Driver

ThomasDesloovere, 2024-09-27:
### PROBLEM DESCRIPTION
I have multiple types of oled display most of them have the SSD1306 driver. Since Tasmota came out with the Universal Display Driver they are a bit more of a pain in the ass to configure. I have a ESP running Tasmota 13.2.0 with SSD1306 display from Wemos/Lolin that is working great, i wanted to replicate the setup with a newer version of Tasmota with the same display (known good) I configured a new esp with Tasmota 14.2.0.6 Here the lolin display doesn't work, if i connect an off brand random display with the SSD1306 driver it works perfecly fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ *] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ *] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ *] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ *] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ *] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ *] Device used (e.g., Sonoff Basic): Lolin / Wemos ESP8266 D1 mini
- [ *] Tasmota binary firmware version number used: 14.2.0.6 
  - [ ] Pre-compiled
  - [ *] Self-compiled
- [ *] Flashing tools used: ESPWebTool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:59:42.583 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:59:42.791 RSL: RESULT = {"Module":{"0":"Generic"}}
19:59:43.064 RSL: RESULT = {"GPIO0":{"480":"Buzzer"},"GPIO1":{"0":"None"},"GPIO2":{"1376":"WS2812"},"GPIO3":{"5730":"Option A3"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"9248":"LD2410 Tx"},"GPIO14":{"1088":"IRrecv"},"GPIO15":{"9280":"LD2410 Rx"},"GPIO16":{"1056":"IRsend1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
20:00:14.286 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:00:14.495 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:00:14.697 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":168,"Free":343,"Rules":":H,SSD1306,128,64,1,I2C,3c,*,*,* :S,0,2,1,0,30,20 :I AE D5,80 A8,3F D3,00 40 8D,14 20,00 A1 C8 DA,12 81,9F D9,F1 DB,40 A4 A6 AF :o,AE :O,AF :A,00,10,40,00,00 :i,A6,A7 #"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:00:34.417 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_3E53A7","ButtonTopic":"0","Power":"11","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:00:34.422 RSL: STATUS1 = {"StatusPRM":{"Baudrate":254400,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:21:37","StartupUTC":"2024-09-27T17:38:57","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2024-09-27T14:55:59","SaveCount":61,"SaveAddress":"F9000"}}
20:00:34.428 RSL: STATUS2 = {"StatusFWR":{"Version":"14.2.0.6(tasmota)","BuildDateTime":"2024-09-27T14:00:42","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"375/699"}}
20:00:34.433 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi-2.4-7538",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","20006000","00004000","00000000"]}}
20:00:34.446 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":892,"Free":108,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8FBAC787","043682A1","000215CF","010013C0","D000F981","00004024","00001000","54000020","04000480","00000000"],"Drivers":"1,2,!3,4,5,!6,7,!8,9,10,12,13,!16,!18,!19,!20,!21,!22,24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6,7,9,10,14,63,102,111","I2CDriver":"3,5,7,8,10,11,15,43,47,84"}}
20:00:34.451 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-3E53A7-5031","IPAddress":"192.168.1.17","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"B4:E6:2D:3E:53:A7","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:00:34.454 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3E53A7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:00:34.460 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-09-27T18:00:34Z","Local":"2024-09-27T20:00:34","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"07:44","Sunset":"19:36"}}
20:00:34.465 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-09-27T20:00:34","LD2410":{"Distance":[0.0,0.0,0.0],"Energy":[0,0]}}}
20:00:34.471 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-09-27T20:00:34","Uptime":"0T00:21:37","UptimeSec":1297,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"POWER1":"ON","Dimmer":99,"Color":"FC72EC","HSBColor":"307,55,99","Channel":[99,45,93],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WiFi-2.4-7538","BSSId":"58:90:43:A3:75:3E","Channel":1,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Connect and configure a Lolin / Wemos 128*64 SSD1306

### EXPECTED BEHAVIOUR
A functional display.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-09-27:
Did you make sure to include [universal display](https://tasmota.github.io/docs/Universal-Display-Driver/) support in your build?
`#define USE_UNIVERSAL_DISPLAY`

And what did you do to include the display descriptor? Without a file system, that could be using the suggestion for doing it in rule set 3. I've seen it work with this:
```
Rule3 :H,SSD1306,128,64,1,I2C,3c,*,*,* :S,0,2,1,0,30,20 :I AE D5,80 A8,3F D3,00 40 8D,14 20,00 A1 C8 DA,12 81,9F D9,F1 DB,40 A4 A6 AF :o,AE :O,AF :A,00,10,40,00,00 :i,A6,A7 #
```

ThomasDesloovere, 2024-09-27:
Yes, as stated the version I compiled works on other ssd1306 displays, if connect the wemos display it stays blank. Same i2c address

sfromis, 2024-09-27:
Well, if you have verified that Tasmota is doing what it should, it would seem to be up to hardware-level differences, including connections and wiring. "Known good" is not a sufficient indicator of "everything should be ok".

ThomasDesloovere, 2024-09-27:
While that is true, there really can’t be any wiring problem. The display connects with a 4 wire cable pre-made that works on the other one, the connector on this esp is also known good as it functions with temperature sensors. If I scan the i2c bus I get the address back from the display. There seems to be a driver difference between 13.2.0 and the universal driver.

sfromis, 2024-09-27:
Yes, the universal display driver is a total rewrite of the old concept with separate display drivers per display type. But still, since one display works, and not another, that indicates some hardware-level difference, where details of that would be key to any fix for making it work for both.

ThomasDesloovere, 2024-09-27:
Yeah I do understand that.

Lolin display
https://www.wemos.cc/en/latest/d1_mini_shield/oled_0_96.html

Generic display that works
https://www.az-delivery.de/a/s/products/0-96zolldisplay

It's kind of wird because I haven't come across this issue with any other driver. I have used both perfectly fine with Arduino and a SSD1306 library

sfromis, 2024-09-27:
Maybe @gemu2015 has some insights or questions?

gemu2015, 2024-09-28:
according to the driver source code of your display these init cmds are send
try with this slightly modified descriptor 

```
  :H,SSD1306,128,64,1,I2C,3c,*,*,*
  :S,0,2,1,0,30,20
  :I
  AE
  D5,F0
  A8,3F
  D3,00
  40
  8D,14
  20,00
  A0
  C0
  DA,12
  81,CF
  D9,F1
  DB,40
  A4
  A6
  2E
  AF
  :o,AE
  :O,AF
  :A,00,10,40,00,00
  :i,A6,A7
  #
```

ThomasDesloovere, 2024-09-28:
Hi, thanks for looking in to it. I just tried it, the generic one still functions but the Wemos one still stays black.