arendst/Tasmota #22209: All device stopped connecting to MQTT broker

DonatoD, 2024-09-28:
### PROBLEM DESCRIPTION
I don't now why but all devices (shelly1 and 2) stopped working. Devices are installed in different places with different LAN, internet conenction, ecc. So it must happens something general. I presume after letsencrypt certificate automatic renew. But I don't understand what. it's just a while it happens and I've tryed everithing but nothing.

### REQUESTED INFORMATION
Configuration
- devices are conencted to mosquitto broker on a remote server
- devices have tasmota version 11.0.0
- mosquitto configuration does not have any TLS configuration
- secure conenction is handled by ngnix

A) LOG MESSAGES
1) Firtstly I had:
- tasmota's console
TLS connection error: 296
Connection failed to www.mysite.com:2222, rc -2. Retry in 120 sec
- ngnix's error
[info] 2093274#2093274: *330537 client xxxxxxxxxxx:16496 connected to 0.0.0.0:2222
[info] 2093274#2093274: *330537 SSL_do_handshake() failed (SSL: error:0A0000C1:SSL routin es::no shared cipher) while SSL handshaking, client: xxxxxxxxxxx, server: 0.0.0.0:2222

2) Googling, the problem could be the certificate type (I have some doubt since I didn't change any settings at all). So I forzed letsencrypt renew certificate passing from key_type = ECDSA to key_type = rsa. Now this is what I get
- tasmota's console
TLS connection error: 62
Connection failed to www.mysite.com:2222, rc -2. Retry in 120 sec
- ngnix's error
[info] 2159881#2159881: *116021 client xxxxxxxxxxx:16541 connected to 0.0.0.0:63101
[info] 2159881#2159881: *116021 peer closed connection in SSL handshake (104: Connection reset by peer) while SSL handshaking, client: xxxxxxxxxxx, server: 0.0.0.0:63101

B) SETTINGS
- mosquitto:
listener 1883
protocol mqtt
listener 1884
protocol websockets

- ngnix
log_format mqtt '$remote_addr [$time_local] $protocol $status $bytes_received '
                '$bytes_sent $upstream_addr';
upstream msqt_mqtt {
    server 127.0.0.1:1883;
}
server {
    listen 2222 ssl;
    proxy_pass msqt_mqtt;
    ssl_preread on;
	# managed by Certbot
    ssl_certificate /etc/letsencrypt/live/mysite.it/fullchain.pem;
    ssl_certificate_key /etc/letsencrypt/live/mysite.it/privkey.pem;
    ssl_dhparam /etc/letsencrypt/ssl-dhparams.pem;
    ssl_session_cache shared:mqtt_nginx_SSL:10m;
    ssl_session_timeout 1440m;
    ssl_session_tickets off;
    ssl_protocols TLSv1.2 TLSv1.3;
    ssl_prefer_server_ciphers off;
    ssl_ciphers 'HIGH:!aNULL:!MD5:ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES256-GCM-SHA384:ECDHE-RSA-AES256-GCM-SHA384:ECDHE-ECDSA-CHACHA20-POLY1305:ECDHE-RSA-CHACHA20-POLY1305:DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384';
    access_log /var/log/nginx/mqtt_access.log mqtt;
    error_log  /var/log/nginx/mqtt_error.log info;
}

### TO REPRODUCE
I don't know

### EXPECTED BEHAVIOUR
devices still get conneted without problems

barbudor, 2024-09-28:
Can you connect a computer MQTT client such as MQTT Explorer in the same conditions as a Tasmota ?

DonatoD, 2024-09-28:
Yes, they have never stopped working, I'm encountering this problem only on tasmota devices