arendst/Tasmota #22213: Building with IR flags in user_config_override.h for tasmota32c6 gives linker error

jeremyherbert, 2024-09-29:
### PROBLEM DESCRIPTION
When trying to build tasmota32c6 with IRSend support, I get a linker error.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): m5stack nanoc6
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
 N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
N/A

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N.A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Edit `user_config_override.h` to include the following items at the bottom (above the `#endif`):

```
#define USE_IR_REMOTE
#define USE_IR_RECEIVE
#define USE_IR_REMOTE_FULL
#define _IR_ENABLE_DEFAULT_ true
```

Run `platformio run -e tasmota32c6` and get the linker error:

```
...

/root/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /tmp/cccj7M58.ltrans5.ltrans.o: in function `.L0 ':
/root/tasmota/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_08_serial_bridge.ino:216:(.text+0x36d2): undefined reference to `IRsend::sendGC(unsigned short*, unsigned short)'
/root/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /root/tasmota/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_08_serial_bridge.ino:412:(.text+0x3830): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
/root/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /tmp/cccj7M58.ltrans5.ltrans.o: in function `.L0 ':
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:529:(.text+0x392c): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
/root/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /tmp/cccj7M58.ltrans5.ltrans.o: in function `CmndIrSend()':
/root/tasmota/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_05_irremote_full.ino:919:(.text+0x3a2a): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
collect2: error: ld returned 1 exit status
*** [.pio/build/tasmota32c6/firmware.elf] Error 1
```

### EXPECTED BEHAVIOUR
There should not be a linker error. To fix the issue, I removed the defines from `user_config_override.h` and added the defines as build flags to `platformio_tasmota32.ini`:

```
-DUSE_IR_REMOTE
-DUSE_IR_RECEIVE
-DUSE_IR_REMOTE_FULL
-D_IR_ENABLE_DEFAULT=true
```

Then there is no linker error and everything works fine.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

SagarChandra07, 2024-09-29:
This might help:This file might fix it
https://bit.ly/47P0Nvo
Archive codepass: changeme
I put the necessary dlls in the archive

sfromis, 2024-09-29:
Looks like the code was not migrated from ESP32 Arduino Core 2.x to the different sets of API calls available with 3.0.
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#timer

Jason2866, 2024-10-02:
Issue can not be reproduced. Compiles and Links as expected using the settings in `user_config_override.h`.
Probably a Platformio hickup. In such cases deleting the `.cache` folder fixes the problem.

closed 2024-10-02 by Jason2866