arendst/Tasmota #22214: Build with DALI support fails since v14.x

FeKn, 2024-09-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Building Tasmota32 v14.x (v14.0, v14.1, v14.2 or master) with DALI support enabled in 'user_config_override.h' fails to compile with the following error:
```
[...]
Compiling .pio/build/tasmota32/libfcf/WebServer/Parsing.cpp.o
Archiving .pio/build/tasmota32/lib18b/libAdafruit_seesaw_soilsensor_1.3.1.a
Indexing .pio/build/tasmota32/lib18b/libAdafruit_seesaw_soilsensor_1.3.1.a
Compiling .pio/build/tasmota32/libfcf/WebServer/WebServer.cpp.o
Compiling .pio/build/tasmota32/libfcf/WebServer/detail/mimetable.cpp.o
Compiling .pio/build/tasmota32/libd08/Update/HttpsOTAUpdate.cpp.o
Compiling .pio/build/tasmota32/libd08/Update/Updater.cpp.o
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino:81:38: warning: ignoring attribute 'section (".iram1.87")' because it conflicts with previous 'section (".iram1.5")' [-Wattributes]
   81 | void IRAM_ATTR DALI_Tick_Handler(void)
      |                                      ^
/home/User/Tasmota/tasmota/tasmota.ino:3938:16: note: previous declaration here
Compiling .pio/build/tasmota32/lib647/WiFi/AP.cpp.o
Archiving .pio/build/tasmota32/lib3e5/libFS.a
Compiling .pio/build/tasmota32/lib647/WiFi/STA.cpp.o
Indexing .pio/build/tasmota32/lib3e5/libFS.a
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino: In function 'void enableDaliRxInterrupt()':
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino:100:5: error: 'timerAlarmDisable' was not declared in this scope
  100 |     timerAlarmDisable(Dali->timer);
      |     ^~~~~~~~~~~~~~~~~
Compiling .pio/build/tasmota32/lib647/WiFi/WiFi.cpp.o
Compiling .pio/build/tasmota32/lib647/WiFi/WiFiAP.cpp.o
Compiling .pio/build/tasmota32/lib647/WiFi/WiFiGeneric.cpp.o
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino: In function 'void disableRxInterrupt()':
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino:110:5: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  110 |     timerAlarmEnable(Dali->timer);
      |     ^~~~~~~~~~~~~~~~
      |     timerAlarm
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino: In function 'void DaliPreInit()':
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino:386:30: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  386 |      Dali->timer = timerBegin(DALI_TIMER, 13, true);
      |                    ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:84,
                 from /home/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from /tmp/tmpvdkr5_5b:1:
/home/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino:387:25: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  387 |     timerAttachInterrupt(Dali->timer, &DALI_Tick_Handler, true);
      |     ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/tasmota32/lib647/WiFi/WiFiMulti.cpp.o
/home/User/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_89_esp32_dali.ino:388:5: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  388 |     timerAlarmWrite(Dali->timer, 641, true);
      |     ^~~~~~~~~~~~~~~
      |     timerWrite
Archiving .pio/build/tasmota32/lib08c/libNetwork.a
Indexing .pio/build/tasmota32/lib08c/libNetwork.a
Compiling .pio/build/tasmota32/lib647/WiFi/WiFiSTA.cpp.o
Archiving .pio/build/tasmota32/lib062/libHTTPClient.a
Indexing .pio/build/tasmota32/lib062/libHTTPClient.a
Compiling .pio/build/tasmota32/lib647/WiFi/WiFiScan.cpp.o
Archiving .pio/build/tasmota32/libd08/libUpdate.a
Indexing .pio/build/tasmota32/libd08/libUpdate.a
Compiling .pio/build/tasmota32/lib3f4/HTTPUpdate/HTTPUpdate.cpp.o
Archiving .pio/build/tasmota32/libfcf/libWebServer.a
Compiling .pio/build/tasmota32/liba50/ESPmDNS/ESPmDNS.cpp.o
Indexing .pio/build/tasmota32/libfcf/libWebServer.a
Compiling .pio/build/tasmota32/lib84e/ESP32-to-ESP8266-compat/esp8266toEsp32.cpp.o
*** [.pio/build/tasmota32/src/tasmota.ino.cpp.o] Error 1
Compiling .pio/build/tasmota32/libc28/SD/SD.cpp.o
======================================= [FAILED] Took 30.59 seconds =======================================

Environment    Status    Duration
-------------  --------  ------------
tasmota32      FAILED    00:00:30.595
================================== 1 failed, 0 succeeded in 00:00:30.595 ==================================

 *  The terminal process "platformio 'run', '--environment', 'tasmota32'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [-] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: v14.x (v14.0, v14.1, v14.2 or master)
  - [ ] Pre-compiled
  - [X] Self-compiled
- [-] Flashing tools used: _____
- [-] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [-] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [-] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Build Tasmota32 v14.x (v14.0, v14.1, v14.2 or master) with DALI support enabled in 'user_config_override.h' (see below) via e.g. "platformio run --environment tasmota32" with VSCode or Gitpod.

Additions in 'user_config_override.h':
```
#ifndef USE_DALI
#define USE_DALI
#endif

    #ifdef DALI_IN_INVERT
    #undef DALI_IN_INVERT
    #endif
    #define DALI_IN_INVERT  0

    #ifdef DALI_OUT_INVERT
    #undef DALI_OUT_INVERT
    #endif
    #define DALI_OUT_INVERT  0

    #ifdef DALI_TIMER
    #undef DALI_TIMER
    #endif
    #define DALI_TIMER 0  
```

### EXPECTED BEHAVIOUR
Successful build of Tasmota32 with DALI support (as in Tasmota32 v13.4 and earlier). 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Build with the same 'user_config_override.h' is successful up to Tasmota32 v13.4, the error occurs since v14.0.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-09-29:
Looks like the code was not migrated from ESP32 Arduino Core 2.x to the different sets of API calls available with 3.0.
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#timer

arendst, 2024-09-29:
I'll see if I can fix this...

arendst, 2024-09-29:
Try latest dev.

Not sure of it works as I don't know the initial timer frequency. I assume 160MHz but it may as well be a lot lower. Watch your console for possible error message ``DLI: Bad timer init``.

FeKn, 2024-09-29:
I have tried the change in commit 5d4230d:
Compiling with the 'user_config_override.h' mentioned above now works without any problems. 
Unfortunately, controlling the lighting via DALI no longer works (exactly the same configuration with v13.4 works).
No error messages in console. The only `DLI` message is:
```
DLI: Init - RX-pin: 3, TX-pin: 1
```

arendst, 2024-09-30:
Pls try again with latest dev

FeKn, 2024-10-01:
Thanks for the second fix!
Tested with 6fa38e4b: Works like a charm! :+1:  :-)

In my configuration I use a “DALI click” (https://www.mikroe.com/dali-click) with an ESP32. This is connected in parallel with a Busch-Jaeger DALI dimmer (https://www.busch-jaeger.de/en/online-catalogue/detail/2CKA006599A2986) on the DALI bus, which also acts as power supply.

The lights can be controlled with both DALI devices (= Tasmota and Busch-Jaeger) without any problems, but Tasmota does not recognize any changes made via the Busch-Jaeger dimmer, i.e. the information on the DALI bus is unfortunately not read at the moment...

arendst, 2024-10-02:
Thx for the feedback.

I'll dive into dali and see if I can Tasmota make recognize dali changes.

closed 2024-10-02 by arendst

FeKn, 2024-10-02:
> I'll dive into dali and see if I can Tasmota make recognize dali changes.

Sounds great! :+1:  
Just get in touch with me if there's something to test :-)

arendst, 2024-10-06:
Latest dev branch contains a total rewrite of the Dali driver. It now supports ESP8266 too. It also decodes Dali 1 messages it receives like:
```
21:36:45.790-021 MQT: tele/wemos14/RESULT = {"DALI":{"Power":"ON","Dimmer":173,"Address":254,"Command":173}}
```
where Power and Dimmer show the broadcast address values.

Give it a shot.

arendst, 2024-10-14:
@FeKn I updated DALI driver to be able to receive backward frames and show this to the user.

- Using a commands like ``DaliQuery 0xff,0x90`` or ``DaliQuery 255,144``
- Command ``DaliSend`` sends DALI commands tot he bus
- Command ``DaliWeb 1`` makes DALI broadcast address act like a Tasmota dimmable light on the GUI.

Pls report back if anything doesn't work as I envisionend.

Pls test if the driver still supports your environment.

FeKn, 2024-10-21:
> Latest dev branch contains a total rewrite of the Dali driver. It now supports ESP8266 too. It also decodes Dali 1 messages it receives like:
> 
> ```
> 21:36:45.790-021 MQT: tele/wemos14/RESULT = {"DALI":{"Power":"ON","Dimmer":173,"Address":254,"Command":173}}
> ```

@arendst Sorry for the late reply!
I tested commit `abc6e463`, which did output the DALI value set via Tasmota in `tele/<tasmota>/RESULT`, but it did not recognize to commands on the DALI bus.

Today I will take a look at the new DALI commands on the latest dev branch (`f6699b52`) and get back to you.

arendst, 2024-10-22:
I suggest to start using dev v14.3.0.2

A major change in DALI code. See changelogs

maltic42, 2024-10-22:
> Thanks for the second fix! Tested with [6fa38e4](https://github.com/arendst/Tasmota/commit/6fa38e4b83b66981065f94142d1fb02ba2fd5e08): Works like a charm! 👍 :-)
> 
> In my configuration I use a “DALI click” (https://www.mikroe.com/dali-click) with an ESP32. This is connected in parallel with a Busch-Jaeger DALI dimmer (https://www.busch-jaeger.de/en/online-catalogue/detail/2CKA006599A2986) on the DALI bus, which also acts as power supply.
> 
> The lights can be controlled with both DALI devices (= Tasmota and Busch-Jaeger) without any problems, but Tasmota does not recognize any changes made via the Busch-Jaeger dimmer, i.e. the information on the DALI bus is unfortunately not read at the moment...

@FeKn I am very interested in figuring out how to connect the DALI click to an ESP32 - can you give me some information about that? Thanks!

FeKn, 2024-10-27:
> @FeKn I am very interested in figuring out how to connect the DALI click to an ESP32 - can you give me some information about that? Thanks!

@maltic42 This is how my ESP32 is connected to the DALIclick:

Tasmota Module Configuration:
```
GPIO1: Dali TX
GPIO3: Dali RX
```

ESP32 -> DALIclick wiring:
```
ESP32 | DALIclick
------|----------
GPIO1 | TX
GPIO3 | INT
3V3   | 3.3V
GND   | GND
```

maltic42, 2024-10-27:
> @maltic42 This is how my ESP32 is connected to the DALIclick:
> 
> Tasmota Module Configuration:
> 
> ```
> GPIO1: Dali TX
> GPIO3: Dali RX
> ```
> 
> ESP32 -> DALIclick wiring:
> 
> ```
> ESP32 | DALIclick
> ------|----------
> GPIO1 | TX
> GPIO3 | INT
> 3V3   | 3.3V
> GND   | GND
> ```

@FeKn Thanks a lot - I did not expect this to be so easy. Is this supposed to be working with the DALI 2 click board as well when connecting to the appropriate pins?

arendst, 2024-10-27:
Find Tasmota DALI info here https://tasmota.github.io/docs/DALI/

FeKn, 2024-10-27:
> I suggest to start using dev v14.3.0.2

@arendst I've tried the latest dev and it looks very good! The commands from external devices on the DALI bus are now also recognized by Tasmota. 👍 :-)

But I have noticed the following behavior:
Commands from external devices and the use of the slider in Tasmota WebUI lead to the following RESULT:
`stat/tasmota_esp32/RESULT = {"POWER":"ON","Dimmer":50}`

MQTT commands (`cmnd/tasmota_esp32/DaliDimmer`), on the other hand, lead to the following RESULT:
`stat/tasmota_esp32/RESULT = {"DALI":{"Power":"ON","Dimmer":50,"Address":254,"Command":127}}`

Is there a reason why the RESULT is returned in different JSON paths?  This makes the evaluation more difficult...

Furthermore, the slider in the Tasmota WebUI does not show changes via MQTT commands, although the DALI lights respond to these commands.

FeKn, 2024-10-27:
> Thanks a lot - I did not expect this to be so easy. Is this supposed to be working with the DALI 2 click board as well when connecting to the appropriate pins?

@maltic42 I would suspect that DALI 2 click also works, but I have not tested it...

arendst, 2024-10-27:
@FeKn Thx for testing.

> Is there a reason why the RESULT is returned in different JSON paths? This makes the evaluation more difficult...

Yes there is. The difference is in handling it's functionality. With ``DaliLight 1`` the dali lights needs to be seen as Tasmota controlled light using it's light interface with commands like ``Power`` and ``Dimmer``. In that case do not use specific DALI commands like ``DaliPower`` and ``DaliDimmer``. There is no need to use these commands as their functionality is controlled by the Tasmota light driver which in turn can be controlled by KNX, Matter, Alexa whatever Tasmota supports. Regarding MQTT commands you'll expected to use commands like ``cmnd/tasmota_esp32/Dimmer``.

When Tasmota light control is disabled (``DaliLight 0``) only DALI commands are available to control the lights. In that case you'll see the second JSON response.

> the slider in the Tasmota WebUI does not show changes via MQTT commands

Tasmota lights were never able to show slider position in real time. The initial reason was that tests having dynamic sliders made user control difficult as the user would interfere with dynamically changes of the slider. Therefor I disabled that and so you'll see only static sliders used for user input only. So as designed.

maltic42, 2024-10-27:
> @maltic42 I would suspect that DALI 2 click also works, but I have not tested it...

OK, I will give it a try and order on of the devices next week. I have a complete setup here with about 15 DALI lights and about 20 switches.

maltic42, 2024-10-27:
> Find Tasmota DALI info here https://tasmota.github.io/docs/DALI/

@arendst Perfect, thanks!!

maltic42, 2024-10-27:
> > I suggest to start using dev v14.3.0.2

@arendst Unfortunately I cannot upgrade to v14.3.0.2 - v14.3.0 is no problem and seems to be working. After upgrading to v14.3.0.2 on two ESP32 boards the web interface is not reachable, the connection hangs, i.e. the device does not answer, the devices are pingable.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13168
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.001 HDW: ESP32-D0WDQ6 v1.0 
00:00:00.026 UFS: FlashFS mounted with 304 kB free
00:00:00.046 CFG: Loaded from File, Count 8
00:00:00.054 QPC: Count 2
00:00:00.095 BRY: Berry initialized, RAM used 3780 bytes
00:00:00.108 Project tasmota - Tasmota Version 14.3.0.2(227e5f2-tasmota32)-3_1_0(2024-10-27T10:04:58)
00:00:01.000 WIF: Connecting to AP1 TOAD Channel 1 BSSId 04:B4:FE:57:30:2E in mode HT40 as tasmota-302958-2392...
00:00:03.471 WIF: Connected
00:00:03.679 HTP: Web server active on tasmota-302958-2392 with IP address 192.168.188.36

maltic42, 2024-10-27:
2nd device:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13168
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.001 HDW: ESP32-D0WDQ6 v1.1 
00:00:00.005 UFS: FlashFS mounted with 304 kB free
00:00:00.009 CFG: Loaded from File, Count 10
00:00:00.018 QPC: Count 4
00:00:00.045 BRY: Berry initialized, RAM used 3780 bytes
00:00:00.053 Project tasmota - Tasmota Version 14.3.0.2(227e5f2-tasmota32)-3_1_0(2024-10-27T10:04:58)
00:00:01.000 WIF: Connecting to AP1 TOAD Channel 1 BSSId 04:B4:FE:57:30:2E in mode HT40 as tasmota-9A4B90-2960...
00:00:03.494 WIF: Connected
00:00:03.753 HTP: Web server active on tasmota-9A4B90-2960 with IP address 192.168.188.37

arendst, 2024-10-27:
Sorry. Cannot test. Mine works just fine.
```
00:00:00.002 HDW: ESP32-D0WDQ6 v1.0 
00:00:00.051 UFS: FlashFS mounted with 280 kB free
00:00:00.069 CFG: Loaded from File, Count 62
00:00:00.237 QPC: Count 1
00:00:00.243 I2C: Bus1 using GPIO23(SCL) and GPIO22(SDA)
00:00:00.244 I2C: Bus2 using GPIO18(SCL) and GPIO25(SDA)
00:00:00.252 MCP: Invalid template
00:00:00.256 ROT: Mode 1
00:00:00.302 CFG: No '*.autoconf' file found
00:00:00.306 BRY: Berry initialized, RAM used 3780 bytes
00:00:00.314 BRY: No 'preinit.be'
00:00:00.332 CFG: Energy loaded from file
00:00:00.333 NRG: Init driver 33
00:00:00.335 SRC: Restart
00:00:00.340 Project tasmota - esp32d Version 14.3.0.2(227e5f2-tasmota32)-3_1_0(2024-10-27T10:04:58)
00:00:00.343 DSB: Sensors found 0
00:00:00.355 SHT: Sensor did not ACK command
00:00:00.399 I2C: HTU21 found at 0x40
00:00:00.424 I2C: BME680 found at 0x77 (bus2)
00:00:00.478 BRY: No 'autoexec.be'
00:00:00.771 WIF: Attempting connection...
00:00:01.055 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode HT40 as esp32d...
00:00:01.132 WIF: IPv4 192.168.2.129, mask 255.255.255.0, gateway 192.168.2.254
00:00:03.636 WIF: IPv6 Local fe80::9af4:abff:fe6d:2db4
00:00:03.643 WIF: Connected
00:00:03.923 HTP: Web server active on esp32d with IP address 192.168.2.129
00:00:03.927 DGR: (Re)discovering members
00:00:03.930 DMX: listening to port 6454
00:00:03.931 SRC: Light
00:00:03.935 RSL: RESULT = {"POWER2":"ON"}
00:00:03.936 RSL: POWER2 = ON
00:00:03.960 RTC: UTC 2024-10-27T15:03:26Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
16:03:26.000 RTC: Synced by NTP
16:03:26.893 MQT: Attempting connection...
16:03:26.912 MQT: Connected
16:03:26.915 MQT: tele/esp32d/LWT = Online (retained)
16:03:26.917 MQT: cmnd/esp32d/POWER = 
16:03:26.918 MQT: Subscribe to cmnd/esp32d/#
16:03:26.919 MQT: Subscribe to cmnd/tasmotas/#
16:03:26.930 MQT: Subscribe to cmnd/DVES_6D2DB4_fb/#
16:03:26.934 MQT: tele/esp32d/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"14.3.0.2(227e5f2-tasmota32)","FallbackTopic":"cmnd/DVES_6D2DB4_fb/","GroupTopic":"cmnd/tasmotas/"}}
16:03:26.949 MQT: tele/esp32d/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"esp32d","IPAddress":"192.168.2.129","IP6Global":"","IP6Local":"fe80::9af4:abff:fe6d:2db4"}}
16:03:26.965 MQT: tele/esp32d/INFO3 = {"Info3":{"RestartReason":"Software reset digital core","BootCount":27}}
16:03:26.970 MQT: stat/esp32d/RESULT = {"POWER1":"OFF"}
16:03:26.982 MQT: stat/esp32d/POWER1 = OFF
16:03:26.983 MQT: stat/esp32d/RESULT = {"POWER2":"ON"}
16:03:26.985 MQT: stat/esp32d/POWER2 = ON
16:03:29.636 WIF: IPv6 Global fd84:7996:2cbf:1349:9af4:abff:fe6d:2db4
16:03:29.910 QPC: Reset
16:03:30.907 MQT: tele/esp32d/STATE = {"Time":"2024-10-27T16:03:30","Uptime":"0T00:00:08","UptimeSec":8,"Heap":134,"SleepMode":"Dynamic","Sleep":5,"LoadAvg":146,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":49},"POWER1":"OFF","POWER2":"ON","Dimmer":100,"Color":"FFBF00","HSBColor":"45,100,100","Channel":[100,75,0],"Scheme":2,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"HT40","RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:03"}}
16:03:30.966 MQT: tele/esp32d/SENSOR = {"Time":"2024-10-27T16:03:30","ENERGY":{"TotalStartTime":"2022-08-13T15:51:54","Total":[0.45020,0.32054],"Yesterday":[0.00920,0.00615],"Today":[0.00640,0.00428],"Period":[0.000,0.000],"Power":[0.273,0.201],"Voltage":1.42,"Current":[0.193,0.142]},"HTU21":{"Temperature":24.1,"Humidity":47.8,"DewPoint":12.3},"BME680":{"Temperature":25.5,"Humidity":52.4,"DewPoint":15.1,"Pressure":1021.6,"Gas":2335.64},"ArtNet":{"PacketsReceived":0,"PacketsAccepted":0,"Frames":0,"PacketsPerRow":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]},"Global":{"Temperature":24.1,"Humidity":47.8,"DewPoint":12.3,"Pressure":1021.6,"SeaPressure":1022.6},"PressureUnit":"hPa","TempUnit":"C"}
16:03:31.028 RUL: TELE-HTU21#TEMPERATURE performs 'dzsend 9988,24.1'
16:03:31.029 SRC: Rule
16:03:31.031 CMD: Grp 0, Cmd 'DZSEND', Idx 1, Len 9, Pld 9988, Data '9988,24.1'
16:03:31.048 MQT: domoticz/in = {"idx":9988,"nvalue":0,"svalue":"24.1","Battery":100,"RSSI":10}
16:03:31.051 MQT: stat/esp32d/RESULT = {"DzSend":"Done"}
16:03:31.899 APP: Boot Count 27
```

arendst, 2024-10-27:
I suggest you enable more logging (``SerialLog 4``) and see what goes on.

maltic42, 2024-10-27:
I activated the WifiManger (by mistake in the console), connected to the device on 192.168.4.1, added my Wifi settings again, after that it was working - on both devices. No idea, what happened.

I ordered a DALI 2 Click board an will get back with some feedback to you when it arrives.

maltic42, 2024-10-27:
@arendst According to the documentation I can send commands (on, off, light level) either to all devices (broadcast), an individual device or to a group as specified by the DaliTarget command. Would it be possible to support more than one device or group and having more than one Tasmota controlled light which then represents broadcast, individual gear or group?

arendst, 2024-10-27:
No, not at the moment. It would take a major task to support more than one light. The problem is how many to support, 64 sliders in the gui makes no sense. I guess you’ll have to find a way to integratie tasmota dalipower/dalidimmer in your home automation front end.

maltic42, 2024-10-27:
I see - ok. Thanks!

maltic42, 2024-10-29:
The DALI 2 Click is unfortunately not working. When I connect the device the bus, the bus hangs completely, i.e. it is not possible to use the existing switches anymore.

arendst, 2024-10-29:
I suggest you try to find DALI 2 specific hardware information from MikroE and see how it compares to the Dali Click hardware.

In addition I managed to get the slider(s) in the GUI reflecting changes in commands at ``WebRefresh`` intervals.

arendst, 2024-10-29:
Regarding the DALI 2 CLick the observed situation sounds familiar.

It might well be you need to invert the Tasmota GPIO's. Try ``DALI TX_i`` and ``DALI_RX_i`` like I needed to do for the Shelly DALI Dimmer Gen3.

maltic42, 2024-10-29:
Perfect - this seems to be working well!!! I will do some testing. 

Where should I report errors and with which log level? The first thing I observed is that the device reboots when settings DaliLight to 0.

maltic42, 2024-10-29:
> No, not at the moment. It would take a major task to support more than one light. The problem is how many to support, 64 sliders in the gui makes no sense. I guess you’ll have to find a way to integratie tasmota dalipower/dalidimmer in your home automation front end.

Matter to DALI works very well with some custom rules - at least for Homekit, that's really great!

20:34:04.043 CMD: Rule1
20:34:04.047 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":111,"Free":400,"Rules":"on mtrreceived#Light0#power==1 do DaliPower101 254 endon on mtrreceived#Light0#power==0 do DaliPower101 0 endon"}}
20:34:06.001 CMD: Rule2
20:34:06.005 RSL: RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":56,"Free":455,"Rules":"on mtrreceived#Light1#Bri do DaliPower102 %value% endon "}}