arendst/Tasmota #22225: Switch at Gpio3 (RX)

bananajoe86, 2024-09-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Connect a Reed-Contact at Sonoff Basic R3 at GPIO3 RX. I´m sure it work very well for few month and didn´t know what happend, that it didn´t work anymore. Using GPIO1 TX will work, so wiring is fine. Permanently using GPIO1 isn´t possible because at startup you got problems wenn Pin is high.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Basic R3 
- [x ] Tasmota binary firmware version number used: 14.2 (also trying downgrade to 9.3)
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: esp-homeflasher
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:31:44.259 MQT: stat/SonoffR3-TreppeLampiTor/RESULT = {"NAME":"Basic R3","GPIO":[32,1,0,1,0,0,1,1,224,320,0,0,1,0],"FLAG":0,"BASE":1}
18:31:44.484 MQT: stat/SonoffR3-TreppeLampiTor/RESULT = {"Module":{"0":"Basic R3"}}
18:31:44.690 MQT: stat/SonoffR3-TreppeLampiTor/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"162":"Switch3"},"GPIO2":{"0":"None"},"GPIO3":{"225":"Relay2"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"226":"Relay3"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:32:14.448 MQT: stat/SonoffR3-TreppeLampiTor/STATUS = {"Status":{"Module":0,"DeviceName":"SonoffR3-TreppeLampiTor","FriendlyName":["SonoffR3-LampiTreppeTor","",""],"Topic":"SonoffR3-TreppeLampiTor","ButtonTopic":"0","Power":"100","PowerLock":"000","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:32:14.469 MQT: stat/SonoffR3-TreppeLampiTor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T01:32:23","StartupUTC":"2024-09-30T14:59:51","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2020-09-29T23:21:33","SaveCount":253,"SaveAddress":"F7000"}}
18:32:14.487 MQT: stat/SonoffR3-TreppeLampiTor/STATUS2 = {"StatusFWR":{"Version":"14.2.0(release-tasmota)","BuildDateTime":"2024.08.14 12:41:05","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"452/699"}}
18:32:14.504 MQT: stat/SonoffR3-TreppeLampiTor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!7390",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
18:32:14.530 MQT: stat/SonoffR3-TreppeLampiTor/STATUS4 = {"StatusMEM":{"ProgramSize":649,"Free":352,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
18:32:14.553 MQT: stat/SonoffR3-TreppeLampiTor/STATUS5 = {"StatusNET":{"Hostname":"SonoffR3-Gartentreppe","IPAddress":"192.168.178.84","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"DC:4F:22:AB:74:34","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:32:14.571 MQT: stat/SonoffR3-TreppeLampiTor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.103","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AB7434","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:32:14.591 MQT: stat/SonoffR3-TreppeLampiTor/STATUS7 = {"StatusTIM":{"UTC":"2024-09-30T16:32:14Z","Local":"2024-09-30T18:32:14","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"07:21","Sunset":"19:02"}}
18:32:14.612 MQT: stat/SonoffR3-TreppeLampiTor/STATUS10 = {"StatusSNS":{"Time":"2024-09-30T18:32:14","Switch3":"ON"}}
18:32:14.633 MQT: stat/SonoffR3-TreppeLampiTor/STATUS11 = {"StatusSTS":{"Time":"2024-09-30T18:32:14","Uptime":"0T01:32:23","UptimeSec":5543,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"FRITZ!7390","BSSId":"E0:28:6D:D8:C0:A5","Channel":11,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-09-30:
If it suddenly no longer works, you may have fried the hardware. The "antenna" (wire from the reed) *might* pick up electrical noise, including static electricity, or whatever, blasting the gpio input signal handling.

Your config looks odd, 3 Relay pins configured, and your `Switch` is badly configured, you should always start numbers with 1; you cannot control what Relay a Switch acts on.

More generally, you skipped an important step, as you did not provide a log with details. Those questions are asked for good reasons.

bananajoe86, 2024-10-02:
I would rule out a hardware defect, I took an identical R3 to recreate the configuration there... same behavior. I am working on the test setup with a direct connection from gpio to ground. Furthermore, TX works well, RX does not respond.

Where do you find 3 relay pins in the configuration? I used the Blackadder [tamplate ](https://templates.blakadder.com/sonoff_basic_R3.html)and entered the 2 additional switches on Relay1 in the device settings.
[image]

I also added the "output". After restart it´s empty, because it did´t recognise the switch at RX.

sfromis, 2024-10-02:
From your post:
```
{"GPIO0":{"32":"Button1"},"GPIO1":{"162":"Switch3"},"GPIO2":{"0":"None"},"GPIO3":{"225":"Relay2"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"226":"Relay3"},"GPIO17":{"0":"None"}}
```
Or nicely formatted:
```
GPIO0   32   Button1
GPIO1   162  Switch3
GPIO2   0    None
GPIO3   225  Relay2
GPIO4   0    None
GPIO5   0    None
GPIO9   0    None
GPIO10  0    None
GPIO12  224  Relay1
GPIO13  320  Led_i1
GPIO14  0    None
GPIO15  0    None
GPIO16  226  Relay3
GPIO17  0    None
```
Odd.

Anyway, Sonoff Basic devices are **really** not good for adding stuff to, unless you like "a challenge" to make it work reliably.