arendst/Tasmota #22228: It seems that issue https://github.com/arendst/Tasmota/issues/16041 is back in the code.

scoace, 2024-10-02:
### PROBLEM DESCRIPTION
It seems that issue https://github.com/arendst/Tasmota/issues/16041 is back in the code.

Output from curl http://192.168.178.25/cm?cmnd=STATUS+8
{
  "Time": "2024-10-02T16:00:11",
  "LK13BE": {
    "Power_total_in": 3548.2,
    "Power_total_out": 77.3,
    "Power_curr": -294,
    "Power_L1_curr": -375,
    "Power_L2_curr": 1,
    "Power_L3_curr": 79,
    "Volt_L1_curr": 241.1,
    "Volt_L2_curr": 240.1,
    "Volt_L3_curr": 240.0,
    "Amperage_L1_curr": 1.62,
    "Amperage_L2_curr": 0.12,
    "Amperage_L3_curr": 0.42,
    "HZ": 50.00,
    "phase_angle_p1": 163.2,
    "phase_angle_p2": 273.3,
    "phase_angle_p3": 32.9,
    "phase_angle_l2_l1": 121.8,
    "phase_angle_l3_l1": 240.7
  }
}
output from curl http://192.168.178.25/metrics at the same time

# TYPE tasmota_info gauge
tasmota_info{version="14.2.0.6",image="(TasmoCompiler-esp32c3)",build_timestamp="2024.10.02 09:04:37",devicename="Z_BK",friendlyname="Z_BK"} 1
# TYPE tasmota_uptime_seconds gauge
tasmota_uptime_seconds{} 31846
# TYPE tasmota_boot_count counter
tasmota_boot_count{} 25
# TYPE tasmota_flash_writes_total counter
tasmota_flash_writes_total{} 6511
# TYPE tasmota_wifi_station_info gauge
tasmota_wifi_station_info{bssid="0C:72:74:A9:36:9E",ssid="BK"} 1
# TYPE tasmota_wifi_station_signal_dbm gauge
tasmota_wifi_station_signal_dbm{mac_address="0C:72:74:A9:36:9E"} -81
# TYPE tasmota_memory_free_bytes gauge
tasmota_memory_free_bytes{memory="heap"} 10480
# TYPE tasmota_sensors_power_total_in_ gauge
tasmota_sensors_power_total_in_{sensor="lk13be"} 3548.2
# TYPE tasmota_sensors_power_total_out_ gauge
tasmota_sensors_power_total_out_{sensor="lk13be"} 77.3
# TYPE tasmota_sensors_power_l2_curr_ gauge
tasmota_sensors_power_l2_curr_{sensor="lk13be"} 1
# TYPE tasmota_sensors_power_l3_curr_ gauge
tasmota_sensors_power_l3_curr_{sensor="lk13be"} 79
# TYPE tasmota_sensors_volt_l1_curr_ gauge
tasmota_sensors_volt_l1_curr_{sensor="lk13be"} 240.9
# TYPE tasmota_sensors_volt_l2_curr_ gauge
tasmota_sensors_volt_l2_curr_{sensor="lk13be"} 240.1
# TYPE tasmota_sensors_volt_l3_curr_ gauge
tasmota_sensors_volt_l3_curr_{sensor="lk13be"} 239.9
# TYPE tasmota_sensors_amperage_l1_curr_ gauge
tasmota_sensors_amperage_l1_curr_{sensor="lk13be"} 1.62
# TYPE tasmota_sensors_amperage_l2_curr_ gauge
tasmota_sensors_amperage_l2_curr_{sensor="lk13be"} 0.12
# TYPE tasmota_sensors_amperage_l3_curr_ gauge
tasmota_sensors_amperage_l3_curr_{sensor="lk13be"} 0.42
# TYPE tasmota_sensors_hz_ gauge
tasmota_sensors_hz_{sensor="lk13be"} 50.10
# TYPE tasmota_sensors_phase_angle_p1_ gauge
tasmota_sensors_phase_angle_p1_{sensor="lk13be"} 163.2
# TYPE tasmota_sensors_phase_angle_p2_ gauge
tasmota_sensors_phase_angle_p2_{sensor="lk13be"} 273.5
# TYPE tasmota_sensors_phase_angle_p3_ gauge
tasmota_sensors_phase_angle_p3_{sensor="lk13be"} 32.7
# TYPE tasmota_sensors_phase_angle_l2_l1_ gauge
tasmota_sensors_phase_angle_l2_l1_{sensor="lk13be"} 121.6
# TYPE tasmota_sensors_phase_angle_l3_l1_ gauge
tasmota_sensors_phase_angle_l3_l1_{sensor="lk13be"} 240.6

the value Power_curr": -294  and Power_L1_curr": -375 is not in the metrics!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:14:43.033 CMD: Backlog Template; Module; GPIO 255
18:14:43.122 MQT: stat/Z_BK/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:14:43.360 MQT: stat/Z_BK/RESULT = {"Module":{"0":"Generic"}}
18:14:43.566 MQT: stat/Z_BK/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:15:20.628 CMD: Status 0
18:15:20.637 MQT: stat/Z_BK/STATUS = {"Status":{"Module":0,"DeviceName":"Z_BK","FriendlyName":["Z_BK"],"Topic":"Z_BK","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:15:20.642 MQT: stat/Z_BK/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-4M.bin.gz","RestartReason":"Software/System restart","Uptime":"0T09:00:06","StartupUTC":"2024-10-02T07:15:14","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2023-11-05T10:53:03","SaveCount":6511,"SaveAddress":"FA000"}}
18:15:20.646 MQT: stat/Z_BK/STATUS2 = {"StatusFWR":{"Version":"14.2.0.6(TasmoCompiler-esp32c3)","BuildDateTime":"2024.10.02 09:04:37","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"363/699"}}
18:15:20.652 MQT: stat/Z_BK/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT","BK"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:15:20.663 MQT: stat/Z_BK/STATUS4 = {"StatusMEM":{"ProgramSize":662,"Free":1384,"Heap":11,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","0740458A","042E0001","00000003","00000000","C000E888","00100084","00201000","54000000","00000080","00000000"],"Drivers":"1,2,!4,!6,9,10,12,!18,!19,!22,26,29,!30,!35,!37,38,!45,50,62,!68","Sensors":"1,2,53,75"}}
18:15:20.670 MQT: stat/Z_BK/STATUS5 = {"StatusNET":{"Hostname":"Z-BK","IPAddress":"192.168.178.25","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"08:3A:8D:D2:1B:69","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:15:20.675 MQT: stat/Z_BK/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.173.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D21B69","MqttUser":"iot","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

 na
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

If a negative value occurs, it should be in the metrics!

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-10-02:
@arendst You removed my PR #16044 with your commit https://github.com/arendst/Tasmota/commit/c1ea8953cb3f496a5dbf20033a665ad3ad26f136

arendst, 2024-10-02:
Well if it takes two years to become noticable that's not too bad ;-)

Pls resubmit PR.

barbudor, 2024-10-02:
Done

PR #22232 (2024-10-03): fix removal of negative values (recreate #16044)
## Description:

**Related issue (if applicable):** fixes #22228 by recreating #16044

Fix removal of values starting with a non digit, such as starting with '-' for negative values or '.' for floating with leading 0

Originaly tested in #16044.
Not re-tested, just tested for comiplation of tasmota and tasmota32

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] The code c [... cut]
closed 2024-10-03 by arendst