arendst/Tasmota #22234: Tasmota Platform espressif32 2024.09.30 - Arduino Core 3.1.0.240926 break Etehernet support

ZioFabry, 2024-10-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I'm using an Olimex ESP32-PoE, starting from commit id 1248bd138c7eda34fe91d9f60721a78210a27b56 the ethernet is no more visible, anyone else experience that?.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

15:20:00.812 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
15:20:00.864 SRC: Backlog
15:20:00.866 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
15:20:00.873 MQT: pdu/SalaKong/J260/stat/RESULT = {"NAME":"Olimex ESP32-PoE-BioPDU","GPIO":[1,10209,10210,1,10144,1,0,0,5536,640,1,1,608,0,5600,1,1,1,1,5568,1,1,1,1,6210,0,0,0,1,10208,1,1,10176,1,1,1],"FLAG":0,"BASE":1}
15:20:01.205 SRC: Backlog
15:20:01.207 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
15:20:01.216 MQT: pdu/SalaKong/J260/stat/RESULT = {"Module":{"0":"Olimex ESP32-PoE-BioPDU"}}
15:20:01.409 SRC: Backlog
15:20:01.411 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
15:20:01.427 MQT: pdu/SalaKong/J260/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"10209":"BioPDU Bit2"},"GPIO2":{"10210":"BioPDU Bit3"},"GPIO3":{"0":"None"},"GPIO4":{"10144":"BioPDU PZEM0XX Tx"},"GPIO5":{"0":"None"},"GPIO6":{"6210":"Option A3"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"5536":"ETH POWER"},"GPIO13":{"640":"I2C SDA"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"608":"I2C SCL"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"10208":"BioPDU Bit1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"10176":"BioPDU PZEM016 Rx"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:18:24.750 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
15:18:24.756 MQT: pdu/SalaKong/J260/stat/STATUS = {"Status":{"Module":0,"DeviceName":"J260","FriendlyName":["J260","BioPDU2","BioPDU3","BioPDU4","","",""],"Topic":"J260","ButtonTopic":"0","Power":"0000001","PowerLock":"0000001","PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:18:24.804 MQT: pdu/SalaKong/J260/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://cc.biomine.it/ota/tasmota32-biopdu.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:40","StartupUTC":"2024-10-03T13:17:44","Sleep":50,"CfgHolder":4623,"BootCount":19,"BCResetTime":"2024-09-25T15:06:17","SaveCount":79}}
15:18:24.843 MQT: pdu/SalaKong/J260/stat/STATUS2 = {"StatusFWR":{"Version":"14.2.0.6(biopdu)","BuildDateTime":"2024-10-03T15:14:04","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"417/699"}}
15:18:24.860 MQT: pdu/SalaKong/J260/stat/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BioSalaKong",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["04208209","2805C80001000600003C5A0A192800000000","00008080","00006000","0000C000","00000011"]}}
15:18:24.899 MQT: pdu/SalaKong/J260/stat/STATUS4 = {"StatusMEM":{"ProgramSize":2597,"Free":282,"Heap":91,"StackLowMark":19,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9FBAD7DF","001482A1","2143B7CF","054013F5","E0320981","400040F0","00200000","5480482D","81030091","00000004"],"Drivers":"1,2,3,!4,!5,7,!8,9,10,11,12,13,!16,!20,!21,!24,26,!27,29,!34,!35,38,50,52,54,!59,62,!63,!67,!68,82,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,19,21,26,29,31,42,45,55,58,59,66,67,85,92,98,109,127","I2CDriver":"3,5,7,8,9,10,11,12,13,14,15,17,18,20,22,24,29,31,36,41,42,44,46,47,48,58,62,65,69,77,82"}}
15:18:24.961 MQT: pdu/SalaKong/J260/stat/STATUS5 = {"StatusNET":{"Hostname":"J260-7260","IPAddress":"10.1.204.126","Gateway":"10.1.204.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.204.1","DNSServer2":"8.8.8.8","Mac":"40:22:D8:15:1C:5C","IP6Global":"","IP6Local":"fe80::4222:d8ff:fe15:1c5c%st1","Ethernet":{"Hostname":"J260-7260-eth","IPAddress":"0.0.0.0","Gateway":"10.1.204.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.204.1","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":16.0}}
15:18:25.010 MQT: pdu/SalaKong/J260/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"PDU_%12X","MqttClient":"PDU_4022D8151C5C","MqttUser":"BIOPDU_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:18:25.040 MQT: pdu/SalaKong/J260/stat/STATUS7 = {"StatusTIM":{"UTC":"2024-10-03T13:18:25Z","Local":"2024-10-03T15:18:25","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"07:26","Sunset":"19:00"}}
15:18:25.064 MQT: pdu/SalaKong/J260/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0],"PowerLow":[0,0,0,0,0,0],"PowerHigh":[0,0,0,0,0,0],"VoltageLow":[0,0,0,0,0,0],"VoltageHigh":[0,0,0,0,0,0],"CurrentLow":[0,0,0,0,0,0],"CurrentHigh":[0,0,0,0,0,0],"MaxPower":[0,0,0,0,0,0],"MaxPowerHold":[10,10,10,10,10,10],"MaxPowerWindow":[30,30,30,30,30,30],"MaxEnergy":[0,0,0,0,0,0],"MaxEnergyStart":[0,0,0,0,0,0]}}
15:18:25.112 MQT: pdu/SalaKong/J260/stat/STATUS10 = {"StatusSNS":{"Time":"2024-10-03T15:18:25","ENERGY":{"TotalStartTime":"2024-09-25T15:06:17","Total":[0.000,0.000,0.000,0.000,0.000,0.000],"Yesterday":[0.000,0.000,0.000,0.000,0.000,0.000],"Today":[0.000,0.000,0.000,0.000,0.000,0.000],"Power":[0,0,0,0,0,0],"ApparentPower":[0,0,0,0,0,0],"ReactivePower":[0,0,0,0,0,0],"Factor":[0.00,0.00,0.00,0.00,0.00,0.00],"Voltage":[0,0,0,0,0,0],"Current":[0.000,0.000,0.000,0.000,0.000,0.000]},"MCP230XX":{"D0":0,"D1":0,"D2":0,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"OUT":{"OUT_D1":"OFF","OUT_D2":"OFF","OUT_D3":"OFF","OUT_D4":"OFF","OUT_D5":"OFF","OUT_D6":"OFF"}},"ESP32":{"Temperature":46.7},"TempUnit":"C"}}
15:18:25.180 MQT: pdu/SalaKong/J260/stat/STATUS11 = {"StatusSTS":{"Time":"2024-10-03T15:18:25","Uptime":"0T00:00:41","UptimeSec":41,"Heap":89,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":13,"Objects":197},"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","Wifi":{"AP":1,"SSId":"BioSalaKong","BSSId":"2E:C8:1B:F4:45:A2","Channel":1,"Mode":"HT40","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-10-04:
Tested with an https://templates.blakadder.com/wireless_tag_WT32-ETH01.html module works via LAN (phy LAN8720)

Jason2866, 2024-10-04:
Tested your firmware from `OtaUrl":"https://cc.biomine.it/ota/tasmota32-biopdu.bin`
Does work too. Closing since problem is not software related.

closed 2024-10-04 by Jason2866

ZioFabry, 2024-10-04:
My firmware on OTA is on  14.2.0.5... this is compiled at the previous commit...

Please test this one... https://cc.biomine.it/ota/tasmota32-biopdu.v1.4.0.6.bin

Jason2866, 2024-10-04:
works fine too
```python
00:00:00.002 HDW: ESP32-D0WD v1.0 
00:00:00.016 UFS: FlashFS mounted with 304 kB free
00:00:00.038 CFG: Loaded from LKG File, Count 78
00:00:00.263 BRY: Berry initialized, RAM used 5217 bytes
00:00:00.274 Project biopdu - Tasmota Version 14.2.0.6(biopdu)-3_1_0(2024-10-03T15:14:04)
00:00:01.690 ETH: Connected at 100Mbps Full Duplex, Mac C4:DE:E2:DA:0F:97, Hostname sonoff-DA0F94-3988-eth
00:00:03.279 HTP: Web server active on sonoff-DA0F94-3988-eth with IP address 192.168.2.161
21:47:13.266 MQT: Attempting connection...
21:47:13.283 MQT: Connected
21:47:13.286 MQT: tele/sonoff-DA0F94/LWT = Online (retained)
21:47:13.287 MQT: cmnd/sonoff-DA0F94/POWER = 
21:47:13.293 MQT: tele/sonoff-DA0F94/INFO1 = {"Info1":{"Module":"WT32_ETH01","Version":"14.2.0.6(biopdu)","FallbackTopic":"cmnd/sonoff-DA0F94_fb/","GroupTopic":"cmnd/sonoffs/"}}
21:47:13.307 MQT: tele/sonoff-DA0F94/INFO2 = {"Info2":{"WebServerMode":"Admin","Ethernet":{"Hostname":"sonoff-DA0F94-3988-eth","IPAddress":"192.168.2.161"}}}
21:47:13.321 MQT: tele/sonoff-DA0F94/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":41}}
21:47:16.001 WIF: Connecting to AP1 Jason_Home_WLAN Channel 11 BSSId 00:A0:57:2A:BD:19 in mode HT40 as sonoff-DA0F94-3988...
21:47:17.778 MQT: tele/sonoff-DA0F94/STATE = {"Time":"2024-10-04T21:47:17","Uptime":"0T00:00:07","UptimeSec":7,"Heap":116,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":5,"Objects":56}}
21:47:18.384 WIF: Connected
21:48:02.396 CMD: Status 0
21:48:02.400 MQT: stat/sonoff-DA0F94/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"sonoff-DA0F94","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:48:02.435 MQT: stat/sonoff-DA0F94/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"","RestartReason":"Software reset CPU","Uptime":"0T00:00:52","StartupUTC":"2024-10-04T19:47:10","Sleep":10,"CfgHolder":4617,"BootCount":41,"BCResetTime":"2023-09-01T14:52:09","SaveCount":80}}
21:48:02.470 MQT: stat/sonoff-DA0F94/STATUS2 = {"StatusFWR":{"Version":"14.2.0.6(biopdu)","BuildDateTime":"2024-10-03T15:14:04","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":160,"Hardware":"ESP32-D0WD v1.0","CR":"367/699"}}
21:48:02.484 MQT: stat/sonoff-DA0F94/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN","Jason_Home_WLAN"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600003C5A0A192800000000","000080C0","08006000","00006000","00000000"]}}
21:48:02.536 MQT: stat/sonoff-DA0F94/STATUS4 = {"StatusMEM":{"ProgramSize":2597,"Free":282,"Heap":110,"StackLowMark":19,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9FBAD7DF","001482A1","2143B7CF","054013F5","E0320981","400040F0","00200000","5480482D","81030091","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,13,!16,!20,!21,!24,26,!27,29,!34,!35,38,50,52,54,!59,62,!63,!67,!68,82,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,19,21,26,29,31,42,45,55,58,59,66,67,85,92,98,109,127","I2CDriver":"3,5,7,8,9,10,11,12,13,14,15,17,18,20,22,24,29,31,36,41,42,44,46,47,48,58,62,65,69,77,82"}}
21:48:02.594 MQT: stat/sonoff-DA0F94/STATUS5 = {"StatusNET":{"Hostname":"sonoff-DA0F94-3988","IPAddress":"192.168.2.163","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C4:DE:E2:DA:0F:94","IP6Global":"fd4d:d9a8:7683:0:c6de:e2ff:feda:f94","IP6Local":"fe80::c6de:e2ff:feda:f94%st3","Ethernet":{"Hostname":"sonoff-DA0F94-3988-eth","IPAddress":"192.168.2.161","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C4:DE:E2:DA:0F:97","IP6Global":"fd4d:d9a8:7683:0:c6de:e2ff:feda:f97","IP6Local":"fe80::c6de:e2ff:feda:f97%en1"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:48:02.659 MQT: stat/sonoff-DA0F94/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lan","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-DA0F94","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:48:02.684 MQT: stat/sonoff-DA0F94/STATUS7 = {"StatusTIM":{"UTC":"2024-10-04T19:48:02Z","Local":"2024-10-04T21:48:02","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"07:17","Sunset":"18:44"}}
21:48:02.701 MQT: stat/sonoff-DA0F94/STATUS10 = {"StatusSNS":{"Time":"2024-10-04T21:48:02"}}
21:48:02.718 MQT: stat/sonoff-DA0F94/STATUS11 = {"StatusSTS":{"Time":"2024-10-04T21:48:02","Uptime":"0T00:00:52","UptimeSec":52,"Heap":108,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":5,"Objects":56},"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":11,"Mode":"HT40","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```

ZioFabry, 2024-10-04:
can be a Olimex ESP32-PoE related issue ?

Jason2866, 2024-10-04:
What else? The used PHY chip is the same LAN8720. The LAN8720 is known to be problematic to init / reset when starting. Iirc Olimex had (has?) problems in there hardware designs here.

ZioFabry, 2024-10-05:
any suggestion ?

Jason2866, 2024-10-05:
Maybe Olimex support can help?