arendst/Tasmota #22235: tasmota32-display not working anymore with SH1106 128x64 oled

SHA-NDY, 2024-10-03:
### PROBLEM DESCRIPTION
I upgraded tasmota32-display from v13.4.0 to v14.2.0. The SH1106 display now doesn't show a reaction anymore.
It works fine with v13.4.0.

Configuration:

- DisplayModel 7
- DisplayAddress 60

i2cscan finds a device at address 0x3c which is expected to be my display.

In v13.4.0 the display model is shown as found in the console output.
The line with the found display model is missing in v14.2.0.

Built with Gitpod from latest development branch (v14.2.0.6) still has this issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 14.2.0.6
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA update
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

20:21:16.824 CMD: Backlog Template; Module; GPIO 255
20:21:16.825 SRC: WebConsole from 192.168.188.106
20:21:16.827 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
20:21:16.848 SRC: Backlog
20:21:16.850 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
20:21:16.858 MQT: stat/tasmota_015160/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1312,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
20:21:17.077 SRC: Backlog
20:21:17.078 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
20:21:17.084 MQT: stat/tasmota_015160/RESULT = {"Module":{"1":"ESP32-DevKit"}}
20:21:17.327 SRC: Backlog
20:21:17.328 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
20:21:17.342 MQT: stat/tasmota_015160/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
20:21:26.432 WIF: Checking connection...
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

20:18:45.750 MQT: stat/tasmota_015160/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_015160","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:18:45.757 MQT: stat/tasmota_015160/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release-13.4.0/tasmota32-display.bin","RestartReason":"Software reset CPU","Uptime":"0T00:14:58","StartupUTC":"2024-10-03T19:03:47","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2022-01-22T12:29:54","SaveCount":100}}
20:18:45.763 MQT: stat/tasmota_015160/STATUS2 = {"StatusFWR":{"Version":"14.2.0.6(display)","BuildDateTime":"2024-10-03T19:00:31","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"452/699"}}
20:18:45.769 MQT: stat/tasmota_015160/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndyCristiWLAN_Priv",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
20:18:45.781 MQT: stat/tasmota_015160/STATUS4 = {"StatusMEM":{"ProgramSize":1375,"Free":1856,"Heap":155,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","0FAA85CE","000482A1","800000C3","00000000","00000080","000000A0","40200000","54000008","00000080","00000004"],"Drivers":"1,2,!4,!5,!8,9,10,12,13,!16,38,50,52,55,62,!66,!68,!121","Sensors":"1,2,5,6,127","I2CDriver":"3,5,47"}}
20:18:45.789 MQT: stat/tasmota_015160/STATUS5 = {"StatusNET":{"Hostname":"tasmota-015160-4448","IPAddress":"192.168.188.68","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.188.23","DNSServer2":"0.0.0.0","Mac":"FC:F5:C4:01:51:60","IP6Global":"2001:9e8:afb6:7e00:fef5:c4ff:fe01:5160","IP6Local":"fe80::fef5:c4ff:fe01:5160%st1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:18:45.800 MQT: stat/tasmota_015160/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.23","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_015160","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:18:45.809 MQT: stat/tasmota_015160/STATUS7 = {"StatusTIM":{"UTC":"2024-10-03T19:18:45Z","Local":"2024-10-03T20:18:45","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:53","Sunset":"18:24"}}
20:18:45.819 MQT: stat/tasmota_015160/STATUS10 = {"StatusSNS":{"Time":"2024-10-03T20:18:45"}}
20:18:45.829 MQT: stat/tasmota_015160/STATUS11 = {"StatusSTS":{"Time":"2024-10-03T20:18:45","Uptime":"0T00:14:58","UptimeSec":898,"Heap":153,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":33},"POWER":"ON","Wifi":{"AP":1,"SSId":"AndyCristiWLAN_Priv","BSSId":"44:4E:6D:3D:9E:02","Channel":6,"Mode":"HE20","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

DisplayMode 0
DisplayText [z]Hello

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The display should show the text "Hello"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-10-03:
With Tasmota version 14, the display configurations are changed. For migration see:
https://tasmota.github.io/docs/Universal-Display-Driver/
You need to pick a `display.ini` file and to configure an unused gpio as `Option A 3`

SHA-NDY, 2024-10-04:
Works

closed 2024-10-04 by SHA-NDY