arendst/Tasmota #22248: Ethernet fails on ESP32 Core 3.1.0

arendst, 2024-10-07:
### PROBLEM DESCRIPTION
Installing core 3.1.0 on ESP32 hardware with Ethernet fails etyhernet init. Core 3.0.5 still works fine. The issue is related to IDF changes in function `esp_eth_mac_new_esp32()` specific for RMII on ESP32.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Olimex ESP32-PoE and Sonoff SPM
- [x] Tasmota binary firmware version number used: v14.2.0.6
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Correct functioning:
```
00:00:00.124 Project tasmota - Olimex ESP32-PoE Version 14.2.0.6(theo)-3_0_5(2024-10-07T12:30:23)
00:00:01.691 ETH: Attempting connection...
00:00:01.693 ETH: Connected at 100Mbps Full Duplex, Mac 24:0A:C4:9F:00:AB, Hostname esp32e-eth
00:00:01.703 ETH: IPv4 192.168.2.213, mask 255.255.255.0, gateway 192.168.2.254
```
Bad functioning:
```
00:00:00.141 Project tasmota - Olimex ESP32-PoE Version 14.2.0.6(theo)-3_1_0(2024-10-07T12:40:54)
00:00:00.145 ETH: Bad EthType or init error
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install latest binaries.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Ethernet should still work

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Patching Core 3.1.0 file `ETH.cpp` with error return numbers revealed the error is in IDF function `esp_eth_mac_new_esp32()`. This is specific for RMII and is not used on SPI connected ethernet hardware.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-10-07:
@arendst Can not reproduce. My esp32 board with RMI PHY chip LAN8720 works fine.

ZioFabry, 2024-10-07:
I tryed to ask for support on the Olimex forum... maybe in the wrong way... https://www.olimex.com/forum/index.php?topic=9635.0

Jason2866, 2024-10-07:
@ZioFabry Yeah, the answer is as expected from Olimex. Provide a sample sketch build with upcoming Arduino core 3.1.0 where the board fails.
There is still a chance Olimex is doing no investigation since Arduino core 3.1.0 is in Release Candidate state.

ZioFabry, 2024-10-07:
Sorry @Jason2866, I'm not so expert with the Arduino IDE, I'm still trying to understand how to add the library v3.1.0

Jason2866, 2024-10-07:
@ZioFabry You can use pioarduino to use core 3.1.0 rc1. 
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc1/platform-espressif32.zip
```

arendst, 2024-10-07:
OK. Did another test on a Wireless Tag WT32-S1:
```
00:00:00.353 Project tasmota - Esp32g Version 14.2.0.6(theo)-3_1_0(2024-10-07T12:40:54)
00:00:01.592 ETH: Attempting connection...
00:00:01.594 ETH: Connected at 100Mbps Full Duplex, Mac 24:6F:28:3A:B2:0B, Hostname esp32g-eth
00:00:01.602 ETH: IPv4 192.168.2.165, mask 255.255.255.0, gateway 192.168.2.254
```
Seems to work fine.

So we have two failing units, Olimex ESP32-PoE and SOnoff SPM working fine on 3.0.5 but fail on 3.1.0...

I'll test my other Olimex ESP32-PoE-ISO and see how that fares....

ZioFabry, 2024-10-07:
I can confirm the same on several Olimex variants... POE, POE-ISO, POE-ISO-IND

@Jason2866 I hope to found the time to test it before the end of day...

arendst, 2024-10-07:
> @ZioFabry You can use pioarduino to use core 3.1.0 rc1. https://github.com/pioarduino/platform-espressif32
> 
> ```
> platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc1/platform-espressif32.zip
> ```

Interresting compilke error:
```
/home/arendst/tasmota/DevProdTasmota/tasmota/tasmota_xdrv74-122/xdrv_82_esp32_ethernet.ino:108:3: error: 'ETH_PHY_JL1101' was not declared in this scope; did you mean 'ETH_PHY_TLK110'?
  108 |   ETH_PHY_JL1101,       //  7 = JL1101
```

Jason2866, 2024-10-07:
@arendst as expected the jl1101 driver is added in our Tasmota IDF fork.
The Jl1101 is not officially supported, so not in pioarduino which uses official espressif sources.

arendst, 2024-10-07:
I added it to their fork, Next compile error:
```
lib/default/WiFiHelper/src/WiFiHelper_ESP32.cpp: In static member function 'static wl_status_t WiFiHelper::begin(const char*, wpa2_auth_method_t, const char*, const char*, const char*, const char*, const char*, const char*, int32_t, const uint8_t*, bool)':
lib/default/WiFiHelper/src/WiFiHelper_ESP32.cpp:39:31: error: no matching function for call to 'WiFiClass::begin(const char*&, wpa2_auth_method_t&, const char*&, const char*&, const char*&, const char*&, const char*&, const char*&, int32_t&, const uint8_t*&, bool&)'
   39 |   wl_status_t ret = WiFi.begin(wpa2_ssid, method, wpa2_identity, wpa2_username, wpa2_password, ca_pem, client_crt, client_key, channel, bssid, connect);
      |                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arendst/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:33,
                 from lib/default/WiFiHelper/src/WiFiHelper.h:35,
                 from lib/default/WiFiHelper/src/WiFiHelper_ESP32.cpp:22:
/home/arendst/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:116:15: note: candidate: 'wl_status_t WiFiSTAClass::begin(const String&, wpa2_auth_method_t, const String&, const String&, const String&, const String&, const String&, const String&, int, int32_t, const uint8_t*, bool)'
```

arendst, 2024-10-07:
And with fixed the next error pops up:
```
In file included from lib/libesp32_div/esp-nimble-cpp/src/NimBLEAddress.cpp:19:
lib/libesp32_div/esp-nimble-cpp/src/NimBLEAddress.h:21:10: fatal error: nimble/ble.h: No such file or directory
   21 | #include "nimble/ble.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
Compiling /tmp/.pioenvs/tasmota32-theo/lib27a/esp-nimble-cpp/NimBLEEddystoneTLM.cpp.o
In file included from lib/libesp32_div/esp-nimble-cpp/src/NimBLEDevice.h:41,
                 from lib/libesp32_div/esp-nimble-cpp/src/NimBLEDevice.cpp:18:
lib/libesp32_div/esp-nimble-cpp/src/NimBLEUtils.h:16:10: fatal error: host/ble_gap.h: No such file or directory
   16 | #include "host/ble_gap.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```

arendst, 2024-10-07:
And fixed by adding NimBLE to ignore list.

A compiled binary results in:
```
00:00:00.145 Project tasmota - Olimex ESP32-PoE-Iso Version 14.2.0.6(theo)-3_1_0_RC1(2024-10-07T16:10:45)
00:00:00.184 ETH: Bad EthType or init error
```
Still no luck.

I see they use IDF5.3 too. As I expect the error to be in IDF it explains why their 3.1.0_RC1 fails too.

ZioFabry, 2024-10-08:
No help on the olimex support forum... https://www.olimex.com/forum/index.php?topic=9635.0

arendst, 2024-10-08:
Well their suggestion to stay on core 3.0.5 seems to be the current valid response for Tasmota and GPIO17 based (EthClockMode 3) ethernet connections too.

arendst, 2024-10-09:
It seems it's not IDF related but Arduino Core 3.1.0 related.

As a workaround add the below lines to your platformio ``env:....`` to disable PSRAM and re-enable ethernet:
```
build_unflags               = ${env:tasmota32_base.build_unflags}
                              -DBOARD_HAS_PSRAM
```

ZioFabry, 2024-10-09:
I can confirm that my custom build work fine with the workaround...

```
00:00:00.237 Project biopdu - J360 Version 14.2.0.6(biopdu)-3_1_0(2024-10-09T15:15:13)
00:00:00.242 LVG: Allocating buffer1 10 KB in main memory (flushlines 40)
00:00:00.253 LVG: LVGL initialized
00:00:01.977 ETH: Attempting connection...
00:00:01.979 ETH: Connected at 100Mbps Full Duplex, Mac 40:22:D8:15:22:DB, Hostname J360-0728-eth
00:00:01.983 ETH: IPv4 10.1.204.121, mask 255.255.255.0, gateway 10.1.204.1
```

arendst, 2024-10-09:
The issue is with IDF5.3. During testing PSRAM they claim GPIO's also used by Ethernet ClockMode 3. As they didn't release their claim Ethernet fails on non-PSRAM hardware.

I've added the relaese to Tasmota boot code fixing the issue for now.

Final solution should be in IDF but for now we're saved.

ZioFabry, 2024-10-09:
Thanks @arendst , I confirm you that also on my custom build it's working fine.

nccchirag, 2024-10-09:
> Thanks @arendst , I confirm you that also on my custom build it's working fine.

@ZioFabry - can you also confirm if firmware upgrade works via web UI with wifi disabled?

Im using esp32s3 with w5500 ethernet chip [8 = ETH_PHY_W5500 (SPI)] and firmware upgrade via web does not work via ethernet .. the UI is stuck on "switching to safeboot" page .. thereafter i need to serially flash esp to restore.. however in the same setup if I enable wifi and then perform an upgrade it works fine! @arendst may I request you to guide please

Jason2866, 2024-10-09:
@nccchirag This issue is only for RMII (= LAN8720). Please ask questions in discussions

closed 2024-10-10 by arendst