arendst/Tasmota #22262: problem to wake up from deep sleep mode

dieterferdinand, 2024-10-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
<div id="i" name="i" style="padding: 5px; font-size: 16px; color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">
Program Version | 14.2.0 (release-sensors)
-- | --
2024-08-14T12:36:37
2_7_7/2.2.2-dev(38a443e)
0T00:11:09
36 at 0xF8000
11
Software/System restart
 
SSId ferdinandRSSI 72%, -64 dBmMode 11nChannel 1BSSId 34:31:C4:B1:B3:91
 
tasmota-B3D152-4434
50:02:91:B3:D1:52
192.168.1.202
192.168.1.3
255.255.255.0
192.168.1.3
192.168.1.225
 
Enabled
 
192.168.1.225
1883
DVES_USER
DVES_B3D152
tasmota_%06X
cmnd/tasmotas/
cmnd/tasmota_B3D152/
cmnd/DVES_B3D152_fb/
Disabled
 
None
 
11784530 (ESP8266EX)
0x164068 (DOUT)
4096 KB
1024 KB
715 KB
288 KB
19.1 KB

</div><br class="Apple-interchange-newline">

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:16:44.431 MQT: stat/tasmota_B3D152/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B3D152","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:16:44.469 MQT: stat/tasmota_B3D152/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:39","StartupUTC":"2024-10-10T13:07:05","Sleep":250,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2024-09-28T16:42:52","SaveCount":36,"SaveAddress":"F8000"}}
14:16:44.504 MQT: stat/tasmota_B3D152/STATUS2 = {"StatusFWR":{"Version":"14.2.0(release-sensors)","BuildDateTime":"2024-08-14T12:36:37","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"421/699"}}
14:16:44.527 MQT: stat/tasmota_B3D152/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ferdinand",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:16:44.567 MQT: stat/tasmota_B3D152/STATUS4 = {"StatusMEM":{"ProgramSize":715,"Free":288,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!14,!16,!17,!20,!21,!24,29,!34,62,!65,!66,!68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
14:16:44.622 MQT: stat/tasmota_B3D152/STATUS5 = {"StatusNET":{"Hostname":"tasmota-B3D152-4434","IPAddress":"192.168.1.202","Gateway":"192.168.1.3","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.3","DNSServer2":"192.168.1.225","Mac":"50:02:91:B3:D1:52","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:16:44.647 MQT: stat/tasmota_B3D152/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.225","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B3D152","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:16:44.671 MQT: stat/tasmota_B3D152/STATUS7 = {"StatusTIM":{"UTC":"2024-10-10T13:16:44Z","Local":"2024-10-10T14:16:44","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:03","Sunset":"18:09"}}
14:16:44.692 MQT: stat/tasmota_B3D152/STATUS10 = {"StatusSNS":{"Time":"2024-10-10T14:16:44","BH1750":{"Illuminance":929}}}
14:16:44.702 MQT: stat/tasmota_B3D152/STATUS11 = {"StatusSTS":{"Time":"2024-10-10T14:16:44","Uptime":"0T00:09:39","UptimeSec":579,"Heap":17,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ferdinand","BSSId":"34:31:C4:B1:B3:91","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}}
14:16:45.268 MQT: tele/tasmota_B3D152/STATE = {"Time":"2024-10-10T14:16:45","Uptime":"0T00:09:40","UptimeSec":580,"Heap":19,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":21,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ferdinand","BSSId":"34:31:C4:B1:B3:91","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}
1

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
activate deep sleep mode

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
if deep sleep mode is active, the device must waked up twice to connect to the wlan and send the data to the mqtt-server.
but the device itself send only wake up to the reset, so it remains in sleep mode.

first reset: led blinks
second reset: device wake up and send data and goes sleeping again.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-10-10:
No info provided.

closed 2024-10-10 by Jason2866

dieterferdinand, 2024-10-10:
hello,
i check some things and find this problem: if i reset the device, there are no problemes, if i wake up the device from deep sleep, i think it goed in the booloader for flashing the device and every second reset, it boots normaly!

what must i do, that the device always boot in normal mode instead of the loader for flashing?

goodby

sfromis, 2024-10-10:
If it is entering the bootloader, stop using gpio 0 for things which could pull it low at boot.

(You did still not provide the requested information about device and configuration)

dieterferdinand, 2024-10-11:
hello,
this is the first project with an esp an i don't know, how i get the informations, you want. i have only connected a bh1750 to a wemos d1 mini with the tasmota-sensors software.

i try all things, i found to solve the problem, but nothing helps. i change the wiring to other io-pins without success.

every first reset/power up after deep sleep have the same effect: the esp needs 30-40 mA instead of 70 mA in normal mode and only a second reset/power up solve this problem.

on my last search for a solution i find this:
https://esp32.com/viewtopic.php?f=50&t=39737&p=132824&hilit=deep+sleep#p132824

with this list for different chargings of esp-modules:
* esp12f from supplier A: ISSUE REPRODUCABLE
* esp12f from supplier A: WORKS AS EXPECTED
* esp12f from supplier A: ISSUE REPRODUCABLE
* esp12f from supplier A: ISSUE REPRODUCABLE
* esp01 from supplier B: WORKS AS EXPECTED
* esp12f from supplier C: ISSUE REPRODUCABLE
* esp12f from supplier C: ISSUE REPRODUCABLE
* esp12f from supplier C: ISSUE REPRODUCABLE

and this:
https://forum.arduino.cc/t/esp8266-esp-12f-goes-into-wrong-boot-mode-after-deepsleep/1200836
Here is my ESP-12F purchase history:
2022/11/28 ; 10 pcs from [Diymore PCB Store 1](https://fr.aliexpress.com/store/910440219?spm=a2g0o.order_list.order_list_main.27.19215e5b2YUcr9) ==> Working great
2023/11/26 ; 10 pcs from Diymore PCB Store ==> Zombie mode
2023/12/28 ; 4 pcs from [Estardyn Official Store 1](https://www.aliexpress.com/store/910789017?spm=a2g0o.order_list.order_list_main.7.19215e5b2YUcr9) ==> Working great
2024/01/16 ; 5 pcs from Estardyn Official Store ==> Zombie mode*

i think, that the problem comes from the hardware, but i don't know, if the software can trigger this problem if deep sleep is activated!

if it is possible, to solve this problem with the software, this should be do, if not, i can only use a hardware to power on the esp for a short time and power off it for a longer time. i can do this with a ne555 or other hardware. i will see.

this don't make me happy!

goodby

dieterferdinand, 2024-10-11:
hello,
this is the modul: https://www.makershop.de/plattformen/d1-mini/esp8266-mini-board/

and this board have a bh1750 light sensor connected.

and i only use gpio 4+5. first i use 0 and 2 (as in the plan), but i find out, that this pins have special functions.

i can try to use a 3.3V power suply for testing.

but i think, that a too weak voltage regulator can't be the reason for this problem. if the module get power, it works fine, if i reset the module, it works. only if it was in deep sleep, there is a problem after the next reset or power on. but i will test it with an other 3.3V power source. and i can check the 3.3V power with a memory-oscilloscope.

i can also test, if it have a weak voltage regulator.

goodby

dieterferdinand, 2024-10-12:
hello,
i have tested the voltage regulator with 300mA load, no problem.

goodby

dieterferdinand, 2024-10-12:
hello,
i try to find out, if the gpio2 goes to ground on reset.

it seems, that it does, but only if the reset comes in deepsleep-mode, not in other mode.

the led is connected at gpio2 and the led flash one time on reset. but i can't user the multimeter to get a timing info beetween reset and led activity.

is it possible, the led is activated to fast after a reset in deepsleepmode, so gpio2 is down and device boot not from flash?

goodby

sfromis, 2024-10-12:
I'm wondering why you are still "refusing" to supply the requested config data, even after several reminders. That makes it harder to guess at what may be happening.

And if you are connecting your stuff to gpio 2, I'd avoid that as this gpio also *can* interfere with boot. As a starting point, I'd suggest to use to gpio 4,5,12,13,14 which are free of special circumstances to be extra careful about.

dieterferdinand, 2024-10-12:
hello,
i use only gpio 4 + 5. the led on the board is connected to gpio2!

gpio 0 + 2 are connected in the schema i found for connection the bh1750 and the esp. but i changed this, after i see, that deepsleep makes problems. but nothing what i find can solve this problem.

and then i find an entry in some forums, which describe my problem. but without solution. only that some of the board have that problem, but not all!

and now i try to find out, why my board have this problem, but it seems to be impossible to get a board without this problem, if i order some more boards.

i have an other, bigger, esp32 modul, so i can test this with this board.

but why happens this only if the board is started after going to deepsleep mode and not if no deepsleep is used?

there must be a reason for that.

can it be possible, that gpio2 goes down at reset, if deepsleep was active. and then is down, if it is checked for boot-loader mode?
i don't know, if this check is done by software or hardware to switch to the bootloader for flashing the device.

this is the first time, i use an esp for a project.

i have nothing connected to the esp only the bh1750 light sensor at gpio 4+5 and a powersuply with accus or usb-connector.

goodby

sfromis, 2024-10-12:
Doing a quick test with a similar ESP8266 board with only a BH1750 sensor, gpio 16 connected to rst, and a short DeepSleep cycle, I saw just the normal wakeup every `DeepSleepTime`.

dieterferdinand, 2024-10-13:
hello,
i have only this configuration and a jumper between gpio16 and reset and it does not work.

only after deepsleep the esp boot in a false mode, after normal power up or reset it works fine.

goodby

dieterferdinand, 2024-10-16:
hello,
with a bigger esp32 it works fine. the problem must be in the hardware of the d1 mini
or while gpio2 is used for the led.

i will check timing fast as posible.

goodby

dieterferdinand, 2024-10-18:
hello,
i find a solution for that problem:
https://github.com/esp8266/Arduino/issues/6007

a resistor between the 3.3V pin and the second pin at side of the usb-connector on the modul.
with picture:
https://github.com/esp8266/Arduino/issues/6007#issuecomment-2080368936

with this resistor deepsleep works right.

goodby