arendst/Tasmota #22267: Bluetooth BT BLE devives not connecting properly

AleXSR700, 2024-10-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I do not know if this is related to issue https://github.com/arendst/Tasmota/issues/21445, but after upgrading my esp32 devices to 14.2.0.6, my eQ3 devices are no longer connecting reliably.

I have 8 Xiaomi devices and 8 EQ3s. The Xiaomis (running ATC firmware) seem to work but the EQ3s just show `20:44:07.951 EQ3: 001A2208C9FA: trv operation failed - retrying -12` (with the respective device ID).

Interestingly, I see all 8 EQ3 TRVs right after a restart and they briefly show the correct temperatures in the WebUI. But then the connection starts failing and all values are set to 0.

Does anybody have similar issues?

EDIT: Reverted to 13.4.0 (http://ota.tasmota.com/tasmota32/release-13.4.0/tasmota32-bluetooth.bin) and it works again. So seems that something is broken in 14.2.0.6 again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): esp32 and esp32c3
- [x ] Tasmota binary firmware version number used: 14.2.0.x
  - [x ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Upgrade to 14.2.0 and enable BLE (if not already enabled). Check console for EQ3-TRV messages.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-10-12:
@btsimonh Can you please have a look?

SteWers, 2024-10-13:
It is working fine in 14.0.0.1

btsimonh, 2024-10-13:
@Jason2866 - family issues, but I'll try to take a look :(

btsimonh, 2024-10-15:
works on fresh build of 14.2.0.1 on S3.  EQ3 polling all good, out of 10 polls of three devices, only one failure (to be expected).
works on build of 14.2.0.5.....
I note that 1248bd138c7eda34fe91d9f60721a78210a27b56 changes the core version (commit before 14.2.0.6)
works with 14.3.0.1 69f7f155dd3750ff44f620556d3d46c3ae25beda
But... possibly more failed to connect, a few in a row.

btsimonh, 2024-10-15:
@AleXSR700 could you try again?  If with 14.3.0.1 you still have consistent failures, check your wifi traffic (including MQTT, etc.).  I had none beyond the GUI.  Wifi and BLE do NOT play nice....  If possible, check serial logs with all connections closed to the esp.
If necessary, I will build for 'std' esp32 (I have some webcam boards I bought for Eq3 use because of external antennas), and test again; just not sure when.

AleXSR700, 2024-10-15:
I will start a new build and start testing tomorrow.

Btw, you are building the base binaries instead of the `bluetooth` variant?

```
#define USE_MI_ESP32 
#define USE_BLE_ESP32 
#define USE_EQ3_ESP32
```

These are all part of the bluetooth variant (not sure if there is more to it than this plus the libs).

I will test as before with the `bluetooth` variant and if that fails, build and test the base image with those three  enabled manually to countercheck.

btsimonh, 2024-10-16:
I built `tasmota32s3-bluetooth` with the above. (bluetooth variants did not used to have USE_BLE_ESP32 or USE_EQ3_ESP32.  now it seems they all do?) 

I double checked configs, and it seems FIRMWARE_BLUETOOTH contains all for EQ3 except in _mi32 variants.

so double check your `platformio_tasmota_cenv.ini` is the same as the `platformio_tasmota_cenv_sample.ini` (for your c3), and the pure bluetooth variants should be fine.  The fact that you have logs from the EQ3 driver indicate your build contained everything required.

btsimonh, 2024-10-16:
I don't have a c3 to test with :(.  could you try on plain esp32?

Jason2866, 2024-10-16:
There is a big update ahead for the BLE legacy stuff. Look in open PRs ;-)

AleXSR700, 2024-10-16:
> I don't have a c3 to test with :(. could you try on plain esp32?

Yes, I have a devkit esp32. So if you have a build you would like me to test, happy to.

AleXSR700, 2024-10-16:
> There is a big update ahead for the BLE legacy stuff. Look in open PRs ;-) There is no TRV driver for the MI BLE (and there will be none). The MI BLE is the more actual BLE code.

Would kind of be nice to have one BLE variant for all BLE devices :)

Jason2866, 2024-10-16:
> Would kind of be nice to have one BLE variant for all BLE devices :)

Will not happen since the drivers have different goals and design concepts.

btsimonh, 2024-10-16:
looking at the berry in the PR, seems an EQ3 driver could be written in berry over the 'new' legacy stuff? (I am not volunteering!).

AleXSR700, 2024-10-16:
@btsimonh @SteWers 👍 
Did you test with latest TRV firmware 1.46 or above?

I now flashed the tasmota32-bluetooth and tasmota32c3-bluetooth from here: https://github.com/tasmota/install/releases/tag/2024-10-16.d.e39f1cc83f17100adc41fb8b6d18a792ecd50c11

I wanted to rule out any build issues.

Neither the ESP32 nor the ESP32C3 seem to work properly. In the WebUI I see all my ATC information (Xiaomi Mijia devices) correctly but all my EQ3s only show 0 values (except for RSSI, which is not a value published by the EQ3). 

So all set point temperatures, valve positions and batteries are 0 all the time.

I flashed the downloaded binaries and then performed a reset 1. Then connected back to network, turned on BLE and power cycled.

AleXSR700, 2024-10-16:
> I don't know the firmware of the TRV.

It is displayed when you put the batteries in. 
In the newer firmware (1.46 and above) the manufacturer changed the pairing mechanism/key.
https://github.com/arendst/Tasmota/issues/16775

So if the tested EQ3 firmware is older, the issue might be related to the linked one and would not affect all EQ3 devices.
I upgraded of mine to 1.46, so it is not possible to test if it would work with the older ones.

SteWers, 2024-10-16:
> It is displayed when you put the batteries in.

Mine has version 1.20

btsimonh, 2024-10-17:
mine are all older firmware.
it's odd that is works for you with older TAS, but newer EQ3 firmware - the key stuff was some time ago.
I'd suggest that the key stuff was not about the key (I don't think it needed it in the end?  it just worked again after some tweaks?), but about doing some handshake.  Maybe check the commits associated with https://github.com/arendst/Tasmota/issues/16775 and see what was added to make it work then, then check for changes in the idf/nimble?

AleXSR700, 2024-10-17:
> mine are all older firmware.
> it's odd that is works for you with older TAS, but newer EQ3 firmware - the key stuff was some time ago.

No, I deliberately stayed with firmware versions newer than that handshake/pairing change.
So firmware v12 and v13 versions post https://github.com/arendst/Tasmota/issues/16775

> Maybe check the commits associated with https://github.com/arendst/Tasmota/issues/16775 and see what was added to make it work then, then check for changes in the idf/nimble?

Since all of your devices are older firmware but work and the same applies for SteWers, I was thinking the same thing. Seems like a change made back im v12 to fix the compatibility with newer EQ3 firmware is no longer present or working in v14.

Maybe @h2zero has an idea?

btsimonh, 2024-10-19:
looked for my 'spare' eq3 today (the one i dismantled and broke the wheel on...) to see if I could upgrade it and test.
But it's not going to happen:
[image]
Washing has so far not brought it to life....

btsimonh, 2024-10-19:
ok, that one is fubar.  (It does now start, but I can't read the display).
I found another... (one which had broken off the rad).  It's now on 1.46, and I get at a distance of 20cm:
```
15:44:09.723 EQ3: 001A22092FB7: Op queued len now 1
15:44:09.725 EQ3: 001A22092FB7: Op dequeued len now 0
15:44:12.314 BLE: failed subscribe for notify
15:44:12.438 BLE: BLETask: op failed -12
15:44:13.691 EQ3: 001A22092FB7: trv operation failed - retrying -12
15:44:13.697 RSL: BLE = {"BLEOperation":{"opid":"1","stat":"-12","state":"FAILNOTIFY","MAC":"001A22092FB7","u":"128","svc":"3e135142-654f-9090-134a-a6ff5bb77046","char":"3fa4585a-ce4a-3bad-db4b-b8df8179ea09","notifychar":"d0e8434d-cd29-0996-af41-6c90f4e0eb2a","write":"03180A130F2C09000000000000000000"}}
15:44:15.956 BLE: failed subscribe for notify
```
It works with the Calorbt app.
5 other EQ3 have responded....
Tas version 14.3.0.1 on esp32S3

I'll check it out some more.

btsimonh, 2024-10-19:
@Staars Cristian - help :).  You have latest and intimate knowledge of esp-nimble-cpp....
@h2zero too if you are able.

We're hitting AddLog(LOG_LEVEL_ERROR,PSTR("BLE: failed subscribe for notify")); - line 1921 in tasmota\tasmota_xdrv_driver\xdrv_79_esp32_ble.ino

This is happening only for EQ3 firmware versions which have 'new' security.
Seems to me that the device is connected, has a valid characteristic, says that characteristic can notify, but then nimble is telling us it can't subscribe.... (pNCharacteristic->subscribe returns false).

Any advice or guidance appreciated.
br,
Simon

Staars, 2024-10-19:
In line 1911:
`            if(pNCharacteristic->subscribe(true, BLE_ESP32::BLEGenNotifyCB)) {`

You try to subscribe to an indication and not a notification. Is this intended?

Staars, 2024-10-19:
Ahh, no ... it is the opposite. forget it.

Staars, 2024-10-19:
But at least in 1911 an incomplete function signature is used - missing the `response`. IDK if this is a problem.

btsimonh, 2024-10-19:
14.1.0 (lvgl-haspmota) - not working,
trying 14.0.0.2..... not working.
13.4.1.1 (lvgl-haspmota)- BLE does not work at all.
13.4.0(lvgl-haspmota) works.

AleXSR700, 2024-10-19:
> 14.1.0 (lvgl-haspmota) - not working,
> trying 14.0.0.2..... not working.
> 13.4.1.1 (lvgl-haspmota)- BLE does not work at all.
> 13.4.0(lvgl-haspmota) none of the eq3 work.  maybe something is wrong with my build.
> 

See the opening post of this thread. I linked the build for 13.4.0 there. Just download that one. It won't have yourWifi or mqtt settings of course, but should at least work for testing.

h2zero, 2024-10-19:
If I was to guess I would say that it's failing to subscribe because it should write to the descriptor with response request as per BT specifications. The older devices/firmwares wouldn't accept that in the past so that's the reason for writing without response now. At least that's what i remember.

btsimonh, 2024-10-20:
come to it in the morning, and 13.4.0 on S3 is working for all Eq3, including ones running older firmware and the one running 1.46 - it must have just been taking it's time getting the data, and I rushed the assessment.
I'll walk forward again in versions.
@Staars - `response` is defaulted to true in the header for the function.  but from h2zero, maybe this is the issue....  maybe it needs false.
@h2zero - could this https://github.com/h2zero/esp-nimble-cpp/commit/5b24c8d681f967be85dd6dd1566657f7a47b0c06 be involved? - or is that only for BLE servers?

btsimonh, 2024-10-20:
@Staars @h2zero - the PR works with both of the versions of EQ3 firmware that I have.
Please have a quick look and comment.
(I thought this may be in the characteristics properties, but the props are always `characteristic props 0x1A` for both firmwares).

btsimonh, 2024-10-20:
@AleXSR700  - I can't seem to find individual artifacts for the PR, but I assume the firmwares are all in  https://github.com/arendst/Tasmota/actions/runs/11424841639/artifacts/2078856871 - 138mbytes.

Staars, 2024-10-20:
> @Staars @h2zero - the PR works with both of the versions of EQ3 firmware that I have. Please have a quick look and comment. (I thought this may be in the characteristics properties, but the props are always `characteristic props 0x1A` for both firmwares).

I suppose the needed response value is device specific (here for the EQ-3). So the PR looks good to me.

Since a few days we can easily customise the Arduino framework for a specific Tasmota build, which is especially useful for (before impossible) individual Bluetooth settings including the log level.
If it is not enough to add a build flag, like: `-DCONFIG_NIMBLE_CPP_LOG_LEVEL=3` to `build_flags `
you can bake these kind of things into the Arduino framework now with:
```
custom_sdkconfig  =  CONFIG_NIMBLE_CPP_LOG_LEVEL=3
```

I did not test this particular use case yet. Maybe you need to provide both sections, it is still e very new feature and not finally finished.

closed 2024-10-20 by Jason2866

btsimonh, 2024-10-20:
@Staars - did you test notify to get battery from MI in your new implementation yet?
The PR mod will affect MI32 when using BLE32 - i.e. it uses the same notify fn.
If you test and find that `response` MUST be true for the MI battery read fn, then I'll make it EQ3 specific somehow.... - let me know.

AleXSR700, 2024-10-20:
Thank you all!
I was away for the weekend but will also test tomorrow :)

Staars, 2024-10-21:
> @Staars - did you test notify to get battery from MI in your new implementation yet? The PR mod will affect MI32 when using BLE32 - i.e. it uses the same notify fn. If you test and find that `response` MUST be true for the MI battery read fn, then I'll make it EQ3 specific somehow.... - let me know. (I don't have an MI style device that required connection....)

As I converted basically all my devices to PVVX I can not tell ATM. It was a cumbersome procedure anyway and I suggest to wait for "real world" user issues. But if I stumble across findings I will report here.

AleXSR700, 2024-10-21:
Completely off topic, but will you maybe implement BTHome into the normal bluetooth variant so it can be used on devices that also handle EQ3? 

Right now it is either BTHome (tasmota legacy ble) or Pvvx  (soon deprecated) with EQ3 (current `-bluetooth` variant).

AleXSR700, 2024-10-23:
I am not sure if this is possible, but the ble experts here might be able to say:
It seems to me as if the EQ3 makes the device less stable.

I have 8 Mijia temp+hum sensors and 8 EQ3s. My esp32 devices (esp32 & esp32c3) work without crashes when I only use the Mijia devices.
When the TRVPeriod is lowered, it seems to increase the crash frequency a lot.
So with TRVPeriod 300 my esp32c3 devices crash multiple times per day. When I increase to 3600, the devices seem to not crash more than once.

It is still too early since the update to say for sure, but is it possible that somehow the EQ3 driver blocks the wifi-ble module too long or something like that? Layman talking here...

btsimonh, 2024-10-23:
> the versions I built without EQ3 support or when the driver was broken

was it like this?:
```
#define USE_MI_ESP32 
#define USE_BLE_ESP32 
//#define USE_EQ3_ESP32
```
i.e. using the BLE_ESP32 version of BLE?

There is a big interaction between BLE of any form and wifi.  Basically on ESP32, you CAN'T mix them.  Really for BLE and TAS we should have one device doing BLE, and another doing wifi.

e.g. when I built webcam+ble, and then used the webcam UI, no device would last more than ~30 mins, some crashed in < 1 min.

Reducing TRVPeriod -> more MQTT & more time with active BLE connections -> more BLE + more wifi -> more crashes :(.

Using just MI sensors, it's using very little BLE, and so the risk is reduced (basically just hearing adverts, no active connections).

see also https://github.com/arendst/Tasmota/issues/20970 and other reported BLE issues.

AleXSR700, 2024-10-23:
Why is an active connection needed for EQ3 but not Mijia? The esp32 devices receive data without active connection. But they need an active connection for the query?
If so, can the duration if this active connection maybe be reduced/optimized?

Again, just asking as a layman.

I have found that the active checking via the TRVPeriod is rarely needed. The TRV will report any change without being queried (and that is received by the tasmota esp32 without active connection). So only an initial query is needed to get the starting value whenever the esp32 device or the Home Automation server are restarted. Apart from that, any change will be actively broadcast by the TRV and received by the esp32 tasmota devices.

btsimonh, 2024-11-10:
@AleXSR700 - sorry for the delay.  The EQ3 needs to be `read`, which requires a connection, whereas the MI devices broadcast their stats in BLE adverts - so anyone can hear the info without connecting.  The exception to this is some MI devices don't include battery level, but this can be read using a connection.

So far as I remember, 'any change will be actively broadcast by the TRV' is not true.  We will only get updated info from the TRVs when we ask for it.  (i.e. at the TRV interval).

AleXSR700, 2024-11-10:
> So far as I remember, 'any change will be actively broadcast by the TRV' is not true.  We will only get updated info from the TRVs when we ask for it.  (i.e. at the TRV interval).

Thanks for your response.
Nope, the above is definitely true. I have set the TRVPeriod to 1 hour and I get constant updates when the TRV is running. It sends updates whenever there is a change in temperature (I think also valve). No need to ask for it :)

Just manually change the temperature at the dial or via HA and you will see that it broadcasts once it has received the command (be it via dial or via MQTT).

Einstein67, 2024-11-11:
I can agree with btsimonh's observations. If TRVperiod is set to 0, no values come from the valves. Regardless of whether I control them via the app or change the temperature on the dial.
I change the value with the status of my heating circuit. 
Heating circuit pumps on --> TRVPeriod 300
Heating circuit pumps off --> TRVPeriod 0
When the circuit is not running, I can only see the current temperature setting, but only the last valve position. But that doesn't interest me at all at this point. An active query using ‘TRV status’ is always possible.

AleXSR700, 2024-11-11:
> I can agree with btsimonh's observations. If TRVperiod is set to 0, no values come from the valves. Regardless of whether I control them via the app or change the temperature on the dial.

Please set TRVPeriod to something high, like 3600 and then use the dial.
You should see the publish in the console.
I know I do. So either that is new in TRV firmware 1.46 or TRVPeriod 0 disables the TRV all together (have not tested that myself).

AleXSR700, 2024-11-14:
@btsimonh 
I thought about this a bit more and I was wondering if maybe we are staying connected "needlessly".

For sure the TRV publishes whenever there is a change and tasmota does not need to connect to receive that message. Essentially it behaves like a Mijia thermometer etc. They publish and anyone within reach can listen. Not exactly safe, but what the heck...

So I think right now you are connecting, querying the status and waiting for the response before disconnecting.
I think you could simply send a command to the TRV and then wait for the response *without* being connected.

Any valid command will trigger the publish and any device will see it (without being connected).

That should significantly reduce the duration of the connection to the pure time that is needed for sending the command.

AleXSR700, 2024-11-18:
> The advertisments from the TRVs (data send periodically and without connection) do not contain any useful data for us.  The only way to get useful data is to connect to each TRV in turn, and make some request (any - since upon any request, the TRV will send a 'notification' containing the latest data).  In order to receive a notification, you must be connected.  

Here I am not sure if this is not a contradiction to your first paragraph.

As you mentioned, when you send a command, the TRV will respond with the data we need. So you do not need to stay connected and wait for a response, you just need to send a command and then you will automatically get a response.
You could of course add a verification, i.e. send the command, remain disconnected, send another command if there is no response within 5 s of the first command. Only connect if this fails 3 times.

Should this not significantly reduce the connection duration and hence the shared wifi/ble blocking?

lbs4pich, 2024-12-27:
Hello, is there any fix now for Firmware > 1,46

AleXSR700, 2024-12-27:
> Hello, is there any fix now for Firmware > 1,46

Isn't 1.46 the latest (and probably final) version? Or was there another update?
Tasmota worked after the last comitts from this thread.  But I have not tried the latest release. Has that stopped working again?

andrei-marinache, 2024-12-28:
I updated one of my EQ3 TRVs to 1.48 about two weeks ago and since then I can't connect to it using any Tasmota version, I just tried today with Tasmota 14.4.1.1 (de3a7f3-bluetooth).

```
14:30:29.984 CMD: trv devlist
14:30:29.986 SRC: WebConsole from 192.168.69.125
14:30:29.988 CMD: Grp 0, Cmd 'TRV', Idx 1, Len 7, Pld -99, Data 'devlist'
14:30:29.991 EQ3: cmd index: 1
14:30:29.995 MQT: stat/tasmota_F9DCC4/EQ3 = {"devices":{"001A2216F340":-80}}
14:30:29.998 MQT: stat/tasmota_F9DCC4/RESULT = {"trv":"Done"}
14:30:36.932 WIF: Checking connection...
14:30:37.134 EQ3: poll overrun, deferred - last loop only got to 1, not 16
14:30:38.179 BLE: failed to connect to device low level rc 0xd
14:30:38.187 BLE: failed to connect to device
14:30:39.136 EQ3: 001A2216F340: trv operation failed - retrying -11
14:30:39.144 MQT: tele/tasmota_F9DCC4/BLE = {"BLEOperation":{"opid":"3601","stat":"-11","state":"FAILCONNECT","MAC":"001A2216F340","u":"128","svc":"3e135142-654f-9090-134a-a6ff5bb77046","char":"3fa4585a-ce4a-3bad-db4b-b8df8179ea09","notifychar":"d0e8434d-cd29-0996-af41-6c90f4e0eb2a","write":"03180C1C0E1D25000000000000000000"}}
14:30:42.387 CMD: trv 001A2216F340 state 1
14:30:42.389 SRC: WebConsole from 192.168.69.125
14:30:42.391 CMD: Grp 0, Cmd 'TRV', Idx 1, Len 20, Pld 1, Data '001A2216F340 state 1'
14:30:42.395 EQ3: cmd index: 1
14:30:42.397 EQ3: 001A2216F340: Op queued len now 1
14:30:42.398 EQ3: queued
14:30:42.402 MQT: stat/tasmota_F9DCC4/RESULT = {"trv":"queued"}
14:30:56.941 WIF: Checking connection...
14:31:09.229 BLE: failed to connect to device low level rc 0xd
14:31:09.232 BLE: failed to connect to device
14:31:10.143 EQ3: 001A2216F340: trv operation failed - retrying -11
14:31:10.150 MQT: tele/tasmota_F9DCC4/BLE = {"BLEOperation":{"opid":"3602","stat":"-11","state":"FAILCONNECT","MAC":"001A2216F340","u":"128","svc":"3e135142-654f-9090-134a-a6ff5bb77046","char":"3fa4585a-ce4a-3bad-db4b-b8df8179ea09","notifychar":"d0e8434d-cd29-0996-af41-6c90f4e0eb2a","write":"03180C1C0E1D25000000000000000000"}}
14:31:15.134 NTP: Sync time...
14:31:15.217 WIF: DNS resolved 'pool.ntp.org' (44.32.133.2) in 81 ms
14:31:15.294 RTC: UTC 2024-12-28T13:31:16Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
14:31:16.000 RTC: Synced by NTP
14:31:16.950 WIF: Checking connection...
14:31:36.953 WIF: Checking connection...
14:31:37.155 EQ3: poll overrun, deferred - last loop only got to 1, not 16
14:31:40.174 BLE: failed to connect to device low level rc 0xd
14:31:40.178 BLE: failed to connect to device
14:31:41.155 EQ3: 001A2216F340: Poll fail not sent because EQ3HideFailedPoll
14:31:41.157 EQ3: 001A2216F340: trv operation failed - no more retries -11
14:31:41.164 MQT: tele/tasmota_F9DCC4/BLE = {"BLEOperation":{"opid":"3603","stat":"-11","state":"FAILCONNECT","MAC":"001A2216F340","u":"128","svc":"3e135142-654f-9090-134a-a6ff5bb77046","char":"3fa4585a-ce4a-3bad-db4b-b8df8179ea09","notifychar":"d0e8434d-cd29-0996-af41-6c90f4e0eb2a","write":"03180C1C0E1D25000000000000000000"}}
14:31:41.188 EQ3: 001A2216F340: Op dequeued len now 0
```

andrei-marinache, 2026-01-14:
No, I've installed a bluetooth proxy on an esphome device, I am using it in Homeassistant.