arendst/Tasmota #22282: shutter1#position triggers every second

kugelkopf123, 2024-10-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I upgraded my Shelly2.5 from v9.x to the latest dev-branch today. Reason: Had sometimes issues with HW Watchdog and wrong shutterposition after that. The update was done via serial not OTA. 
Since then shutter1#position in rule3 triggers every second. 
Because I thought it was due to the big version jump, I deleted the shellys via serial and reflashed them.

There are other shutters running in the house (v9.x) with the same rule and there are no problems there. 

Rule3: `on SHUTTER1#Position do var1 %value% endon; on shutter#moved do backlog publish stat/%topic%/RESULT {‘ShutterPosition1’:%var1%} endon`

Even if you shorten it to:
`Rule3 on Shutter1#position do var1 %value% endon`

the rule is triggered every second.

By the way, it doesn't matter whether rule3 or rule1 or 2.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly2.5
- [ ] Tasmota binary firmware version number used: 14.2.0.6 (tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VisualStudioCode
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

14:11:54.634 CMD: Backlog Template; Module; GPIO 255
14:11:54.656 RSL: RESULT = {"NAME":"Shelly 2.5BT","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
14:11:54.784 RUL: SHUTTER1#POSITION performs 'var1 100'
14:11:54.789 RSL: RESULT = {"Var1":"100"}
14:11:55.031 RSL: RESULT = {"Module":{"0":"Shelly 2.5BT"}}
14:11:55.286 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp."}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

14:12:45.786 CMD: Backlog Rule1; Rule2; Rule3
14:12:45.866 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:12:45.881 RUL: SHUTTER1#POSITION performs 'var1 100'
14:12:45.886 RSL: RESULT = {"Var1":"100"}
14:12:46.097 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:12:46.347 RSL: RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":42,"Free":469,"Rules":"on Shutter1#position do var1 %value% endon"}}

```

- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:13:18.965 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_F3EF1F","ButtonTopic":"0","Power":"00","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:13:18.975 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software/System restart","Uptime":"0T00:47:21","StartupUTC":"2024-10-13T11:25:57","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2024-10-13T13:16:19","SaveCount":12,"SaveAddress":"1000"}}
14:13:18.981 RSL: STATUS2 = {"StatusFWR":{"Version":"14.2.0.6(tasmota)","BuildDateTime":"2024.10.13 13:25:04","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
14:13:18.987 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C001","2805C80001000600003C5A0A192800000000","40000088","00006000","00004000","00000000"]}}
14:13:18.997 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":460,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","8800058D","04000001","00000006","01000040","00000100","00004080","00200000","00000020","00000000","00000000"],"Drivers":"1,!2,3,10,27,38,50","Sensors":"2,3","I2CDriver":"7"}}
14:13:19.005 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-F3EF1F-3871","IPAddress":"192.168.1.206","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"98:F4:AB:F3:EF:1F","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
14:13:19.011 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-10-13T12:13:19Z","Local":"2024-10-13T14:13:19","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99}}
14:13:19.016 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0,"MaxPower":0,"MaxPowerHold":10,"MaxPowerWindow":30,"MaxEnergy":0,"MaxEnergyStart":0}}
14:13:19.026 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-10-13T14:13:19","ANALOG":{"Temperature":51.9},"ENERGY":{"TotalStartTime":"2024-10-13T13:16:19","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"TempUnit":"C"}}
14:13:19.037 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-10-13T14:13:19","Uptime":"0T00:47:22","UptimeSec":2842,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Eiswuerstchen","BSSId":"DC:15:C8:47:D8:77","Channel":11,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:04"}}}
14:13:19.044 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1"}}}

[... 37 lines cut ...]
14:21:48.791 RUL: SHUTTER1#POSITION performs 'var1 100'
14:21:48.797 RSL: RESULT = {"Var1":"100"}
14:21:49.801 RUL: SHUTTER1#POSITION performs 'var1 100'
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

This Rule should only be triggered if the Position is changing.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The purpose of the rule is that when the blinds are adjusted via the wall switch, the current position must be written to the `Shutterposition1` topic via MQTT.
I use the Yahka adapter, if the current position and the target position are different, because the topic `shutterposition1` is not updated after moving the blinds via the wall switch, ‘Apple Home’ shows ‘is closed’ or ‘is opened’ all the time. 

The above rule is not ideal. 
I would prefer it if the variable Shutterposition1 is also updated when the blinds are adjusted manually. Alternatively, if I didn't have to use var1 at all but could simply do something like this with rules:
If shutter1#moved>1 then publish mqtt shutterposition1 %acutal position% endif. 

Unfortunately, I don't know how to implement this in rules.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2024-10-21:
This works the standard compile (without the conditional one). Maybe this solves your use case.

rule2
 on shutter#moving do rule3 on endon
 on shutter#moved do backlog rule3 off; publish hm/status/ESP_850248/position %var1% endon

rule3
  on shutter1#position do var1 %value% endon

kugelkopf123, 2024-10-21:
> Yes it is correct that the position will updated every second to support other application like HA. to show nice animations where the shutter actually is. There are other triggers like shutter1#moved that you might combine. As said. this is not a bug and since version 9 there where several enhancements made to the shutters.

Ok, thank you very much for your answer.
Also for the help with the rules. I'll give that a try.

closed 2024-10-21 by kugelkopf123

stefanbode, 2024-10-24:
@kugelkopf123 : I have submitted an update right now that fix the issue with the data send every second. Nor the rule with the position is only triggered if one of the defined shutters is moving. Should make it easier to build automation.

kugelkopf123, 2024-10-24:
Perfect. This is more or less the same old behaviour. 
I was also a little worried that the constant switching on and off of the rules might lead to excessive wear on the flash memory.
Many thanks for your help! I'll try it at the weekend.