arendst/Tasmota #22357: Cannot assign to ESP12F_Relay_X8 more than 8 IDX to use them on domoticz

paolomeraviglia, 2024-10-25:
### PROBLEM DESCRIPTION
hi there,

I am using 3 of these:
https://templates.blakadder.com/ESP12F_Relay_X8.html

everything works well except menu under http://ip/dm?

where there is no way to assign to relay 5,6,7,8 and IDX from domoticz.
[image]

That was like that on 13.2, and still on 14.30 that should be the latest.
All the relays are switching properly using webui. They are simply not addressable here.
Template applied is the one on blackadder site:

{"NAME":"ESP12F_Relay_X8","GPIO":[229,1,1,1,230,231,0,0,226,227,225,228,224,1],"FLAG":0,"BASE":18}

Thanks for your help!

paolomeraviglia, 2024-10-25:
seems only a visual issue on webui.
to configure idx from 5th to 8th it is possible to use the console, with command

Dzidx<x> idx

where <x> is the relay, idx is the idx number of the device on domoticz.

arendst, 2024-10-25:
As designed. You found the workaround already.

closed 2024-10-25 by arendst

paolomeraviglia, 2024-12-17:
I saw that after a reboot, the function don't save the id, so need to re-insert the dzidx manually.
Is there a way to save it and preserve this info in the saved config?