arendst/Tasmota #22361: SetOption69 doesn't work to invert serial input for the serial bridge

kobuki, 2024-10-25:
### PROBLEM DESCRIPTION
SetOption69 doesn't work to invert the serial RX signal for the serial bridge. Received data is garbage using ON and OFF. It's ineffective.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-DevKit
- [ ] Tasmota binary firmware version number used: 14.3.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: online tasmotizer + manual fw update on web ui
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:39:44.237 MQT: stat/ma105-meter/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:39:44.462 MQT: stat/ma105-meter/RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:39:44.671 MQT: stat/ma105-meter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
STATUS 0 output here:
21:40:52.188 MQT: stat/ma105-meter/STATUS = {"Status":{"Module":1,"DeviceName":"ma105-meter","FriendlyName":["P1 port"],"Topic":"ma105-meter","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:40:52.197 MQT: stat/ma105-meter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:04:14","StartupUTC":"2024-10-25T20:36:38","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2024-10-25T14:28:04","SaveCount":51}}
21:40:52.202 MQT: stat/ma105-meter/STATUS2 = {"StatusFWR":{"Version":"14.3.0(tasmota32)","BuildDateTime":"2024-10-23T15:55:30","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"423/699"}}
21:40:52.210 MQT: stat/ma105-meter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"10.11.12.88","LogPort":514,"SSId":["altair",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00408009","2805C80001000600003C5A0A192800000000","00080080","00006000","00004000","00000000"]}}
21:40:52.233 MQT: stat/ma105-meter/STATUS4 = {"StatusMEM":{"ProgramSize":2060,"Free":819,"Heap":127,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","979AD7DF","001DA001","B7F7BFCF","05DA9BC4","E0360DCF","480840D2","20200000","D4BC482D","810A80B1","00000014"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,53,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
21:40:52.241 MQT: stat/ma105-meter/STATUS5 = {"StatusNET":{"Hostname":"ma105-meter","IPAddress":"10.11.12.111","Gateway":"10.11.12.1","Subnetmask":"255.255.255.0","DNSServer1":"10.11.12.1","DNSServer2":"0.0.0.0","Mac":"FC:F5:C4:31:62:40","IP6Global":"","IP6Local":"fe80::fef5:c4ff:fe31:6240%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.11.12.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:40:52.245 MQT: stat/ma105-meter/STATUS6 = {"StatusMQT":{"MqttHost":"10.11.12.246","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_316240","MqttUser":"iot","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:40:52.250 MQT: stat/ma105-meter/STATUS7 = {"StatusTIM":{"UTC":"2024-10-25T20:40:52Z","Local":"2024-10-25T21:40:52","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:27","Sunset":"17:40"}}
21:40:52.259 MQT: stat/ma105-meter/STATUS10 = {"StatusSNS":{"Time":"2024-10-25T21:40:52"}}
21:40:52.269 MQT: stat/ma105-meter/STATUS11 = {"StatusSTS":{"Time":"2024-10-25T21:40:52","Uptime":"0T00:04:14","UptimeSec":254,"Heap":129,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":46},"Wifi":{"AP":1,"SSId":"altair","BSSId":"B0:48:7A:99:AC:84","Channel":6,"Mode":"HT40","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
For instance, hook up a smart meter P1 port serial output to the device. The output is inverted TTL serial. In theory, `SetOption69` can be used to allow interpreting it. Use commands below first.
```
setoption69 1
serialsend2 1111
```

### EXPECTED BEHAVIOUR
I expect the serial data to appear but instead, garbage is received. It seems that SetOption69 is not effective.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The smart meter interface can deal with the inverted input. I use the meter special command `1,=so2,4` for that and the input is decoded properly and can be dumped with `Sensor53 d1`. That means the serial output of meter is working properly.

I also tried a small Berry POC:

```
# the last parameter enables RX inverting
ser = serial(3, 1, 115200, serial.SERIAL_8N1, true)
msg = ser.read()
print(msg.asstring())
```

The incoming message is properly displayed. Only the serial bridge has problems.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-10-25:
Well, the configuration output you posted has no pins configured at all, and the `Status 0` output has `!8` under "Drivers", indicating that the serial bridge driver was not enabled at that time. But I could confirm than `SetOption69` was enabled.

Not sure if  what you posted was not when trying to use serial bridge, but the point of those questions is that you should fill them in with information directly relevant to having the issue you post about.

kobuki, 2024-10-26:
Well, I have a running serial bridge now, as I had when I posted my OP. I updated the `status 0` output for you, but !8 is still present in the drivers section, yet the bridge is working. Maybe it's not relevant? On the other hand, you don't need to set pins for the serial bridge to work, it's also in the docs.

Sample output from a few mins ago:

```
10:24:57.545 CMD: serialsend2 1111
10:24:57.551 MQT: stat/ma105-meter/RESULT = {"SerialSend":"Done"}
10:25:07.076 MQT: tele/ma105-meter/RESULT = {"SerialReceived":"=3"}
10:25:16.753 MQT: tele/ma105-meter/RESULT = {"SerialReceived":"4"}
10:25:26.813 MQT: tele/ma105-meter/RESULT = {"SerialReceived":"F>^>>"}
```

Obviously, I wouldn't have made a bug report if all worked flawlessly... Please believe me that I tried all possibilities I could think of. I would be glad if it worked or a mistake pointed out that would fix it in my setup, but right now I think it's a bug in Tasmota. I can do any tests or settings you want to help find the root cause.

sfromis, 2024-10-26:
But you still did not post the configuration details (first question), and as you have that `!8`, it looks like you are not even configuring for serial bridge.

kobuki, 2024-10-26:
Sorry, I don't understand. Please tell me exactly what you need. There is no pin configuration set up at all, and I filled the questionnaire, too. The GPIO report didn't change. The The serial bridge works without configuring any pin - in that case, default serial setup is in effect (also for the bridge): GPIO 1, 3 for TX, RX, 115200, 8N1. And, as you see, the bridge does work, there is data coming in the serial interface and it's logged in the weblog. After `seriallog 0` it stops. I'm not using Tasmota32 and these special functions since too long ago, please bear with me.

sfromis, 2024-10-26:
The configuration details are after "Configuration output here", first of the copy/paste question you are asked to fill in from the device.

Sure, you had something there, but as I observed, all the pins in that are not configured, hence I see no signs of serial bridge. Thus, I'm inclined to assume that you executed the commands at a different time from the issue you post about; that you did not post the configuration details **from the relevant point in time**.

And since your `Status 0` output also had `!8`, that strengthens my assumption about serial bridge not being enabled at the time of you getting the details. Maybe you did not configure serial bridge, but instead just "Serial" pins? But without relevant details in what you post, I can only make vague guesses.

kobuki, 2024-10-26:
You don't need to make vague guesses, though. You just need to understand and believe what I write. The serial bridge does work. The input appears not only in the weblog but also MQTT messages are sent. Why don't you believe me is beyond me, really. Timestamps differ in my OP because I updated the logs after your first comment. It's very hard to make a step forward this way.

However, I'll attach a new log where you can see everything in the order I do. Still, if anything's missing, I'll provide it for you, just tell me exactly what you need. Note that after issuing a `SetOption69 x` command, the device restarts itself. The option state remains after reset. I'll start the logging after a manual reset.

I still see `!8` in the drivers section every time, but the serial bridge is visibly working. This must be a bug, then?

When the bridge is running, I see MQTT messages like this:

```
tele/ma105-meter/RESULT {"SerialReceived":"E6F3E6E666CDE6F3E6E666A5CDCDE6F3E6E666A ... truncated ..."}
```

Wouldn't it indicate that the serial bridge is working? The decoded `SerialSend5` output above is still garbage, when I decode the hex string, but it shows that there is more data coming in than the few ASCII characters the log shows.

[github1.log](https://github.com/user-attachments/files/17530213/github1.log)

sfromis, 2024-10-26:
Well then, you did not configure serial bridge, and hence the feature is not active. You need to configure the relevant gpios for `SerBr Tx/Rx`.

kobuki, 2024-10-26:
No. Let me quote from the [docs](https://tasmota.github.io/docs/Commands/#serial-bridge).

- - - - -
Hardware Serial Bridge uses GPIO1 (Tx) and GPIO3 (Rx) or GPIO15 (Tx) and GPIO13 (Rx) pins of your device. Software Serial Bridge can use any other GPIO to be configured as components Serial Tx and Serial Rx (or SerBr Tx and SerBr Rx). **_If Tx and Rx components are not assigned in the Template or Module, GPIO1 and GPIO3 will be used._** Note that changing serial logging ([SerialLog](https://tasmota.github.io/docs/Commands/#seriallog) 0) will disable the hardware Serial Bridge.

Information received by Tasmota over the serial bridge is captured automatically. **_Before data will be received, a properly formatted [SerialSend<x> or SSerialSend<x>](https://tasmota.github.io/docs/Commands/#serialsend) command must be executed._** This must be done any time the device restarts (e.g., via a System#Boot triggered rule). This command is required in order to set how the expected serial data will be formatted and interpreted (i.e., which option).
- - - - -

There is no mention of `SerBr`. But I have tried setting the pins to `Serial RX/TX` and also `SerBr Rx/Tx`, with no change in results.

Let me ask again: **seeing the serial data coming in the weblog and also in MQTT messages - isn't it an indication that the serial bridge feature itself is working?**

Ths issule must lie elsewhere, not in my config.

Jason2866, 2024-10-26:
To be clear not providing **any** data of the issues template is not a valid issue.
You have been kindly asked to provide all data.

kobuki, 2024-10-26:
@Jason2866 of course, you're right, but I have filled in the issues template. While I think I have provided everything it asked for and provided additional details for @sfromis , I might be mistaken and something could be still missing. Could you please tell me what it is. I will provide everything if I've overlooked something. I also attached a log where you can see how the issue can be reproduced.

To re-iterate: I am willing to do everything you ask me of, but I can't see what I can add. But I will if you kindly shed some light on that. I really want to solve a problem if I started on it or see it's either my mistake or can't be fixed.

sfromis, 2024-10-26:
The text you just quoted does very explicitly mention `SerBr Tx/Rx`. The text may not be very clear (seems to combine words about hardware serial vs software serial and plain serial vs serial bridge), but those components configured is what it takes configure serial bridge, as indicated by the `Br` part of the component names. If you do not configure any sort of serial pins, you do indeed get plain serial on gpio1/3, but that's not serial *bridge*.

And having that `!8` in "Drivers" is a clear indication of the serial bridge driver not being enabled.

What I'm posting is to make sure that you are even using serial bridge at all, before which SetOption69 cannot work, but I've not checked if the inversion works when actually using the serial bridge feature.

kobuki, 2024-10-26:
Alright, I've found what the issue was, just before the last post from @sfromis .

According to the [docs](https://tasmota.github.io/docs/Commands/#serial-bridge), the serial bridge can work with both HW and SW serial. I looked at the source of the driver (`xdrv_08_serial_bridge.ino`) and it's for software serial only. `SetOption69` only relates to that, not HW serial, it appears. My apologies for the confusion created. This information might have been obvious for you, it wasn't for me. Maybe the docs could be amended and these details explained in greater detail. The docs for `SetOption69` still describe the old, deprecated function (for Tuya dimmers).

So, setting the module pins for `SerBr RX` (3), `SerBr TX` (1) (these are the HW serial pins as well) and then issuing the following commands make it work.

```
SetOption69 1
SBaudRate 115200
SSerialConfig 8N1
SSerialSend2 1111
```

Note that extra `S` in the front for SW serial. Now I get output similar to this:

```
14:49:36.316 MQT: tele/ma105-meter/RESULT = {"SSerialReceived":"/SAG5SAG-METER\r\n\r\n0-0:1.0.0(redacted)\r\n0-0:42.0.0(redacted)\r\n0-0:96.1.0(redacted)\r\n0-0:96.14.0(0002)\r\n0-0:96.3.10(1)\r\n0-0:17.0.0(24.000*kW)\r\n1-0:1.8.0(000113.244*kWh)\r\n1-0:1.8.1(000047.806*kWh)\r\n1-0:1.8.2(000065.438*kWh)\r\n1-0:1.8.3(000000.000*kWh)\r\n1-0:1.8.4(000000.000*kWh)\r\n ... "}
```

And like that, all is good.

Now, one question remains: should `SetOption69` work for HW serial bridge as well? Because the same functionality (barring inversion of RX) does work the same way: received serial data is logged in the weblog, sent to syslog (when configured) and also published as MQTT messages. This latter is the core of the serial bridge and both HW and SW mode of operation provides this, regardless of driver 08. It would then follow that for both it should enable the inverting of the received signal.

sfromis, 2024-10-26:
There is no such thing as "hardware serial bridge", and hardware/software serial is not about what components you configure. On ESP32, *every* pair of serial pins will be inherently be hardware serial, as Tasmota does not use software serial on ESP32 due to this. Commands SSerialSend vs SerialSend is not about software serial, but instead about serial bridge, with the driver for serial bridge being where the `S`-prefixed commands are implemented, also when using ESP32 with hardware serial by configuration.

On ESP8266, you instead have dedicated hardware serial pins, and Tasmota has an implementation for lower-performance software serial on other pins. The decision to use hardware/software serial is done in the serial driver, with hardcoded gpio numbers 1/3 and 15/13 for hardware serial.

Much of the confusion around wording comes from Tasmota originally only working on ESP8266 where you could not have "extra" serial pins, except with the feature added for that.

kobuki, 2024-10-26:
By HW serial bridge I meant the functionality for the bridge that uses HW serial, be it ESP32 or ESP8266. I also concluded what you stated, as in, for the ESP32, only HW serial is used and can be configured (almost) freely on pins. I'm still learning to use ESP32, I used 826x previously. However, I still think that `SetOption69` should work whichever is in use. Seeing the option having no effect at all is baffling, to say at the least.

I think we can close this now. Thanks for both of you for your time you took to respond. Actually, having this arrive to a resolution is pretty good in such a short time, considering how huge the project is right now.

closed 2024-10-26 by kobuki