arendst/Tasmota #22365: MiElHVAC Remote Temp and reverting to internal temp

mamrai1, 2024-10-27:
### PROBLEM DESCRIPTION
After sending command with remote temperature to miel hvac unit the unit correctly reports the new room temperature BUT 
the unit never reverts to internal temp.(waiting time of over 10minutes)
Tried sending  0 temp  but instead of reverting to internal temp sensor it shows 10.
(cmnd/tasmota_xxx/HVACRemoteTemp) 

only power cycling the unit fixes the problem

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mamrai1, 2024-10-27:
The library has support for external temp sensors.
Use method setRemoteTemperature(float celsius)
Send setRemoteTemperature(0) to revert back to internal sensors.

https://github.com/SwiCago/HeatPump/discussions/208

grzegorz914, 2024-10-27:
Yes, this is normal behavior. Looks like our driver have not implemented this function now. I will add it but I think will be better revert to Room Temp after some time if Remote Temp stop updating. What do You think?

grzegorz914, 2024-10-27:
Please try to send `clear` instead `0`

mamrai1, 2024-10-27:
It does work with sending ‘clear’!!
Now i think we need need a time period to automatically revert if remote temp not updating…

grzegorz914, 2024-10-27:
I will add it.

mamrai1, 2024-10-29:
HVACRemoteTempClearTime command works as expected when sending time in ms but the command without a number does not show the saved time.

screenshot:
22:41:02.291 CMD: HVACRemoteTempClearTime 60000
22:41:02.300 MQT: stat/tasmota_AD9FA1/RESULT = {"HVACRemoteTempClearTime":60000}
22:41:11.266 CMD: HVACRemoteTempClearTime
22:41:11.274 MQT: stat/tasmota_AD9FA1/RESULT = {"Command":"Error","Input":"HVACREMOTETEMPCLEARTIME"}

@grzegorz914 Excellent work by the way!

grzegorz914, 2024-10-29:
Empty command is not supported, will not display current time. I can add display current set time to the sensor output

grzegorz914, 2024-10-29:
Will add it in this PR #22379

mamrai1, 2024-10-29:
Perfect!Sent from my iPhone

grzegorz914, 2024-10-29:
Added, wait to be merged

mamrai1, 2024-10-29:
thanks. one last question:
setting a new time with HVACRemoteTempClearTime command shouldn't survive a restart?
now after restarting it reverts to 10000

grzegorz914, 2024-10-29:
Please read this comment from @arendst  https://github.com/arendst/Tasmota/pull/22370#issuecomment-2441259392