arendst/Tasmota #22367: Device ETH does not work in Release 14.3.0 of "tasmota32-zbbrdgpro.bin"

pstzh, 2024-10-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._I'm using [tasmota32-zbbrdgpro.bin](https://ota.tasmota.com/tasmota32/release/tasmota32-zbbrdgpro.bin) for my Athom Zigbee Gateway connected my wire (ethernet). It worked perfect until Relese 14.2.0. After upgrade to Releses 14.3.0, the ZB-Gatewywas only available via Wifi - Ethernet was enabled (according console) but did not work. - Upgrade to latest development Release 14.3.0.2 did not help.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____Athom Zigbee Gateway (ZG01)
- [ x] Tasmota binary firmware version number used: _____14.3.0, 14.3.0.2
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: _____Firmware Upgrade via Web-Interface
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`: 18:35:04.362 CMD: Backlog Template; Module; GPIO 255
18:35:04.379 RSL: RESULT = {"NAME":"Athom Zigbee Bridge","GPIO":[32,0,0,0,0,0,0,0,3552,0,3584,544,0,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,1,5792,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
18:35:04.610 RSL: RESULT = {"Module":{"0":"Athom Zigbee Bridge"}}
18:35:04.868 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"3552":"Zigbee Tx"},"GPIO13":{"0":"None"},"GPIO14":{"3584":"Zigbee Rx"},"GPIO15":{"544":"LedLink"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"5792":"Zigbee Rst1"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
- [ ] 
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:45:31.218 CMD: status 0
18:45:31.222 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_95EDE8","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:45:31.231 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:15:09","StartupUTC":"2024-10-27T17:30:22","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2024-10-27T16:51:53","SaveCount":23}}
18:45:31.238 RSL: STATUS2 = {"StatusFWR":{"Version":"14.3.0.2(faf6e66-zbbrdgpro)","BuildDateTime":"2024-10-27T16:35:27","Core":"3_1_0","SDK":"5.3.1.241016","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"389/699"}}
18:45:31.243 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Lazy-Net",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","0405C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000001"]}}
18:45:31.256 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1501,"Free":354,"Heap":157,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":8192,"FlashSize":8192,"FlashChipId":"1740C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","0F8087CE","00548001","000000C3","00000000","C0000881","40020080","00200000","54000808","00000080","00000004"],"Drivers":"1,2,!4,!8,9,10,12,!16,!20,!21,23,!24,29,!35,38,!41,50,52,!59,62,!68,82,!121","Sensors":"1,2,5,6,127","I2CDriver":""}}
18:45:31.264 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-95EDE8-3560","IPAddress":"192.168.240.97","Gateway":"192.168.240.9","Subnetmask":"255.255.255.0","DNSServer1":"192.168.240.9","DNSServer2":"62.2.17.60","Mac":"94:E6:86:95:ED:E8","IP6Global":"","IP6Local":"fe80::96e6:86ff:fe95:ede8","Ethernet":{"Hostname":"tasmota-95EDE8-3560-eth","IPAddress":"0.0.0.0","Gateway":"192.168.240.9","Subnetmask":"255.255.255.0","DNSServer1":"192.168.240.9","DNSServer2":"62.2.17.60","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:45:31.270 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_95EDE8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:45:31.274 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-10-27T17:45:31Z","Local":"2024-10-27T18:45:31","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:30","Sunset":"17:37"}}
18:45:31.282 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-10-27T18:45:31","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
18:45:31.289 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-10-27T18:45:31","Uptime":"0T00:15:09","UptimeSec":909,"Heap":159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":37},"Wifi":{"AP":1,"SSId":"Lazy-Net","BSSId":"C8:7F:54:BE:6D:90","Channel":2,"Mode":"HT40","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}}
18:45:31.864 RSL: STATE = {"Time":"2024-10-27T18:45:31","Uptime":"0T00:15:09","UptimeSec":909,"Heap":161,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":37},"Wifi":{"AP":1,"SSId":"Lazy-Net","BSSId":"C8:7F:54:BE:6D:90","Channel":2,"Mode":"HT40","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:45:31.871 RSL: SENSOR = {"Time":"2024-10-27T18:45:31","ESP32":{"Temperature":53.3},"TempUnit":"C"}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_Install Firmware above 14.2.0 on Athom Zigbee Bridge and try to connect via Ethernet

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._Ethernet connection works

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
After some trials with self compilation - I could make Ethernet working in zbbrdgpro firmware with leaving away (commenting out) 
 the -DFRAMEWORK_ARDUINO_ITEAD - It works for my Athom Zigbee Gateway but will probably not work for other Zigbee Bridges.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-10-27:
Strange. @Jason2866 any hint what `FRAMEWORK_ARDUINO_ITEAD` does?

Jason2866, 2024-10-27:
The difference is maybe, in the standard framework the PSRAM GPIOs are "freed" after check for PSRAM. Could it be that this is not done or not working when the framework ITEAD is used where the PSRAM GPIOs are changed to `CONFIG_D0WD_PSRAM_CLK_IO=5 and CONFIG_D0WD_PSRAM_CS_IO=18`?

s-hadinger, 2024-10-27:
I don't have the board at hand until in a few days. That's worth digging indeed

Jason2866, 2024-10-27:
@s-hadinger Will this done with changed PSRAM GPIOs?
https://github.com/arendst/Tasmota/blob/faf6e66e322d8d2858de39c822eee2bf5e2522bd/tasmota/tasmota.ino#L404-L424
Probably not since PSRAM is found.

arendst, 2024-10-28:
@s-hadinger I have one. Let me chk.

arendst, 2024-10-28:
Yep this fixed it:
```
  // restore GPIO16/17 if no PSRAM is found
  if (!FoundPSRAM()) {
    // test if the CPU is not pico
    uint32_t pkg_version = bootloader_common_get_chip_ver_pkg();
    if (pkg_version <= 3) {         // D0WD, S0WD, D2WD
      gpio_reset_pin(GPIO_NUM_16);  // D0WD_PSRAM_CS_IO
      gpio_reset_pin(GPIO_NUM_17);  // D0WD_PSRAM_CLK_IO

      gpio_reset_pin(GPIO_NUM_5);   // D0WD_PSRAM_CLK_IO
      gpio_reset_pin(GPIO_NUM_18);  // D0WD_PSRAM_CS_IO

      // IDF5.3 fix esp_gpio_reserve used in init PSRAM
      esp_gpio_revoke(BIT64(GPIO_NUM_5) | BIT64(GPIO_NUM_16) | BIT64(GPIO_NUM_17) | BIT64(GPIO_NUM_18));
    }
  }
```

Now I have to find a way how to detect which pins are used specifically for PSRAM....

Notice good bootlog:
```
00:00:00.002 HDW: ESP32-D0WD-V3 v3.0 
00:00:00.038 UFS: FlashFS mounted with 5424 kB free
00:00:00.053 CFG: Loaded from File, Count 35
00:00:00.063 QPC: Count 1
00:00:00.066 TYA: Active=0
00:00:00.068 ROT: Mode 1
00:00:00.102 CFG: No '*.autoconf' file found
00:00:00.105 BRY: Berry initialized, RAM used 3478 bytes
00:00:00.112 BRY: No 'preinit.be'
00:00:00.115 ZIG: Using Hardware Serial UART2
00:00:00.115 SRC: Restart
00:00:00.117 Project tasmota - AthomZbBrdg1 Version 14.3.0.2(zbbrdgpro)-3_1_0(2024-10-28T10:50:01)
00:00:01.690 ETH: Attempting connection...
00:00:01.692 ETH: Connected at 100Mbps Full Duplex, Mac 94:E6:86:92:22:8F, Hostname athomzbbrdg1-eth
00:00:01.700 ETH: IPv4 192.168.2.152, mask 255.255.255.0, gateway 192.168.2.254
00:00:01.740 ZIG: Loading Zigbee data
00:00:01.746 ZIG: No Zigbee device information
00:00:01.749 ZIG: No Zigbee device data
00:00:01.751 ZIG: ZbLoad '<internal_plugin>' loaded successfully
00:00:01.840 BRY: No 'autoexec.be'
00:00:01.921 RTC: UTC 2024-10-28T09:52:45Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
10:52:45.000 RTC: Synced by NTP
10:52:45.126 WIF: Attempting connection...
10:52:46.000 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode HT40 as athomzbbrdg1...
10:52:46.003 ETH: IPv6 Local fe80::96e6:86ff:fe92:228f
10:52:46.016 HTP: Web server active on athomzbbrdg1-eth with IP address 192.168.2.152
10:52:46.087 WIF: IPv4 192.168.2.143, mask 255.255.255.0, gateway 192.168.2.254
10:52:47.534 WIF: Connected
10:52:47.784 MQT: Attempting connection...
10:52:47.812 MQT: Connected
10:52:47.816 MQT: tele/athomzbbrdg1/LWT = Online (retained)
10:52:47.817 MQT: cmnd/athomzbbrdg1/POWER = 
10:52:47.818 MQT: Subscribe to cmnd/athomzbbrdg1/#
10:52:47.820 MQT: Subscribe to cmnd/tasmotas/#
10:52:47.821 MQT: Subscribe to cmnd/DVES_92228C_fb/#
10:52:47.823 MQT: tele/athomzbbrdg1/INFO1 = {"Info1":{"Module":"Athom Zigbee Bridge","Version":"14.3.0.2(zbbrdgpro)","FallbackTopic":"cmnd/DVES_92228C_fb/","GroupTopic":"cmnd/tasmotas/"}}
10:52:47.828 MQT: tele/athomzbbrdg1/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"athomzbbrdg1","IPAddress":"192.168.2.143","IP6Global":"","IP6Local":"fe80::96e6:86ff:fe92:228c","Ethernet":{"Hostname":"athomzbbrdg1-eth","IPAddress":"192.168.2.152"}}}
10:52:47.831 MQT: tele/athomzbbrdg1/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":12}}
10:52:48.055 WIF: IPv6 Local fe80::96e6:86ff:fe92:228c
10:52:50.055 WIF: IPv6 Global fd84:7996:2cbf:1349:96e6:86ff:fe92:228c
10:52:50.958 QPC: Reset
10:52:51.960 MQT: tele/athomzbbrdg1/STATE = {"Time":"2024-10-28T10:52:51","Uptime":"0T00:00:07","UptimeSec":7,"Heap":143,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"HT40","RSSI":100,"Signal":-16,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:52:52.951 APP: Boot Count 12
10:52:53.430 CFG: Saved, Count 36, Bytes 4096
10:53:00.276 ZIG: rebooting ZNP device
```

Jason2866, 2024-10-28:
@arendst 
> Now I have to find a way how to detect which pins are used specifically for PSRAM....

You have this in sdkconfig
```
CONFIG_D0WD_PSRAM_CLK_IO=5
CONFIG_D0WD_PSRAM_CS_IO=18
```
or GPIO 16/17 there when standard framework is used.

arendst, 2024-10-28:
Adding some debug info shows that the follwoing GPIO's are reserved:
```
  theo_map1 = esp_gpio_revoke(0);
...
  AddLog(LOG_LEVEL_INFO, PSTR("*** TEO: theo_map %_X"), &theo_map1);
```
shows FFFFFFFCF1040FE0 = GPIO's 5,6,7,8,9,10,11,18,24,28,29,30,31,34,35,36,37,38,39.

I think it's safe to revoke all four GPIO's in case no PSRAM is found. So final fix will be:
```
  // restore GPIO5/16/17/18 if no PSRAM is found which may be used by Ethernet among others
  if (!FoundPSRAM()) {
    // test if the CPU is not pico
    uint32_t pkg_version = bootloader_common_get_chip_ver_pkg();
    if (pkg_version <= 3) {         // D0WD, S0WD, D2WD
      // Normal PSRAM connection
      gpio_reset_pin(GPIO_NUM_16);  // D0WD_PSRAM_CS_IO
      gpio_reset_pin(GPIO_NUM_17);  // D0WD_PSRAM_CLK_IO
      // iTEAD PSRAM connection
      gpio_reset_pin(GPIO_NUM_5);   // D0WD_PSRAM_CLK_IO
      gpio_reset_pin(GPIO_NUM_18);  // D0WD_PSRAM_CS_IO
      // IDF5.3 fix esp_gpio_reserve used in init PSRAM
      esp_gpio_revoke(BIT64(GPIO_NUM_5) | BIT64(GPIO_NUM_16) | BIT64(GPIO_NUM_17) | BIT64(GPIO_NUM_18));
    }
  }
```

arendst, 2024-10-28:
Oh I just missed your amendmend. Let me test these defines.....

Jason2866, 2024-10-28:
Isn't it still failing when PSRAM is found when GPIOs 5 and 18 are used?

arendst, 2024-10-28:
The PSRAM test is in IDF file ``esp_psram_impl_quad.c``. Here it does the reserve without the revoke in function ``psram_gpio_config``.

So if it finds PSRAM, ethernet will fail.

Anyway, your suggestion to use sdkconfig defines works out well. This will be the final fix:
```
  // restore GPIO5/16/17/18 if no PSRAM is found which may be used by Ethernet among others
  if (!FoundPSRAM()) {
    // test if the CPU is not pico
    uint32_t pkg_version = bootloader_common_get_chip_ver_pkg();
    if (pkg_version <= 3) {         // D0WD, S0WD, D2WD
      gpio_reset_pin((gpio_num_t)CONFIG_D0WD_PSRAM_CS_IO);   // D0WD_PSRAM_CS_IO
      gpio_reset_pin((gpio_num_t)CONFIG_D0WD_PSRAM_CLK_IO);  // D0WD_PSRAM_CLK_IO
      // IDF5.3 fix esp_gpio_reserve used in init PSRAM
      esp_gpio_revoke(BIT64(CONFIG_D0WD_PSRAM_CS_IO) | BIT64(CONFIG_D0WD_PSRAM_CLK_IO));
    }
  }
```

Jason2866, 2024-10-28:
Yes, all not needed libs for zbbrdgpro are removed so it compiles very fast.

Jason2866, 2024-10-28:
@arendst Close to solution. Since no PSRAM is used in safeboot variants it fails there.
For safeboot the complete check for PSRAM can be left out. Even smaller code size :-)

arendst, 2024-10-28:
Bugger. I just noticed too.

Will change.

sfromis, 2024-10-28:
Finally works again for me.

pstzh, 2024-10-28:
works also for me - thanks for the quick solution!

closed 2024-10-28 by pstzh

shlomishm, 2024-11-05:
@arendst, I attempted to compile Tasmota for the ESP32 using the Arduino framework, but the build failed due to a missing declaration for esp_gpio_revoke.

I'm receiving the following error:
C:/Users/.../tasmota/tasmota.ino:421:7: error: 'esp_gpio_revoke' was not declared in this scope
 421 |       esp_gpio_revoke(BIT64(CONFIG_D0WD_PSRAM_CS_IO) | BIT64(CONFIG_D0WD_PSRAM_CLK_IO));

After commenting out this line, the compilation completes successfully.

I’m new to this development environment, so it’s possible I may have overlooked something in my setup. However, as far as I know, esp_gpio_revoke is not included in the Arduino core for ESP32, which could be the source of this issue. Any insights or guidance on this would be greatly appreciated!