arendst/Tasmota #22369: WT32-ETH01 v1.4 board  - infinite reset loop after flashing via web flasher.

K1WIZ, 2024-10-27:
### PROBLEM DESCRIPTION
After flashing WT32-ETH01, and device resets from web flasher, ethernet does not work nor does wifi (ie. not broadcasting "TasmotaXXX" AP.

Seeing this message repeating in the serial log:

00:00:00.002 HDW: ESP32-D0WD v1.0 
00:00:00.009 UFS: FlashFS mounted with 308 kB free
00:00:00.015 CFG: Loaded from File, Count 2
00:00:00.017 FRC: Some settings have been reset (18)
00:00:00.027 Project tasmota - Tasmota Version 14.3.0(release-tasmota32)-3_1_0(2024-10-15T08:21:27)
00:00:00.379 WIF: WifiManager active for 3 minutes
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13168
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.002 HDW: ESP32-D0WD v1.0 
00:00:00.009 UFS: FlashFS mounted with 308 kB free
00:00:00.015 CFG: Loaded from File, Count 2
00:00:00.017 FRC: Some settings have been reset (19)
00:00:00.027 Project tasmota - Tasmota Version 14.3.0(release-tasmota32)-3_1_0(2024-10-15T08:21:27)
00:00:00.379 WIF: WifiManager active for 3 minutes
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13168
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.002 HDW: ESP32-D0WD v1.0 
00:00:00.009 UFS: FlashFS mounted with 308 kB free
00:00:00.015 CFG: Loaded from File, Count 2
00:00:00.017 FRC: Some settings have been reset (20)
00:00:00.027 Project tasmota - Tasmota Version 14.3.0(release-tasmota32)-3_1_0(2024-10-15T08:21:27)
00:00:00.379 WIF: WifiManager active for 3 minutes
ets Jun  8 2016 00:22:57

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
flash WT32-ETH01 v1.4 via web flasher, remove IO0 jumper, powercycle, and view serial console. At time of this issue, version was:
Project tasmota - Tasmota Version 14.3.0(release-tasmota32)-3_1_0(2024-10-15T08:21:27)

### EXPECTED BEHAVIOUR
I expected to see tasmota boot on the device and spawn an initial AP "TasmotaXXXX" with which to use to setup the ethernet port connectivity.  

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-10-27:
This bootloop is typical of a failing power supply. How do you power the device?

K1WIZ, 2024-10-27:
5 volts via the 5v pin.   lab power supply.

s-hadinger, 2024-10-27:
Can you try a complete flash erase and flash again? Does the bootloop happen before you configure ethernet?

K1WIZ, 2024-10-27:
it was before configuring ethernet.   will try a reflash.

s-hadinger, 2024-10-27:
Then it may be a 3.3V vreg problem on the board. You can try lowering `wifipower` to 5 or 10. Unfortunately it's difficult to type the command fast enough before the crash. Better option is to compile a version with lower wifi power

K1WIZ, 2024-10-27:
Ok did a complete erase and reflash via the web flasher and the tasmota (english).   Board is up now, and I was able to configure ethernet, and turn off wifi.   Strange thing is that I'm seeing a lot of dropped packets:

`
CORE-SW#pi 10.50.150.135 r 100
Type escape sequence to abort.
Sending 100, 100-byte ICMP Echos to 10.50.150.135, timeout is 2 seconds:
!.!!!!..!!!!!!!!!!!!.!!!!!!!!!!!!.!!!!!!!!!!!.!!.!!!!!.!!!!!!!!!!!!!!.
!!!!!!!!!!!!!!!.!.!!!!!...!!!!
Success rate is 86 percent (86/100), round-trip min/avg/max = 1/3/16 ms
`

Any idea why?  Plugged the same cable into a laptop and zero loss.

s-hadinger, 2024-10-27:
No. This is not normal. But I have doubts about the quality of your board

s-hadinger, 2024-10-27:
Just an idea out of the blue. Check if the port is in full duplex both on Tasmota and on your switch

K1WIZ, 2024-10-27:
switch shows:

CORE-SW#sh int status | i 1/0/39  
Gi1/0/39  TEST_PORT          connected    50         a-full  a-100 10/100/1000BaseTX

How do I check the board?  (I don't know what console commands for tasmota)

K1WIZ, 2024-10-27:
on the board I see in console:

00:00:00.045 Project tasmota - wt32eth01 Version 14.3.0(release-tasmota32)-3_1_0(2024-10-15T08:21:27)
00:00:01.873 ETH: Connected at 100Mbps Full Duplex, Mac 94:3C:C6:3D:2B:97, Hostname w32eth01-eth
22:03:47.365 HTP: Web server active on w32eth01-eth with IP address 10.50.150.135
22:03:48.441 RSL: INFO1 = {"Info1":{"Module":"WT32_ETH01","Version":"14.3.0(release-tasmota32)","FallbackTopic":"cmnd/DVES_3D2B94_fb/","GroupTopic":"cmnd/tasmotas/"}}

So the 100 full auto seems to agree with the switch

K1WIZ, 2024-10-27:
pinging the wifi interface:
CORE-SW#pi 10.50.50.139 r 100
Type escape sequence to abort.
Sending 100, 100-byte ICMP Echos to 10.50.50.139, timeout is 2 seconds:
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Success rate is 100 percent (100/100), round-trip min/avg/max = 1/9/84 ms

Comparing the ethernet (garbage):   (for this application, ethernet is desired, but with this much loss, will be unreliable)

CORE-SW#pi 10.50.150.135 r 100
Type escape sequence to abort.
Sending 100, 100-byte ICMP Echos to 10.50.150.135, timeout is 2 seconds:
!!!!!!!!!!.!!!!!!!!!!!!!!!!!!!!!!!!!!..!.!..!!!!.!!!!!!!!!!!!!!!!!!!!!
!!!!!!.!!.!!!!!!!!!!!!!!!!!.!.
Success rate is 89 percent (89/100), round-trip min/avg/max = 1/3/17 ms

Jason2866, 2024-10-27:
I have the previous rev. of the board. Only after adding Caps the board started to work reliable. Big cap in 3.3V rail. And 100nf Filter cap directly soldered on the esp32 module between GND and 3.3V input.

K1WIZ, 2024-10-27:
closing this issue (for the reset loop)   continuing network related issue on new issue:
https://github.com/arendst/Tasmota/issues/22372

closed 2024-10-27 by K1WIZ