arendst/Tasmota #22371: shelly_plus_pm_mini support with DALI breaks from 14.2 to 14.3

stefanbode, 2024-10-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
https://templates.blakadder.com/shelly_plus_pm_mini.html does not show power readings after upgrade. Instead the console log ile shows following line
"DALI":{"Power":"OFF","Dimmer":0,"Address":255,"Command":44}}

Reverting back to 14.2 and restore Config file fixes the problem

Standard Configuration https://templates.blakadder.com/shelly_plus_pm_mini.html

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-10-28:
Pls provide more information.

What version fails? And if so provide the result from command ``status 0``.

As USE_DALI is enabled for the first time in version v14.3.0.2 I don't see how v4.3.0 could end up like you report.

arendst, 2024-10-28:
Oh dear! Just verified your experience! Investigating.....

arendst, 2024-10-28:
Thx. Fixed and Tested.

closed 2024-10-28 by arendst