arendst/Tasmota #22378: 14.3.0.2 on ESP32 breaks matter support with Google Home

maltic42, 2024-10-28:
### PROBLEM DESCRIPTION
With Tasmota 14.3.0.2 on ESP32 (virtual) matter devices can be commissioned but stay as offline in Google Home. The device works in Homekit. In 14.3.0 the same configuration makes no problems at all, devices are commissioned and are then working.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nothing else
- [x] Tasmota binary firmware version number used: 14.3.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: web installer w/ Chrome
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:33:22.110 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
19:33:22.334 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
19:33:22.541 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:33:57.806 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9A4B90","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:33:57.862 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:19:26","StartupUTC":"2024-10-28T18:14:31","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2024-10-28T19:10:33","SaveCount":19}}
19:33:57.900 RSL: STATUS2 = {"StatusFWR":{"Version":"14.3.0.2(27db263-tasmota32)","BuildDateTime":"2024-10-28T13:08:59","Core":"3_1_0","SDK":"5.3.1.241016","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.1","CR":"401/699"}}
19:33:57.917 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TOAD",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000020"]}}
19:33:57.955 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2002,"Free":877,"Heap":99,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000014"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
19:33:58.029 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9A4B90-2960","IPAddress":"192.168.188.37","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.188.1","DNSServer2":"2003:f2:9f3f:b400:6b4:feff:fe57:302c","Mac":"08:3A:8D:9A:4B:90","IP6Global":"2003:f2:9f3f:b400:a3a:8dff:fe9a:4b90","IP6Local":"fe80::a3a:8dff:fe9a:4b90","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.188.1","DNSServer2":"2003:f2:9f3f:b400:6b4:feff:fe57:302c","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
19:33:58.090 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9A4B90","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:33:58.107 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-10-28T18:33:58Z","Local":"2024-10-28T19:33:58","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:31","Sunset":"17:35"}}
19:33:58.135 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-10-28T19:33:58"}}
19:33:58.157 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-10-28T19:33:58","Uptime":"0T00:19:27","UptimeSec":1167,"Heap":104,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":27,"Objects":313},"Wifi":{"AP":1,"SSId":"TOAD","BSSId":"04:B4:FE:57:30:2E","Channel":1,"Mode":"HT40","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
19:35:59.539 HTP: Tools
19:36:00.578 HTP: Console
19:36:03.650 CMD: WebLog 4
19:36:03.651 SRC: WebConsole from 192.168.188.90
19:36:03.653 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 4, Data '4'
19:36:03.656 RSL: RESULT = {"WebLog":4}
19:36:07.304 HTP: Tools
19:36:08.435 HTP: Main Menu
19:36:08.617 BRY: GC from 32412 to 22088 bytes, objects freed 160/265 (in 2 ms) - slots from 495/562 to 242/562
19:36:08.643 BRY: GC from 32372 to 22104 bytes, objects freed 87/262 (in 3 ms) - slots from 262/562 to 241/562
19:36:08.666 BRY: GC from 32396 to 22121 bytes, objects freed 90/261 (in 2 ms) - slots from 253/562 to 237/562
19:36:08.691 BRY: GC from 32422 to 22144 bytes, objects freed 89/262 (in 2 ms) - slots from 253/562 to 241/562
19:36:11.182 BRY: GC from 32422 to 22166 bytes, objects freed 90/265 (in 2 ms) - slots from 345/562 to 241/562
19:36:11.206 BRY: GC from 32457 to 22136 bytes, objects freed 94/262 (in 2 ms) - slots from 261/562 to 241/562
19:36:11.233 BRY: GC from 32391 to 22080 bytes, objects freed 86/262 (in 2 ms) - slots from 257/562 to 241/562
19:36:11.259 BRY: GC from 32388 to 22056 bytes, objects freed 93/262 (in 2 ms) - slots from 257/562 to 241/562
19:36:13.839 BRY: GC from 32363 to 22204 bytes, objects freed 95/265 (in 3 ms) - slots from 356/562 to 241/562
19:36:13.864 BRY: GC from 32532 to 22158 bytes, objects freed 95/262 (in 2 ms) - slots from 261/562 to 241/562
19:36:13.889 BRY: GC from 32474 to 22104 bytes, objects freed 88/262 (in 2 ms) - slots from 257/562 to 241/562
19:36:13.914 BRY: GC from 32378 to 22096 bytes, objects freed 90/262 (in 2 ms) - slots from 257/562 to 241/562
19:36:16.511 BRY: GC from 32432 to 22040 bytes, objects freed 104/262 (in 2 ms) - slots from 365/562 to 241/562
19:36:16.531 BRY: GC from 32317 to 22182 bytes, objects freed 88/262 (in 2 ms) - slots from 253/562 to 241/562
19:36:16.556 BRY: GC from 32488 to 22130 bytes, objects freed 89/262 (in 2 ms) - slots from 257/562 to 241/562
[... 571 lines cut ...]
19:37:21.895 MTR: decode header: local_session_id=35508 message_counter=220904873
19:37:21.899 MTR: .          Duplicate encrypted message = 220904873 counter=220904873
19:37:21.913 MTR: UDP received from [192.168.188.45]:49653
[... cut ...]
### TO REPRODUCE, 
Start with a fresh setup, enable matter, force static endpoints, add a virtual device type Light 0, start commissioning, start Google Home on iOS, add matter device, go through the wizard

### EXPECTED BEHAVIOUR
The device is created in Google Home and works

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

maltic42, 2024-10-28:
`19:37:21.975 MTR: >Command   ( 35508) [00]0030/0000 ArmFailSafe 
19:37:21.984 BRY: GC from 44669 to 33933 bytes, objects freed 99/390 (in 3 ms) - slots from 508/592 to 371/562
19:37:21.989 MTR: <Replied   ( 35508) [00]0030/0001 ArmFailSafeResponse
19:37:22.013 MTR: <snd       ( 35508) id=100397622 exch=18155 rack=220904875
19:37:22.019 MTR: sending packet to '[192.168.188.45]:49653'
19:37:22.041 MTR: UDP received from [192.168.188.45]:49653
19:37:22.045 MTR: decode header: local_session_id=35508 message_counter=220904876
19:37:22.052 MTR: >          Decrypted message: protocol_id:0 opcode=16 exchange_id=18155
19:37:22.055 MTR: .          Removed packet from sending list id=100397622
19:37:23.086 MTR: UDP received from [fe80::9:50d3:3f9b:ea82]:62774
19:37:23.097 MTR: >Received  (     0) CASE_Sigma1 from [fe80::9:50d3:3f9b:ea82]:62774
19:37:23.114 MTR: SEARCHING: destinationId=9E32BC177E9C88DD9F159FDA215A005D3216EAB3CF0138CA80252D6CAFB83767
19:37:23.122 BRY: GC from 44221 to 34052 bytes, objects freed 135/369 (in 3 ms) - slots from 520/562 to 343/562
19:37:23.154 MTR: SIGMA1: candidateDestinationId=B3073013CCB712921BB031316A5D4CBC8539FFC8060B3D4B73D589F7F5A26465
19:37:23.187 MTR: SIGMA1: candidateDestinationId=9E32BC177E9C88DD9F159FDA215A005D3216EAB3CF0138CA80252D6CAFB83767
19:37:23.196 BRY: GC from 44360 to 33803 bytes, objects freed 149/364 (in 5 ms) - slots from 432/562 to 336/562
19:37:23.198 MTR: +Session   ( 61133) from '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:23.578 MTR: <Replied   (     0) CASE_Sigma2
19:37:23.585 BRY: GC from 44046 to 38820 bytes, objects freed 57/389 (in 3 ms) - slots from 410/562 to 349/562
19:37:23.588 MTR: New Connection (CASE id=61133) from [fe80::9:50d3:3f9b:ea82]:62774
19:37:23.603 MTR: error sending packet to '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:23.756 MTR: UDP received from [fe80::9:50d3:3f9b:ea82]:62774
19:37:23.763 MTR: .          Duplicate unencrypted message = 205240369 ref = 205240369
19:37:23.925 MTR: .          Resending packet id=1296624
19:37:23.929 MTR: error sending packet to '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:24.271 MTR: .          Resending packet id=1296624
19:37:24.274 MTR: error sending packet to '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:24.288 WIF: Checking connection...
19:37:24.494 MTR: UDP received from [fe80::9:50d3:3f9b:ea82]:62774
19:37:24.501 MTR: .          Duplicate unencrypted message = 205240369 ref = 205240369
19:37:24.619 MTR: .          Resending packet id=1296624
19:37:24.622 MTR: error sending packet to '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:25.173 MTR: .          Resending packet id=1296624
19:37:25.176 MTR: error sending packet to '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:25.416 MTR: UDP received from [fe80::9:50d3:3f9b:ea82]:62774
19:37:25.422 MTR: .          Duplicate unencrypted message = 205240369 ref = 205240369
19:37:26.131 MTR: .          Resending packet id=1296624
19:37:26.134 MTR: error sending packet to '[fe80::9:50d3:3f9b:ea82]:62774'
19:37:37.403 MTR: UDP received from [192.168.188.45]:49653
19:37:37.410 MTR: .          Duplicate unencrypted message = 205240371 ref = 205240371
19:37:37.414 MTR: UDP received from [192.168.188.45]:49653
19:37:37.418 MTR: decode header: local_session_id=22993 message_counter=23758076
19:37:37.436 MTR: >          Decrypted message: protocol_id:1 opcode=3 exchange_id=18158
19:37:37.452 MTR: <Ack*      ( 22993) ack=23758076 id=55621824 
19:37:37.459 MTR: sending packet to '[192.168.188.45]:49653'
19:37:37.475 BRY: GC from 58525 to 37386 bytes, objects freed 154/413 (in 5 ms) - slots from 602/638 to 385/638
19:37:37.477 MTR: >Subscribe ( 22993) [**]****/**** (min=0, max=60, keep=1) sub=34781 fabric_filtered=false
19:37:37.491 MTR: >Read_Attr ( 22993) [**]****/****
19:37:37.495 MTR: >Read_Event( 22993) [**]****/** 
19:37:37.576 BRY: GC from 47708 to 37592 bytes, objects freed 158/397 (in 4 ms) - slots from 563/638 to 360/623
19:37:37.817 BRY: GC from 47773 to 41227 bytes, objects freed 81/427 (in 3 ms) - slots from 511/623 to 361/623
19:37:37.885 MTR: sending packet to '[192.168.188.45]:49653'
19:37:37.890 MTR: UDP received from [192.168.188.45]:49653
19:37:37.897 MTR: .          Removed packet from sending list id=185545512
19:37:37.902 MTR: >rcv Ack   ( 22993) rid=205240372 exch=0 ack=18157 185545512from [192.168.188.45]:49653
19:37:38.012 MTR: UDP received from [192.168.188.45]:49653
19:37:38.016 MTR: decode header: local_session_id=22993 message_counter=23758077
19:37:38.023 MTR: >          Decrypted message: protocol_id:1 opcode=1 exchange_id=18158
19:37:38.025 MTR: .          Removed packet from sending list id=55621825
19:37:38.042 MTR: >Sub_OK    ( 22993) sub=34781
19:37:38.049 MTR: <Ack*      ( 22993) ack=23758077 id=55621827 
19:37:38.055 MTR: sending packet to '[192.168.188.45]:49653'
19:37:38.076 BRY: GC from 51488 to 37492 bytes, objects freed 109/385 (in 4 ms) - slots from 452/623 to 360/623
19:37:38.208 BRY: GC from 47741 to 38384 bytes, objects freed 124/390 (in 4 ms) - slots from 497/623 to 363/623
19:37:38.334 BRY: GC from 48660 to 38552 bytes, objects freed 150/386 (in 4 ms) - slots from 508/623 to 362/623
19:37:38.405 MTR: sending packet to '[192.168.188.45]:49653'
19:37:38.411 MTR: UDP received from [192.168.188.45]:49653
19:37:38.415 MTR: decode header: local_session_id=22993 message_counter=23758078
19:37:38.422 MTR: >          Decrypted message: protocol_id:1 opcode=8 exchange_id=18159
[... cut ...]
19:40:18.927 MTR: error sending packet to '[fe80::4ad6:d5ff:fe67:6c4b]:5540'
19:40:19.805 MTR: UDP received from [fe80::4ad6:d5ff:fe67:6c4b]:5540
19:40:19.813 MTR: .          Duplicate unencrypted message = 123575509 ref = 123575509
19:40:20.383 MTR: .          Resending packet id=61253051
19:40:20.386 MTR: error sending packet to '[fe80::4ad6:d5ff:fe67:6c4b]:5540'
19:40:22.806 MTR: .          (     0) Unacked packet '[fe80::4ad6:d5ff:fe67:6c4b]:5540' msg_id=61253051
19:40:23.958 MTR: -Session   (     0) (removed)
19:40:27.920 MTR: -Session   (     0) (removed)`

s-hadinger, 2024-10-29:
I will have a look in the following days. For the future, loglevel 3 is enough for this kind of problem. Loglevel 4 is too verbose.

maltic42, 2024-10-29:
> I will have a look in the following days.

Thanks!!

Jason2866, 2024-10-29:
> Hmmm `19:39:58.805 MTR: UDP received from [fe80::c0af:3d89:e933:b760]:5540` shows that the zone I'd has disappeared.
> 
> @Jason2866 this must be linked with the commit removing the zone id from String. But the problem is that sending packets seems to fail too

Most probably [the commit](https://github.com/espressif/arduino-esp32/commit/20a28b58bc3fd5ff613e2860d65e0953446f264b) we talked about.

s-hadinger, 2024-10-29:
The bad news is that this commit causes Matter to fail for all builds...

maltic42, 2024-10-29:
> The bad news is that this commit causes Matter to fail for all builds... I will try to find a work-around

Thanks a lot - that would be great!!

s-hadinger, 2024-10-29:
We should have shortly a work around. Thanks for reporting the problem

maltic42, 2024-10-30:
> The fix is included in #22384 now in development branch

@s-hadinger Excellent - I just updated and tested it - matter now works again on Google Home. Thanks!!

PR #22384 (2024-10-30): Tasmota Platform 2024.11.30 / Arduino 3.1.0.241030 / IDF 5.3.1.241024
## Description:

- Arduino core fix IPv6 zones Fixes #22378

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.8
  - [x] The code change is tested and works with Tasmota core ESP32 V.3.1.0.241023
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTIN [... cut]