arendst/Tasmota #22404: Update to 14.3 fails: Upload Failed  Not enough space

Taxxian, 2024-11-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Blitzwolf BW-SHP2 16A device using Tasmota.bin updated many times up to 12.2 with 1024KB memory.
Now trying to update to 14.3 minimal worked but the normal tasmota.bin does not upload.
It says:

Upload Failed

Not enough space

I downloaded tasmota.bin.gz and tasmota.bin multiple times, updated to tasmota 14.3 minimal multiple times, thats not the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

01:54:00.396 CMD: STATUS 0
01:54:00.402 MQT: strom/stat/RESULT = {"Command":"Unknown","Input":"STATUS 0"}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

not applicable
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I do not know if I can downgrade to 12.2 after intalling 14.3 minimal and I am hesitant to upgrade any other device to 14.3 right now because this one is at this time basically unusable. So I did not undertake any more tests because I fear bricking more devices.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-11-01:
See https://github.com/arendst/Tasmota/discussions/18983

sfromis, 2024-11-01:
To upgrade, you should use the normal `tasmota.bin.gz` instead of `tasmota.bin`; the latter is only for wired initial flash.