arendst/Tasmota #22405: Rel. 14.3 KNX: scenes are no longer working

clkempe, 2024-11-01:
Since Version 14.3 scenes are no longer processed by the device.

Here the protocol of a 14.3 device:
08:49:44.814 KNX: Received from 0/0/14 Command Write: 0.00 to KNX RX 6
08:49:44.820 RSL: RESULT = {"Event":"Done"}
08:50:10.258 KNX: Received from 0/0/14 Command Write: 0.00 to KNX RX 6
08:50:10.264 RSL: RESULT = {"Event":"Done"}

Here the protocol of a 14.2 device (same scene events):
08:49:45.473 KNX: Received from 0/0/14 Command Write: 31 to KNX SCENE RX
08:49:45.478 RSL: RESULT = {"Event":"Done"}
08:49:45.507 RUL: EVENT#KNX_SCENE=31 performs 'Power1 0'
08:49:45.513 KNX: Output 1 = 0 sent to 0/6/25
08:49:45.515 RSL: RESULT = {"POWER":"OFF"}
08:49:45.517 RSL: POWER = OFF
08:50:10.912 KNX: Received from 0/0/14 Command Write: 29 to KNX SCENE RX
08:50:10.917 RSL: RESULT = {"Event":"Done"}
08:50:10.938 RUL: EVENT#KNX_SCENE=29 performs 'Power1 1'
08:50:10.943 KNX: Output 1 = 1 sent to 0/6/25
08:50:10.946 RSL: RESULT = {"POWER":"ON"}
08:50:10.948 RSL: POWER = ON

After a downgrade to 14.2 it works again.

Jason2866, 2024-11-01:
This PR added new features and added more Slots. https://github.com/arendst/Tasmota/pull/22071

clkempe, 2024-11-01:
Ok, but why not adding them _after_ KNX SCENE RX?

clkempe, 2024-11-26:
> This PR added new features and added more Slots. #22071

No reply? So poor!

Jason2866, 2024-11-26:
There is a reply, the reason for the changed behaviour is identified and mentioned.

clkempe, 2024-11-27:
> There is a reply, the reason for the changed behaviour is identified and mentioned. Tasmota is a Open Source project, it needs someone with the time and skills to look at wanted changes. You are invited to provide a PR to change the behaviour you like.

The point is: what does that mean for the future? Whenever a new option is added to this list, "KNX Scene RX" will be the victim and moved down, meaning that all users using scenes need to reconfigure their devices? There should be an awareness to take care of having a backwards compatibility for the settings.

Jason2866, 2024-11-27:
Tasmota tries to avoid breaking changes whenever possible. That's why this is still open. Someone has to take the job ;-)

miesi, 2024-11-30:
I would like to note that there is another unexpected change in behavior in this pr:
in the drop down list it is still "KNX_ SCENE_RX" but in the rules you now have to write EVENT#KNX_SCENE==24

anyway, already migrated to 14.3 devices are broken and can not really be fixed automatically I think.

I have no clue if a simple migration step is possible to copy the content of the old slot to a new slot is possible.

Not renaming the EVENT#KNX_SCENE_RX should be easy, right?

anyway reviewing a pr is a tough thing.

mi-hol, 2024-12-04:
The inventor Theo acknowledged the issue in Discord discussion https://discord.com/channels/479389167382691863/479389167382691865/1313408499643846687 
But he needs help to fix it. 
Who takes the lead?

arendst, 2024-12-04:
As a quick fix I'll move the four new KNX_SLOTS to the end of the list. This should solve the issues observed.

arendst, 2024-12-04:
Try latest change. It restores the indexes as before. It also extents with four slots as in the original PR.

GUI drop down still needs some rework but at least the scenes should work fine now.

BTW. Users who updated their scene index need to revert their update ;-)

arendst, 2024-12-05:
Just fixed select drop down list order

[image]

Also fixed some possible input buffer overflow exceptions.

This should solve the observed errors. Pls report your results and/or close this issue.

pbrinette, 2024-12-05:
Thank you @arendst for the fix. I'm too busy to fix it myself at the moment. I will try to test it during the weekend, I hope.

miesi, 2024-12-08:
I upgraded a tasmota 2022.something to the binaries of December 7th and the rules kept working. Thanks for adjusting the code.

closed 2024-12-08 by arendst