arendst/Tasmota #22446: Button_tc missing

dingausmwald, 2024-11-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Button_tc is missing in newest verion 14.3. Or did i miss something? Docs still say set button_tc in config

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-11-09:
@arendst @Jason2866 
The touch button feature seem to be controlled by `defines SOC_TOUCH_VERSION_1` or `SOC_TOUCH_VERSION_2`
Could those be defines from core or IDF that has been removed/changed during an change of version ?

arendst, 2024-11-09:
Searching through core it looks like SOC_TOUCH_VERSION_1 / _2 have been replaced by SOC_TOUCH_SENSOR_VERSION.

I'll update Tasmota accordingly.

arendst, 2024-11-09:
Try latest dev branch. I can see the touch buttons are re-enabled but I wonder if they work as before....

Give it a try.

Jason2866, 2024-11-09:
The should. The esp32-p4 does have a different touch driver implementation
As currently P4 not supported from Tasmota not really a problem ;-)

sfromis, 2024-11-09:
On an ESP32-D0WD-V3 v3.1, `Button_tc` was indeed gone, but now appears again, and works too.

closed 2024-11-09 by Jason2866